EttusResearch/uhd #379: uhd 4.0.0.0 test failure

ZeroChaos-, 2020-09-16:
```
98% tests passed, 2 tests failed out of 81

Total Test time (real) =   0.74 sec

The following tests FAILED:
	 50 - pyranges_test (Failed)
	 52 - pychdr_parse_test (Failed)
Errors while running CTest
```
I am ready and able to troubleshoot this, please let me know what you need.  I suspect most of you already know that you can find me on freenode if desired.

ZeroChaos-, 2020-09-16:
```
theprophet uhd-4.0.0.0 # python3 host/tests/pychdr_parse_test.py
Traceback (most recent call last):
  File "host/tests/pychdr_parse_test.py", line 11, in <module>
    from uhd import chdr
ImportError: cannot import name 'chdr' from 'uhd' (/usr/lib/python3.7/site-packages/uhd/__init__.py)
theprophet uhd-4.0.0.0 # python3 host/tests/pyranges_test.py    
theprophet uhd-4.0.0.0 # echo $?
0
```
pychdr_parse_test has an obvious issue when I run it manually, I have no idea why pyranges_test fails.

Unfortunately the pychdr_parse_test is a painful one, it's trying to import something which isn't installed yet.  This isn't just a problem for me, this is a problem for everyone.  When testing a new version, this test is going to import the installed version instead of the version attempting to be tested.

ZeroChaos-, 2020-09-16:
running make test with 'PYTHON_PATH=python make test' works.  I leave this bug open and up to the discretion of the team to close or do something about.  I would recommend changing the python to makefile to ensure it never accidentally pulls the installed version for testing, but it's up to you all.

mbr0wn, 2021-09-03:
Closed by ea5a08dbc385c38688833fd04c.

mbr0wn, 2021-09-03:
Wait, that commit didn't touch this.

mbr0wn, 2021-09-03:
I can't repro this... Python paths seem to have a mind of their own. On my systems, the paths are always correct -- meaning we never test the install version, and never fail if there's nothing installed. Maybe we did fix this in the almost-a-year this has been open.
That said, I *will* add the `build/python` path to PYTHONPATH just for the expressiveness.

mbr0wn, 2021-09-03:
Patch will look something like this:
```diff
diff --git a/host/cmake/Modules/UHDUnitTest.cmake b/host/cmake/Modules/UHDUnitTest.cmake
index 519be9ca04..4e6651fd97 100644
--- a/host/cmake/Modules/UHDUnitTest.cmake
+++ b/host/cmake/Modules/UHDUnitTest.cmake
@@ -133,9 +133,12 @@ function(UHD_ADD_PYTEST test_name)
# Include ${CMAKE_BINARY_DIR}/utils/ for testing the python utils
if(APPLE)
set_tests_properties(${test_name} PROPERTIES
-            ENVIRONMENT "DYLD_LIBRARY_PATH=${CMAKE_BINARY_DIR}/lib/;PYTHONPATH=${CMAKE_SOURCE_DIR}/tests/common:${CMAKE_BINARY_DIR}/utils/")
+            ENVIRONMENT
+            "DYLD_LIBRARY_PATH=${CMAKE_BINARY_DIR}/lib/;PYTHONPATH=${UHD_BINARY_DIR}/python:${CMAKE_SOURCE_DIR}/tests/common:${CMAKE_BINARY_DIR}/utils/")
else()
set_tests_properties(${test_name} PROPERTIES
-            ENVIRONMENT "LD_LIBRARY_PATH=${CMAKE_BINARY_DIR}/lib/;PYTHONPATH=${CMAKE_SOURCE_DIR}/tests/common:${CMAKE_BINARY_DIR}/utils/")
+            ENVIRONMENT
+            "LD_LIBRARY_PATH=${CMAKE_BINARY_DIR}/lib/;PYTHONPATH=${UHD_BINARY_DIR}/python:${CMAKE_SOURCE_DIR}/tests/common:${CMAKE_BINARY_DIR}/utils/"
+            )
endif()
endfunction(UHD_ADD_PYTEST)
```

Closing this for now.

closed 2021-09-03 by mbr0wn