EttusResearch/uhd #380: B200 overrun causes flowgraph to freeze

radio-dan, 2020-09-16:
## Issue Description ##
When running a GRC flowgraph using a UHD: USRP Source with a B200, if the flowgraph is slow enough that an overrun results from USRP, the entire flowgraph freezes permanently.

## Setup Details ##
Linux Mint 20 running in VirtualBox hosted on Windows 10
USRP B200 connected via USB 3
UHD_4.0.0.0-0-g90ce6062
Gnuradio maint-3.8 branch, b0619899

## Expected Behavior ##
Continuous overruns reported and flowgraph tries to keep up.

## Actual Behaviour ##
A single overrun is reported and the flowgraph freezes.

## Steps to reproduce the problem ##
A GRC flowgraph consisting of just a USRP Source -> QT GUI Frequency Sink, attempting to run at 40Msps sample rate.

## Additional Information ##
This problem does not occur when building from UHD-3.15.LTS (UHD_3.15.0.0-54-gbc585b43

radio-dan, 2020-09-16:
Example GRC
[b200 test.grc.zip](https://github.com/EttusResearch/uhd/files/5233589/b200.test.grc.zip)

mbr0wn, 2020-09-17:
Confirmed with `uhd_fft`.

michael-west, 2020-10-12:
The root cause has been found and fixed on the master branch (commit 70e1f0f0c697d6522e3a87e2de93e34e375decb4).  Thanks for reporting!

closed 2020-10-12 by michael-west