EttusResearch/uhd #381: n310 image build with zeus and 3.15 fails

msherman64, 2020-09-19:
## Issue Description ##
When trying to use the oe-build docker container to build against the zeus branch, UHD 4.0 builds succeed, while zeus + 3.15 fails.

## Setup Details ##
Using the OE-Build docker image.
Added a separate output directory at `/output/bitbake`

added `PREFERRED_UHD_VERSION = "3.15%"` to:
`/output/bitbake/conf/site.conf`

Inside the docker image:
Set up repo with:
`repo init -u git://github.com/EttusResearch/oe-manifests.git -b zeus`
`repo sync --force-sync`

Executed build with:
`./meta-ettus/contrib/build_imgs_package.sh n3xx zeus_v3.15 . /output/bitbake/ `

## Expected Behavior ##
I expected the build to create the artifacts with UHD 3.15, instead of 4.0.

## Actual Behaviour ##
The build exits with the following error:
`mv: cannot stat '/output/bitbake/tmp-glibc/work/cortexa9t2hf-neon-oe-linux-gnueabi/mpmd/3.15.0.0-r0/image/lib/systemd/network/*.network': No such file or directory`

Reverting commits 8446217, 2f22978, 3038026, 12dce34 from metta-ettus allowed the build to continue further.

After reverting those MPMD related commits, the build halts on the error
`CMake Error at /output/bitbake/tmp-glibc/work/cortexa9t2hf-neon-oe-linux-gnueabi/gr-ettus/0.0.4+gitAUTOINC+30640b7396-r0/recipe-sysroot/usr/lib/cmake/gnuradio/FindUHD.cmake:97 (message):
The found UHD version (3.15.0.0-0-gaea0e2de) is not compatible with the version required (4.0).

In meta-sdr/recipes-support/gr-ettus/gr-ettus_git.bb, it appears to be targeting the gr-ettus master branch, and recent commits have hard coded uhd4.0 as a dependency. I attempted to change this, but am not familiar with the bitbake syntaax around rebuilds and git revisions.

## Steps to reproduce the problem ##
Set `PREFERRED_UHD_VERSION = "3.15%"` and execute a build on branch zeus.

michael-west, 2020-09-21:
@msherman64 UHD 3.15 requires thud.  Please use the thud branch of oe-manifests for UHD 3.15.  The

closed 2020-09-21 by michael-west

msherman64, 2020-09-22:
Hi Michael,
    Thanks for that information. In that case, the documentation in the
meta-ettus readme is a bit unclear.
What is the intended usage of the preferred_uhd_version flag?

Thanks for your help,
Mike Sherman