EttusResearch/uhd #382: 4.0.0.0 Python-3.8 link problem in Mageia Cauldron.

barjac, 2020-09-21:
Simplest to show you the full build log:
http://mtf.duckdns.org/pub/linux/barjac/distrib/cauldron/x86_64/log/uhd-4.0.0.0-1.mga8_bcj.src.rpm/build.0.20200921171431.log

All package versions comprising the build chroot can be seen here:
http://mtf.duckdns.org/pub/linux/barjac/distrib/cauldron/x86_64/log/uhd-4.0.0.0-1.mga8_bcj.src.rpm/rpm_qa.0.20200921171431.log

Any help would be greatly appreciated.

Barry
(Package maintainer for Mageia)

barjac, 2020-10-01:
OK I finally figured what was missing.
LDFLAGS need  -lpython3 maybe this can be added somewhere in the CMake setup?
There are also still some python2 syntax issues which are fixed with 2to3, but only with 2to3 errors ignored. The errors don't affect the build but 2to3 is needed to fix bytecompile.
Michael, sorry about the direct mail, but I have been struggling with this. 
Always the case when I finally give up and ask for help, the answer appears. :)

michael-west, 2020-10-06:
We left out the linking to the python3 library for all Linux distros according to the pybind11 documentation (see the Note at the end of the "Building manually" section):

https://pybind11.readthedocs.io/en/stable/compiling.html#building-manually

This was not needed by Ubuntu or Fedora, so I'm not sure why Mageia seems to need it.

mbr0wn, 2021-09-17:
@barjac What's the current status of this? Still failing?

Also, I noticed you have a bunch of custom flags:
```
+ CFLAGS='-O2 -g -pipe -Wformat -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fstack-protector --param=ssp-buffer-size=4 -fasynchronous-unwind-tables'
+ export CFLAGS
+ CXXFLAGS='-O2 -g -pipe -Wformat -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fstack-protector --param=ssp-buffer-size=4 -fasynchronous-unwind-tables'
+ export CXXFLAGS
+ FFLAGS='-O2 -g -pipe -Wformat -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fstack-protector --param=ssp-buffer-size=4 -fasynchronous-unwind-tables '
+ export FFLAGS
+ FCFLAGS='-O2 -g -pipe -Wformat -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fstack-protector --param=ssp-buffer-size=4 -fasynchronous-unwind-tables '
+ export FCFLAGS
+ LDFLAGS=' -Wl,--as-needed -Wl,--no-undefined -Wl,-z,relro -Wl,-O1 -Wl,--build-id -Wl,--enable-new-dtags'
```

In particular, the linker flags are different from what we use on Ubuntu and Fedora. If you really need special linker flags, can you just add them to your build script? (I mean, add the `-lpython3` to the other special linker flags that you're using).

mbr0wn, 2021-09-17:
Also, since Mageia is RPM-based, maybe you can check how Fedora packages UHD?

barjac, 2021-09-20:
This seems resolved in latest versions without any patching or adding special linker flags so closing as fixed.
Thanks for your help and sorry for my long delay replying, I missed the earlier notifications.

closed 2021-09-20 by barjac