EttusResearch/uhd #387: X310 Build Issue

arash1366j, 2020-10-06:
Dear All,

I followed the build instruction for generation 3 on the website.
All requirement and dependencies are fulfilled as well as Xilinx Vivado 2019.1.
I would like to save the project in Vivado for future customization, therefore I ran "make X310_HG GUI=1".
The makefile launch Vivado successfully, after configuration is done when I save the project using save as option in the Vivado. as I press generate bitstream in the Vivado, It automatically starts from synthesis phase, and it gives me the following errors in the synthesis phase:
My UHD version is currently 3.14 LTS, I tried the FPGA source from different branches (e.g uhd3.14 LTS and uhd3.15 LTS) always results in the same errors.

[Synth 8-1766] cannot open include file ../../lib/io_port2/LvFpga_Chinch_Interface.vh ["/home/scoe-adm/Downloads/uhd-master/fpga/usrp3/top/x300/build-X310_HG/X310.srcs/sources_1/imports/usrp3/top/x300/x300.v":18]

[Synth 8-2715] syntax error near ; ["/home/scoe-adm/Downloads/uhd-master/fpga/usrp3/top/x300/build-X310_HG/X310.srcs/sources_1/imports/usrp3/top/x300/x300.v":656]

[Synth 8-1031] DMA_STREAM_WIDTH is not declared ["/home/scoe-adm/Downloads/uhd-master/fpga/usrp3/top/x300/build-X310_HG/X310.srcs/sources_1/imports/usrp3/top/x300/x300.v":664]

[Common 17-69] Command failed: Synthesis failed - please see the console or run log file for details

I'm absolutely blocked, any help is highly appreciated.

Kind Regards,

-----------------------------------------
Dipl.-Ing. Arash Jafari 

Phone: +43 650 844 3617
E-mail: arash.jafari.telecom@gmail.com

mbr0wn, 2020-10-06:
Does it work without GUI=1?

mbr0wn, 2020-10-06:
Also note the paths you are printing are for UHD 4, and not for UHD-3.15. Have you tried cloning https://github.com/EttusResearch/fpga and building that for UHD-3.15?

arash1366j, 2020-10-06:
Dear mbr0wn,
Thank you very mush for your response.
without GUI=1(in batch mode), it generates the **.bit** file without any problem!
But in GUI mode the configuration is not correct!
I've tried uhd 3.14 L, uhd 3.15 LTS, uhd master (4). All of them result in the same issue.

Kind Regards

jpendlum, 2020-10-06:
@arash1366j 

Make sure when you save your Vivado project to **not** select the option to copy the source files. I believe that is the cause of your issue.

arash1366j, 2020-10-07:
Dear Jonathon,

Thank you for your response 

The option you are talking about is available in Add sources not in save as (See  attachment), I would be grateful if you can explain in details?

Regards
[image]
[image]

mbr0wn, 2020-10-08:
Since this is resolve; closing.

closed 2020-10-08 by mbr0wn