EttusResearch/uhd #388: USRP 2974 FPGA Not discoverable in Vivado Over JTAG

kamranayub1100, 2020-10-07:
## Issue Description ##
The X3x0 of USRP 2974 is not discoverable in Vivado 2015.4/2019.1. But appears in lsusb as FTDI.

## Setup Details ##
The block diagram of USRP 2974<https://kb.ettus.com/File:2974_blk_dia_hiLevel_v01.png> suggests that the SoM (host) has a USB2.0 connected with FPGA JTAG. In order to program the FPGA directly, I made following changed to the device:
1.	Wiped-off NI Linux RT
2.	Installed Ubuntu 18.04.5
3.	Make UHD 3.15.0.0 and installed with success
4.	Setup the correct IP of internal enp1s0f0 to 192.168.40.1. I was able to successfully ping FPGA @ 192.168.40.2
5.	The device was discoverable with uhd_find_devices.
6.	I had to update the image for running uhd probe command. Image was updated successfully from 19 to 36. Uhd probe output can be found: https://pastebin.com/ZZ71fKTG
7.	Followed exactly : https://kb.ettus.com/X300/X310_Device_Recovery for discovery of the device in Vivado via JTAG. Failed to discover one.
8.	Double checked the driver installation, power cycle, etc. No luck. Here is lsusb which shows device connected but not discoverable in Vivado : https://pastebin.com/vFLB3rM6
9.	Installed Vivado 2019.1 on similar lines, reinstalled the latest drivers, power cycle the device, however, exactly the same results.

## Expected Behavior ##
The X3x0 FPGA (Kintex7) should be discoverable in Vivado automatically.

## Actual Behaviour ##
Despite all the dependencies installed, the FPGA is not discoverable.

## Steps to reproduce the problem ##
As given above in the setup details. The same steps were followed by my clients on another 2974 machine. They gave exact same results.

## Additional Information ##
NA

michael-west, 2020-10-12:
Did the Digilent drivers install successfully (see 
https://kb.ettus.com/X300/X310_Device_Recovery#Installing_the_Digilent_Cable_Driver)?

Redirecting to Support for assistance.

michael-west, 2020-10-14:
Tracking this issue internally.  User is working with R&D to resolve.

kamranayub1100, 2020-10-15:
Thanks @michael-west for the comments here.

gb-esat, 2022-11-02:
We have the same problem--was there any resolution? Thanks.

marino-gnss, 2023-07-13:
Have you checked that the FTDI module has been programmed correctly for it to be recognised as a Digilent JTAG