EttusResearch/uhd #389: it seems DMAFIFO doesn't work correctly in UHD4.0.0.0 Release (N310)

dmitryshidlovsky, 2020-10-12:
## Issue Description ##
I'd like to capture 1 250 000 samples in 125 MHz sample rate, and I'm using DMAFIFO for that. I have my own fpga part, but the issue can be reproduced with default bit file with DMAFIFO (using usrp_n310_fpga_AA.bit) and with my uhd program (rx_multi_samples.cpp with some modification to use DMAFIFO). 
 

## Setup Details ##
UHD\FPGA master (4.0.0.0), N310

## Expected Behavior ##
capturing of 1 250 000 samples at sample rate 125 MHz through DMAFIFO

## Actual Behaviour ##
[ERROR] [STREAMER] The receive transport caught a value exception.
ValueError: Bad CHDR header or invalid packet length.
Error: Receiver error: ERROR_CODE_BAD_PACKET

## Steps to reproduce the problem ##
with default 4.0.0.0 sdi  (command: uhd_images_downloader -t n3xx_common_sdimg_default)
load bit file with DMAFIFO: 
uhd_image_loader --args type=n3xx --fpga-path /usr/share/uhd/images/usrp_n310_fpga_AA.bit
run capture program, I use modified version of rfnoc_rx_to_file (see Additional Information)
./rfnoc_rx_to_file --rate=125000000 --nsamps=1250000

## Additional Information ##

//
// Copyright 2011 Ettus Research LLC
// Copyright 2018 Ettus Research, a National Instruments Company
//
// SPDX-License-Identifier: GPL-3.0-or-later
//

#include <uhd/usrp/multi_usrp.hpp>
#include <uhd/utils/safe_main.hpp>
#include <uhd/utils/thread.hpp>
#include <boost/algorithm/string.hpp>
#include <boost/format.hpp>
#include <boost/program_options.hpp>
#include <chrono>
#include <complex>
#include <iostream>
#include <thread>
#include <fstream>
#include <uhd/rfnoc/ddc_block_control.hpp>
//#include <uhd/rfnoc/radarxcorr_control.hpp>
#include <uhd/rfnoc/dmafifo_block_control.hpp>
#include <uhd/rfnoc/defaults.hpp>
#include <uhd/rfnoc/mb_controller.hpp>
#include <uhd/rfnoc/radio_control.hpp>
#include <uhd/rfnoc_graph.hpp>

namespace po = boost::program_options;

int UHD_SAFE_MAIN(int argc, char* argv[])
{
    // variables to be set by po
    std::string args, sync, subdev, channel_list;
    double seconds_in_future;
    size_t total_num_samps, shift;
    double rate, freq, gain, s;

    // setup the program options
    po::options_description desc("Allowed options");
    // clang-format off
    desc.add_options()
        ("help", "help message")
        ("args", po::value<std::string>(&args)->default_value(""), "single uhd device address args")
        ("secs", po::value<double>(&seconds_in_future)->default_value(1.5), "number of seconds in the future to receive")
        ("nsamps", po::value<size_t>(&total_num_samps)->default_value(10000), "total number of samples to receive")
        ("rate", po::value<double>(&rate)->default_value(100e6/16), "rate of incoming samples")
        ("freq", po::value<double>(&freq)->default_value(0.0), "RF center frequency in Hz")
[... 344 lines cut ...]
        num_acc_samps += num_rx_samps;
        
        std::cout << "num_rx_samps:" << num_rx_samps << std::endl;
        
        if (!s) {
        
        if (outfile.is_open()) {
        	for (size_t i = 0; i < buffs.size(); i++) {
        	
        			//std::cout << "i:" << i << std::endl;
            			outfile.write((const char*)&buffs[i].front(), num_rx_samps * 2 *sizeof(short));
            		}
        }
        
        }
    }

    if (num_acc_samps < total_num_samps)
        std::cerr << "Receive timeout before all samples received..." << std::endl;

    // finished
    std::cout << std::endl << "Done!" << std::endl << std::endl;

    return EXIT_SUCCESS;
}

michael-west, 2020-10-13:
@dmitryshidlovsky Thanks for reporting!  Please provide the following additional information:

1.  Is this being run from the local host on the N310 or a remote host?
2.  What spp sizes have you tried?  I see 2044 with 360 commented on the side.  Did the same error occur with the 360 value?

This could be an issue with the transport and not necessarily the DMA FIFO block, so I have changed the title.  The error indicates corrupted or lost data in a packet.  I suspect some type of buffer issue.  If the smaller spp size works, then it is likely the frames created with the larger spp are too large.  If both sizes fail, it is more likely to be an issue with buffering in the transport.  In that case, you could try experimenting with different values for the `recv_buff_size`, `recv_frame_size`, and `num_recv_frames` device arguments.

In UHD 4.0.0.0, the default images (other than the AA image) now contain the Replay block.  I would encourage you to move to using the Replay block as it has proven to be more reliable and versatile than the DMA FIFO.  You could try using the HG image and modifying the code to use the Replay block instead of the DMA FIFO.  All that is needed in the code is to directly replace the DMA FIFO with the Replay block and then just add calls to record() and play() on the Replay block.

dmitryshidlovsky, 2020-10-13:
@michael-west Thank you for the answer!
Regarding additional information:
1. It run from local host (on N310).
2. With spp=360 and less it works for a little total number of samples (ERROR_CODE_BAD_PACKET isn't seen), but with total number of samples more then 8000 overflow happens, and I lost some samples.

Thank you for directing me to Replay block! I think it will help me.

michael-west, 2020-10-14:
@dmitryshidlovsky The overflow is likely due to the occasional latency seen during DRAM refreshing.  The way to mitigate that is to make sure the DMA FIFO is on a separate endpoint and use a large ingress buffer on the endpoint.  You will need to do that when using the Replay block as well.

Since reducing the spp to 360 resolves the original issue of the ERROR_CODE_BAD_PACKET, I am closing this issue.  Thank you!

closed 2020-10-14 by michael-west