EttusResearch/uhd #391: CMake errot :The source directory "/home/...." does not appear to contain CMakeLists.txt.

Csysongyi, 2020-10-19:
Hello,
when I installing the uhd on my Ubuntu 18.04 system, I met a problem,  I fellow the steps on the web page:

https://kb.ettus.com/Building_and_Installing_the_USRP_Open-Source_Toolchain_(UHD_and_GNU_Radio)_on_Linux

mkdir build
 cd build
 cmake ../

when I use  "cmake ../ ", an error happened:

CMake errot :The source directory "/home/...." does not appear to contain CMakeLists.txt.

And I found that there doesn't have the CMakeLists.txt file in my uhd folder
how can I  solve this problem 
thanks in advanced.

michaelld, 2020-10-19:
Here are the instructions:
```
Next, create a build folder within the repository, invoke CMake and build UHD.

   cd host
   mkdir build
   cd build
   cmake ../
```

Did you do `cd host` part before the `mkdir build` part? That's critical to successfully building UHD from source!

michaelld, 2020-10-19:
FYI this issue is better off served by either [the USRP email](https://lists.ettus.com/mailman/listinfo/usrp-users_lists.ettus.com) list or [support@ettus.com](mailto:support@ettus.com). UHD builds fine from source, so this is not an 'issue' per se -- which is really what these GitHub issues are about. Hence I'm closing this.

closed 2020-10-19 by michaelld

Csysongyi, 2020-10-19:
Thank you very much, I worked it out under your help, and I will join the mailing list of UHD, thanks for your recommendation.

best wishes!

Songyi.Cao