EttusResearch/uhd #392: `rfnoc_image_builder -g `, Not opening Vivado GUI

ctonermw, 2020-11-06:
## Issue Description ##
rfnoc_image_builder does not pass the GUI option to the build environment and therefore the Vivado GUI never opens.

## Setup Details ##

UHD-4.0

## Expected Behavior ##
The Vivado GUI to open

## Actual Behaviour ##
The make command at the end does not include 'GUI=1' and therefore the Vivado GUI does not open.

## Steps to reproduce the problem ##
`rfnoc_image_builder -g -y any_image.yml`

## Additional Information ##
The main rfnoc_image_builder file sets the gui argument to 'gui'. However the image_builder.py line 750 checks the argument 'GUI' as the cases are different the condition is always false. Setting line 203 in rfnoc_image_builder to `GUI=args.GUI,` enables the correct functionality of  image_builder.py. However somewhere in the build environment this option is lost as the final make command still does not include 'GUI=1' as I would expect.

MungerNI, 2020-11-13:
@ctonermw I can reproduce the issue. I have applied the mentioned change to uppercase to rfnoc_image_builder and this seems to fix the issue for me (i.e. GUI opens with either -g or --GUI argument to rfnoc_image_builder). Do you have any more details on what might go wrong afterwards?
I'll prioritize getting this typo fixed.

wordimont, 2020-11-13:
Note that you can also run rfnoc_image_builder with the `--generate-only` option, then run make with the `GUI=1` option to get the desired behavior.

mbr0wn, 2021-06-28:
Fixed by cb68d724e6c146036b18b88a9f78254412bee154

closed 2021-06-28 by mbr0wn