EttusResearch/uhd #394: DBSRX detected but errors on USRP1

lrowen, 2020-11-16:
## Issue Description ##
Version: 3.15.0 as packaged by Anaconda's conda-forge as well as UHD 3.15.0 packaged by EPEL for CEntOS 8  Operating system: CentOS 8.  On multiple USRP1 units, running uhd_usrp_probe results in an error that the daughterboard manager encountered a recoverable error in init.  Run of uhd_usrp_probe with logging set to 'trace' below.

## Setup Details ##
<!CentOS 8, EPEL install of UHD 3.15 or Anaconda3 conda-forge install of UHD 3.15.

## Expected Behavior ##
DBSRX is used and reports correct frequencies, not erroring out.

## Actual Behaviour ##
The USRP1 is accessible, but no programs using UHD can actually receive signal through the DBSRX

## Steps to reproduce the problem ##
On CentOS 8, install the EPEL-packged or the conda-forge UHD 3.15.  Set up udev rules for non-root user in /etc/udev/rules.d/10-usrp-uhd.rules.  If using Anaconda3, run the 'conda activate' command to obtain the anaconda environment.  Run uhd_usrp_probe

## Additional Information ##
TRACE level logging output:
```
(base) [pari-sdr@dhcp-pool167 ~]$ UHD_LOG_CONSOLE_LEVEL=0 uhd_usrp_probe
[INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_107400; UHD_3.15.0.HEAD-release
[DEBUG] [MPMD] Discovering MPM devices on port 49600
[DEBUG] [MPMD] Discovering MPM devices on port 49600
[TRACE] [UDP] Creating udp transport for 127.255.255.255 49600
[TRACE] [UDP] Creating udp transport for 192.168.1.255 49600
[DEBUG] [MPMD] Discovering MPM devices on port 49600
[TRACE] [UDP] Creating udp transport for 192.168.122.255 49600
[DEBUG] [USRP1] USRP1 firmware image: /home/pari-sdr/anaconda3/share/uhd/images/usrp1_fw.ihx
[INFO] [FX2] Loading firmware image: /home/pari-sdr/anaconda3/share/uhd/images/usrp1_fw.ihx...
[INFO] [FX2] Firmware loaded
[TRACE] [UDP] Creating udp transport for 192.168.1.255 49152
[TRACE] [UDP] Creating udp transport for 192.168.122.255 49152
[TRACE] [UDP] Creating udp transport for 192.168.1.255 49152
[TRACE] [UDP] Creating udp transport for 192.168.122.255 49152
[TRACE] [NIRIO] rpc_client connection request cancelled/aborted.
[TRACE] [UDP] Creating udp transport for 192.168.1.255 49152
[TRACE] [UDP] Creating udp transport for 192.168.122.255 49152
[TRACE] [UDP] Creating udp transport for 192.168.1.255 50000
[TRACE] [UDP] Creating udp transport for 192.168.122.255 50000
[TRACE] [UHD] Device hash: 11462434024067858173
[DEBUG] [PREFS] Loaded system config file /etc/uhd/uhd.conf
[DEBUG] [PREFS] Loaded user config file /home/pari-sdr/.uhd/uhd.conf
[INFO] [USRP1] Opening a USRP1 device...
[DEBUG] [USRP1] USRP1 FPGA image: /home/pari-sdr/anaconda3/share/uhd/images/usrp1_fpga.rbf
[INFO] [FX2] Loading FPGA image: /home/pari-sdr/anaconda3/share/uhd/images/usrp1_fpga.rbf...
[INFO] [FX2] FPGA image loaded
[TRACE] [USRP1] poke32(13, 0x       0)
[TRACE] [USRP1] poke32(14, 0x       0)
[TRACE] [USRP1] peek32( 3)
[TRACE] [USRP1] peek32( 3)
[TRACE] [USRP1] peek32( 3)
[TRACE] [USRP1] peek32( 3)
[DEBUG] [USRP1] USRP1 Capabilities    number of duc's: 2    number of ddc's: 2    rx halfband:     1    tx halfband:     0
[INFO] [USRP1] Using FPGA clock rate of 64.000000MHz...
[TRACE] [USRP1] codec control write reg: 0x      20
[TRACE] [USRP1] transact_spi:   slave: 2  bits: 32  num_bits: 16  readback: 0
[TRACE] [USRP1] codec control write reg: 0x       0
[TRACE] [USRP1] transact_spi:   slave: 2  bits: 0  num_bits: 16  readback: 0
[TRACE] [USRP1] codec control write reg: 0x     106
[TRACE] [USRP1] transact_spi:   slave: 2  bits: 262  num_bits: 16  readback: 0
[TRACE] [USRP1] codec control write reg: 0x     280
[TRACE] [USRP1] transact_spi:   slave: 2  bits: 640  num_bits: 16  readback: 0
[TRACE] [USRP1] codec control write reg: 0x     380
[TRACE] [USRP1] transact_spi:   slave: 2  bits: 896  num_bits: 16  readback: 0
[TRACE] [USRP1] codec control write reg: 0x     400
[TRACE] [USRP1] transact_spi:   slave: 2  bits: 1024  num_bits: 16  readback: 0
[TRACE] [USRP1] codec control write reg: 0x     504
[TRACE] [USRP1] transact_spi:   slave: 2  bits: 1284  num_bits: 16  readback: 0
[TRACE] [USRP1] codec control write reg: 0x     608
[TRACE] [USRP1] transact_spi:   slave: 2  bits: 1544  num_bits: 16  readback: 0
[TRACE] [USRP1] codec control write reg: 0x     700
[... 311 lines cut ...]
[TRACE] [USRP1] codec control write reg: 0x    2a00
[TRACE] [USRP1] transact_spi:   slave: 2  bits: 10752  num_bits: 16  readback: 0
[TRACE] [USRP1] codec control write reg: 0x    2b00
[TRACE] [USRP1] transact_spi:   slave: 2  bits: 11008  num_bits: 16  readback: 0
[TRACE] [USRP1] codec control write reg: 0x     107
[TRACE] [USRP1] transact_spi:   slave: 2  bits: 263  num_bits: 16  readback: 0
[TRACE] [USRP1] codec control write reg: 0x     80f
[TRACE] [USRP1] transact_spi:   slave: 2  bits: 2063  num_bits: 16  readback: 0
[TRACE] [USRP1] codec control write reg: 0x    2400
[TRACE] [USRP1] transact_spi:   slave: 4  bits: 9216  num_bits: 16  readback: 0
[TRACE] [USRP1] codec control write reg: 0x    2500
[TRACE] [USRP1] transact_spi:   slave: 4  bits: 9472  num_bits: 16  readback: 0
[TRACE] [USRP1] codec control write reg: 0x    2600
[TRACE] [USRP1] transact_spi:   slave: 4  bits: 9728  num_bits: 16  readback: 0
[TRACE] [USRP1] codec control write reg: 0x    2a00
[TRACE] [USRP1] transact_spi:   slave: 4  bits: 10752  num_bits: 16  readback: 0
[TRACE] [USRP1] codec control write reg: 0x    2b00
[TRACE] [USRP1] transact_spi:   slave: 4  bits: 11008  num_bits: 16  readback: 0
[TRACE] [USRP1] codec control write reg: 0x     107
[TRACE] [USRP1] transact_spi:   slave: 4  bits: 263  num_bits: 16  readback: 0
[TRACE] [USRP1] codec control write reg: 0x     80f
[TRACE] [USRP1] transact_spi:   slave: 4  bits: 2063  num_bits: 16  readback: 0
(base) [pari-sdr@dhcp-pool167 ~]$ 
```
In looking at the dbsrx setup code in question (host/lib/usrp/dboard/db_dbsrx.cpp) I see a similar 'for(auto ..." construct as seen in the resolution to issue #304 ( [https://github.com/EttusResearch/uhd/issues/304](url) ); could this be the issue?

mbr0wn, 2021-07-06:
@lrowen sadly, we can't reproduce this because we can't find the appropriate hardware. However, I'm pretty sure you're right. Can you try simply removing the `auto` keyword and giving it a spin?

mbr0wn, 2021-07-06:
You can try this:
```patch
From 6b240326286db65e372eb6368c3c69bc1fbdc147 Mon Sep 17 00:00:00 2001
From: Martin Braun <martin.braun@ettus.com>
Date: Tue, 6 Jul 2021 17:42:54 +0200
Subject: [PATCH] dbsrx: Fix issue with loop variable

In 26cc208, we accidentally added an `auto` into a loop, making the loop
variable's scope local. However, this variable lives outside this for
loop.
---
 host/lib/usrp/dboard/db_dbsrx.cpp | 4 ++--
 1 file changed, 2 insertions(+), 2 deletions(-)

diff --git a/host/lib/usrp/dboard/db_dbsrx.cpp b/host/lib/usrp/dboard/db_dbsrx.cpp
index 09690a3d16..c2f3e6b429 100644
--- a/host/lib/usrp/dboard/db_dbsrx.cpp
+++ b/host/lib/usrp/dboard/db_dbsrx.cpp
@@ -272,7 +272,7 @@ double dbsrx::set_lo_freq(double target_freq)
     std::vector<double> clock_rates =
         this->get_iface()->get_clock_rates(dboard_iface::UNIT_RX);
     const double max_clock_rate = uhd::sorted(clock_rates).back();
-    for (auto ref_clock : uhd::reversed(uhd::sorted(clock_rates))) {
+    for (ref_clock : uhd::reversed(uhd::sorted(clock_rates))) {
         // USRP1 feeds the DBSRX clock from a FPGA GPIO line.
         // make sure that this clock does not exceed rate limit.
         if (this->get_iface()->get_special_props().soft_clock_divider) {
@@ -298,7 +298,7 @@ double dbsrx::set_lo_freq(double target_freq)
             continue;
 
         // choose R
-        for (auto r = 0; r <= 6; r += 1) {
+        for (r = 0; r <= 6; r += 1) {
             // compute divider from setting
             R = 1 << (r + 1);
             UHD_LOGGER_TRACE("DBSRX") << boost::format("DBSRX R:%d\n") % R;
```

lrowen, 2021-07-14:
Martin, thanks for taking a look.  It will be a bit before I can check it out; moved on to other projects here since reporting the issue.  But I will check it again.

closed 2021-07-20 by atrnati

mbr0wn, 2021-07-21:
Fix is on internal master and will go public soon.