EttusResearch/uhd #396: Zeus includes an incompatible version of gcc for fftw

bmagistro, 2020-12-06:
Opening an issue here for tracking/linking purposes.  The initial discussion occurred on the mailing list (http://lists.ettus.com/pipermail/usrp-users_lists.ettus.com/2020-December/063312.html).

Short version, when fftw is compiled against gcc 9.x and NEON is enabled, fftw may segfault.  Another user opened an issue with the fftw project (https://github.com/FFTW/fftw3/issues/213).  A possible solution would be to add gcc-8.x as a devtools recipe that could be used to override the default compiler shipped in Zeus (and likely Dunfell).  There is an open request with the yocto project to have them revert a recent commit and continue including gcc 8.x due to this issue.

Believe the issue better resides in [meta-ettus](https://github.com/EttusResearch/meta-ettus) but don't see a way to add issues there.

cc: @michaelld @balister

michaelld, 2020-12-07:
@bmagistro thanks for creating this public UHD issue. I've tagged it in our private / internal issues as you can see. I've also marked is as "escalated" since I think it's critical to get it fixed. We'll see what others think.

andrepuschmann, 2021-07-02:
Hey,
we are just in the progress of updating an E310 that was still running UHD 3.9.2 to a more recent version I found the same with UHD 4.0. Is there a solution yet? Is 4.1 any better? If not, what's the last known working version?

Thanks
Andre

mbr0wn, 2024-06-21:
In the next release, we'll be moving to kirkstone which contains a newer gcc.

closed 2024-06-21 by mbr0wn