EttusResearch/uhd #397: UHD 4.0.0 X310 uhd_udrp_probe runtime error: Property types do not match

icpz, 2020-12-21:
## Issue Description ##

After update the fpga image with uhd_image_loader, `uhd_usrp_probe` produce runtime error:

```bash
$ uhd_usrp_probe
[INFO] [UHD] Mac OS; Clang version 12.0.0 (clang-1200.0.32.27); Boost_107400; UHD_4.0.0.0-0-unknown
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 8000 bytes.
[INFO] [GPS] Found an internal GPSDO: LC_XO, Firmware Rev 0.929a
[INFO] [X300] Radio 1x clock: 200 MHz
Error: RuntimeError: Cannot access! Property types do not match at: /mboards/0/eeprom
```

## Setup Details ##

USRP device: USRP-2953R
FPGA image: x310_HG
uhd version: download from 'https://github.com/EttusResearch/uhd/archive/v4.0.0.0.tar.gz'
OS: macOS 11.1

## Expected Behavior ##

`uhd_usrp_probe` produces correct result without error.

## Actual Behaviour ##

```
Error: RuntimeError: Cannot access! Property types do not match at: /mboards/0/eeprom
```

## Steps to reproduce the problem ##

```bash
uhd_image_loader --args "type=x300,addr=172.16.39.5,fpga=HG"

# power cycle USRP

uhd_usrp_probe
```

## Additional Information ##

I've tried recovery EEPROM by exec `usrp_burn_mb_eeprom --args="type=x300,recover_mb_eeprom,disable_adc_self_test" --values="revision=8"`

but `usrp_burn_mb_eeprom` still reports the not match error:

```bash
$ /usr/local/opt/uhd/lib/uhd/utils/usrp_burn_mb_eeprom --args="type=x300,recover_mb_eeprom,disable_adc_self_test" --values="revision=8"
Creating USRP device from address: type=x300,recover_mb_eeprom,disable_adc_self_test
[INFO] [UHD] Mac OS; Clang version 12.0.0 (clang-1200.0.32.27); Boost_107400; UHD_4.0.0.0-0-unknown
[INFO] [X300] X300 initialization sequence...
[WARNING] [X300] UHD is operating in EEPROM Recovery Mode which disables hardware version checks.
Operating in this mode may cause hardware damage and unstable radio performance!
Error: RuntimeError: Cannot access! Property types do not match at: /mboards/0/eeprom
```

michaelld, 2020-12-21:
Does this USRP work on this macOS using UHD 3.15?

michaelld, 2020-12-21:
I also experience the `Cannot access! Property types do not match at: /mboards/0/eeprom` but not the `UHD is operating in EEPROM Recovery Mode`.

icpz, 2020-12-21:
> Does this USRP work on this macOS using UHD 3.15?

Yes, it works if I downgrade uhd to 3.15 and re-flash the fpga image.

> I also experience the `Cannot access! Property types do not match at: /mboards/0/eeprom` but not the `UHD is operating in EEPROM Recovery Mode`.

`UHD is operating in EEPROM Recovery Mode` this only shows when I try to recovery the EEPROM with the instructions from [here](https://files.ettus.com/manual/page_usrp_x3x0.html#x3x0_corrupt_eeprom).

michaelld, 2020-12-21:
OK; thanks for that added information. In my testing on my macOS computers running 11.0 or newer, the X310 shows these issues when doing a `uhd_usrp_probe` or even `uhd_find_devices` ... but, it does work for data transport e.g. `benchmark_rate`. Admittedly this was with 11.0, not 11.1 or 11.2 beta, so things might have changed with those and/or newer Xcode. FYI the issue is not with your USRP; the issue is some strangeness in how Clang does runtime type inference. I hope to look into this issue in the coming weeks / early 2021. In the meantime, if you can stick to either just doing data transport (assuming that works with your X310 + UHD 4.0) or just use UHD 3.15, then those would be the best ways to go at the moment. Given the US holidays, I doubt we'll see much movement here for at least the next 2 weeks ... so, thank you for your patience!

skotopes, 2021-02-03:
@michaelld I have same issue with b200/b205

michaelld, 2021-03-26:
Finally getting back to some of these older issues ...

The `Cannot access! Property types do not match at: /mboards/0/eeprom` error has been fixed in our internal UHD repo; it will be pushed to our public UHD repo "soon".

Is there an update from anyone else on this issue for the other issue (`UHD is operating in EEPROM Recovery Mode`)?

icpz, 2021-03-26:
> Finally getting back to some of these older issues ...
> 
> The `Cannot access! Property types do not match at: /mboards/0/eeprom` error has been fixed in our internal UHD repo; it will be pushed to our public UHD repo "soon".

Good news! Thanks for the fix!

> Is there an update from anyone else on this issue for the other issue (`UHD is operating in EEPROM Recovery Mode`)?

I don't think it's an issue since I did operate EEPROM in recovery mode. I guess that output line is by design :)

mbr0wn, 2021-05-04:
Fixed by f22e3c2d475f4579d9fdd0cd533c509a52a47ad1.

closed 2021-05-04 by mbr0wn

hassanmaree, 2023-09-08:
[image]
can any one help me resolve this issue.