EttusResearch/uhd #398: Receiving too many "L" from uhd driver

astro7x, 2020-12-29:
## Issue Description ##
Receiving too many "L" on the host which indicates late packets on the transmit channel when running OAI eNodeB connected via SPF+ 10GigE to USRPn310

## Setup Details ##

The server has an installed NIC `Intel 82599ES 10-Gigabit SFI/SFP+` connected directly to the USRPn310 via a 10G SFP+ port 1.

1. Server: Dell Precision 7920 Rack Workstation

The **server/host** is running OAI5G eNodeB on top of ubuntu18.04.5 and latest low-latency kernel `5.4.0-58-lowlatency`  with uhd version `UHD_3.15.0.HEAD-0-gaea0e2de` built from source.
_Network configuration on the host ` ip a `_

Installed NIC `Intel 82599ES 10-Gigabit SFI/SFP+` and upgraded to version number `ixgbe-5.7.1`
```
enb@node:~$ lspci | egrep -i --color 'network|ethernet'
b3:00.0 Ethernet controller: Intel Corporation 82599ES 10-Gigabit SFI/SFP+ Network Connection (rev 01)
b3:00.1 Ethernet controller: Intel Corporation 82599ES 10-Gigabit SFI/SFP+ Network Connection (rev 01)
```

```
enb@node:~$  ethtool -i enp179s0f0
driver: ixgbe
version: 5.1.0-k
firmware-version: 0x000161c1
expansion-rom-version: 
bus-info: 0000:b3:00.0
supports-statistics: yes
supports-test: yes
supports-eeprom-access: yes
supports-register-dump: yes
supports-priv-flags: yes
```

``` bash
enb@node:~$ ip a
5: enp179s0f0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 9000 qdisc mq state UP group default qlen 1000
    link/ether f8:f2:1e:94:4a:78 brd ff:ff:ff:ff:ff:ff
    inet 192.168.20.1/24 brd 192.168.20.255 scope global noprefixroute enp179s0f0
       valid_lft forever preferred_lft forever
    inet6 fe80::dafe:1d74:ebc:337d/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
```

```
enb@node:~$ uhd_usrp_probe 
[INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_3.15.0.HEAD-0-gaea0e2de
[INFO] [MPMD] Initializing 1 device(s) in parallel with args: mgmt_addr=192.168.20.2,type=n3xx,product=n310,serial=31BEB90,claimed=False,addr=192.168.20.2
[INFO] [MPM.PeriphManager] init() called with device args `mgmt_addr=192.168.20.2,product=n310,clock_source=internal,time_source=internal'.
[INFO] [0/Replay_0] Initializing block control (NOC ID: 0x4E91A00000000004)
[INFO] [0/Radio_0] Initializing block control (NOC ID: 0x12AD100000011312)
[INFO] [0/Radio_1] Initializing block control (NOC ID: 0x12AD100000011312)
[INFO] [0/DDC_0] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DDC_1] Initializing block control (NOC ID: 0xDDC0000000000000)
[INFO] [0/DUC_0] Initializing block control (NOC ID: 0xD0C0000000000002)
[INFO] [0/DUC_1] Initializing block control (NOC ID: 0xD0C0000000000002)
[INFO] [0/FIFO_0] Initializing block control (NOC ID: 0xF1F0000000000000)
[INFO] [0/FIFO_1] Initializing block control (NOC ID: 0xF1F0000000000000)
[INFO] [0/FIFO_2] Initializing block control (NOC ID: 0xF1F0000000000000)
[INFO] [0/FIFO_3] Initializing block control (NOC ID: 0xF1F0000000000000)
  _____________________________________________________
 /
|       Device: N300-Series Device
|     _____________________________________________________
|    /
|   |       Mboard: ni-n3xx-31BEB90
|   |   eeprom_version: 3
|   |   mpm_version: 3.15.0.0-gaea0e2de
|   |   pid: 16962
|   |   product: n310
|   |   rev: 10
[... 362 lines cut ...]
[TMR]   enter blocking wait for TASK_RRC_ENB
LL[PHY]   td:0 dei:0 dem:0 llr:0 tci:0 [PHY]   tca:0 tcb:0 tcg:0 tce:0 l1:0 l2:0]

L
Run time: 0h 11s
	Time executing user inst.: 1s 447402us
	Time executing system inst.: 2s 14885us
	Max. Phy. memory usage: 468688kB
	Page fault number (no io): 108052
	Page fault number (requiring io): 0
	Number of file system read: 0
	Number of filesystem write: 0
	Number of context switch (process origin, io...): 283551
	Number of context switch (os origin, priority...): 124
LLinux signal Interrupt...
/opt/oai5g/openairinterface5g/executables/softmodem-common.c:187 signal_handler() Exiting OAI softmodem: softmodem starting exit procedure

LLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLL[PHY]   south_out_cnt 2231
LLLL[PHY]   south_in_cnt 2232
[PHY]   RU 0/0 TS 17134080 (off 7680000), frame 223, subframe 1
L[PHY]   south_in/rx_rf: RU 0/0 TS 17134080 (off 7680000), frame 223, subframe 1
[PHY]   Exiting ru_thread
L[PHY]   RU 0 rf device stopped
LLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLLL
```

astro7x, 2020-12-29:
This is the reported usrp driver configuration as well. Something seems invalid here
``` bash
root@ni-n3xx-31BEB90:~# ethtool -i sfp1
driver: nixge
version:
firmware-version:
expansion-rom-version:
bus-info: platform
supports-statistics: no
supports-test: no
supports-eeprom-access: no
supports-register-dump: no
supports-priv-flags: no
```

let me know your recommendation in that matter.

astro7x, 2021-03-17:
To conclude the problem:

1.  USRPn310 has a problem on receiving the samples from the server claiming that the server is not fast enough.
2.  Accordingly, we used to encounter Late packets and around 30% losses in (BLER)
3.  Verification done using OAI5G Stack and IPERF3 to measure the throughput and it 100% clear that the problem is on the receiving side on USRP (vice versa test with IPERF3 is perfect).
4.  Double verification achieved by using B210 instead of N310 to verify our software OAI5G (no problem at all, and traffic was over expectation)

NOTE: **We tested over 4 n310 devices with the same problem. (I wish you can check this critical problem as it
s coming from USRP side)**

closed 2021-05-11 by astroa-git

bams-10, 2023-02-28:
Hi.

I'm having much of the same problem with the same setup as yours, can you please tell me how you solved this issue?

Thanks. Looking forward to hearing from you.