EttusResearch/uhd #401: uhd_image_builder building incorrect .v file

mgarrett1955, 2021-01-13:
## Issue Description ##
In the 3.15.0.0LTS branch (3/15/0/0) running uhd_image_builder or uhd_image_builder_gui creates the file
rfnoc_ce_auto_inst_e310.v but the makefile chain uses the file rfnoc_ce_auto_inst_e31x.v to build the .bit file.
## Setup Details ##
Executing the following commend line:
./uhd_image_builder.py duc ddc -t E310_RFNOC_sg3 -d E310 -m 4 --fill-with-fifos

## Expected Behavior ##
./uhd_image_builder.py duc ddc -t E310_RFNOC_sg3 -d E310 -m 4 --fill-with-fifos
will update the rfnoc_ce_auto_inst_e310.v  file in the directory:
~/rfnoc/src/uhd/fpga-src/usrp3,top,e31x
The build should occur using the uhd_image_builder created .v file.

## Actual Behavior ##
./uhd_image_builder.py duc ddc -t E310_RFNOC_sg3 -d E310 -m 4 --fill-with-fifos
will update the rfnoc_ce_auto_inst_e310.v  file in the directory:
~/rfnoc/src/uhd/fpga-src/usrp3,top,e31x
The build will occur using:
rfnoc_ce_auto_inst_e31x.v in the same directory.

## Steps to reproduce the problem ##
./uhd_image_builder.py duc ddc -t E310_RFNOC_sg3 -d E310 -m 4 --fill-with-fifos
will update the rfnoc_ce_auto_inst_e310.v  as indicated above. 

## Additional Information ##
Here is th generated command line information:

mgarrett@ubuntu:~/rfnoc/src/uhd/fpga-src/usrp3/tools/scripts$ ./uhd_image_builder.py ddc duc -t E310_RFNOC_sg3 -d E310 -m 4 --fill-with-fifos
--Using the following blocks to generate image:
    * ddc
    * duc
Adding CE instantiation file for 'E310_RFNOC_sg3'
changing temporarily working directory to /home/mgarrett/rfnoc/src/uhd/fpga-src/usrp3/tools/scripts/../../top/e31x
/bin/bash -c ". ./setupenv.sh && make E310_RFNOC_sg3"
Setting up a 64-bit FPGA build environment for the USRP-E31x...
- Vivado: Found (/opt/Xilinx/Vivado/2018.3/bin)

Environment successfully initialized.
make -f Makefile.e31x.inc bin NAME=E310_RFNOC_sg3 ARCH=zynq PART_ID=xc7z020/clg484/-3 RFNOC=1 E310_SG3=1 TOP_MODULE=e31x EXTRA_DEFS="RFNOC=1 E310_SG3=1"
make[1]: Entering directory '/home/mgarrett/rfnoc/src/uhd/fpga-src/usrp3/top/e31x'
BUILDER: Checking tools...
* GNU bash, version 4.4.20(1)-release (x86_64-pc-linux-gnu)
* Python 2.7.17
* Vivado v2018.3 (64-bit)

jpendlum, 2021-01-13:
@mgarrett1955 This issue is already fixed on the fpga repo with commit https://github.com/EttusResearch/fpga/commit/45db3d04fa08b0bf83b6379d9dc4a8144e8b1f72, but the fpga-src submodule needs to be updated to point to the new commit. For now, you can go into the fpga-src directory and checkout the UHD-3.15.LTS tag to get the fix.

jpendlum, 2021-01-13:
@michael-west ^

mgarrett1955, 2021-01-13:
jpendlum, Thank you very much that solved my issue!! Is there an updated document for the 3.15LTS install cross environment?

Thank you again for the quick response!

mbr0wn, 2021-06-17:
We have a fix in the pipeline for updating the submodule pointer.

mbr0wn, 2021-06-30:
Fixed with 1b60cf86d39354b5e46f4e250b9e59758ac50ec9.

closed 2021-06-30 by mbr0wn