EttusResearch/uhd #405: uhd-4.0.0.0, boost-1.75.0, gcc-11 build failure

yarda, 2021-01-26:
## Issue Description ##
uhd-4.0.0.0 build failure

## Setup Details ##
Fedora rawhide (f34)

## Expected Behavior ##
Build

## Actual Behaviour ##
Build error

## Steps to reproduce the problem ##
Build for fedora rawhide (f34)

## Additional Information ##
Build log fragment:
```
[47%] Building CXX object lib/CMakeFiles/uhd.dir/usrp/dboard/magnesium/magnesium_gain_table.cpp.o
cd /builddir/build/BUILD/uhd-4.0.0.0/host/x86_64-redhat-linux-gnu/lib && /usr/bin/g++ -DBOOST_ASIO_DISABLE_STD_EXPERIMENTAL_STRING_VIEW -DBOOST_ASIO_DISABLE_STD_STRING_VIEW -DBOOST_ERROR_CODE_HEADER_ONLY -DHAVE_CONFIG_H -DUHD_DLL_EXPORTS -DUHD_LOG_CONSOLE_COLOR -DUHD_LOG_CONSOLE_LEVEL=2 -DUHD_LOG_FILE_LEVEL=2 -DUHD_LOG_MIN_LEVEL=1 -I/builddir/build/BUILD/uhd-4.0.0.0/host/x86_64-redhat-linux-gnu/include -I/builddir/build/BUILD/uhd-4.0.0.0/host/include -I/builddir/build/BUILD/uhd-4.0.0.0/host/lib/include -I/builddir/build/BUILD/uhd-4.0.0.0/host/lib/deps/flatbuffers/include -I/builddir/build/BUILD/uhd-4.0.0.0/host/x86_64-redhat-linux-gnu/lib/ic_reg_maps -I/builddir/build/BUILD/uhd-4.0.0.0/host/lib/convert -I/builddir/build/BUILD/uhd-4.0.0.0/host/x86_64-redhat-linux-gnu/lib/convert -I/builddir/build/BUILD/uhd-4.0.0.0/host/lib/usrp -I/builddir/build/BUILD/uhd-4.0.0.0/host/lib/usrp/common/ad9361_driver -I/builddir/build/BUILD/uhd-4.0.0.0/host/lib/usrp/common -I/builddir/build/BUILD/uhd-4.0.0.0/host/x86_64-redhat-linux-gnu/lib/transport/nirio/lvbitx -I/usr/include/libusb-1.0 -I/builddir/build/BUILD/uhd-4.0.0.0/host/lib/deps/rpclib/include -I/builddir/build/BUILD/uhd-4.0.0.0/host/x86_64-redhat-linux-gnu/_cmrc/include -O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1  -m64  -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fvisibility=hidden -fvisibility-inlines-hidden -DNDEBUG -fPIC   -Wall -Wextra -Wsign-compare -std=gnu++14 -o CMakeFiles/uhd.dir/usrp/dboard/magnesium/magnesium_gain_table.cpp.o -c /builddir/build/BUILD/uhd-4.0.0.0/host/lib/usrp/dboard/magnesium/magnesium_gain_table.cpp
In file included from /builddir/build/BUILD/uhd-4.0.0.0/host/lib/usrp/dboard/e3xx/e3xx_ad9361_iface.hpp:14,
                 from /builddir/build/BUILD/uhd-4.0.0.0/host/lib/usrp/dboard/e3xx/e3xx_ad9361_iface.cpp:7:
/builddir/build/BUILD/uhd-4.0.0.0/host/lib/include/uhdlib/utils/rpc.hpp: In function 'bool {anonymous}::await_rpc_connected_state(std::shared_ptr<rpc::client>, std::chrono::milliseconds)':
/builddir/build/BUILD/uhd-4.0.0.0/host/lib/include/uhdlib/utils/rpc.hpp:29:27: error: 'sleep_for' is not a member of 'std::this_thread'
   29 |         std::this_thread::sleep_for(std::chrono::milliseconds(1));
      |                           ^~~~~~~~~
/builddir/build/BUILD/uhd-4.0.0.0/host/lib/include/uhdlib/utils/rpc.hpp:29:27: note: suggested alternatives:
In file included from /usr/include/boost/thread/thread_only.hpp:17,
                 from /usr/include/boost/thread/thread.hpp:12,
                 from /builddir/build/BUILD/uhd-4.0.0.0/host/include/uhd/utils/log.hpp:12,
                 from /builddir/build/BUILD/uhd-4.0.0.0/host/include/uhd/types/filters.hpp:11,
                 from /builddir/build/BUILD/uhd-4.0.0.0/host/lib/usrp/dboard/e3xx/e3xx_ad9361_iface.hpp:11,
                 from /builddir/build/BUILD/uhd-4.0.0.0/host/lib/usrp/dboard/e3xx/e3xx_ad9361_iface.cpp:7:
/usr/include/boost/thread/pthread/thread_data.hpp:295:14: note:   'boost::this_thread::sleep_for'
  295 |         void sleep_for(const chrono::duration<Rep, Period>& d)
      |              ^~~~~~~~~
/usr/include/boost/thread/pthread/thread_data.hpp:341:16: note:   'boost::this_thread::no_interruption_point::sleep_for'
  341 |           void sleep_for(const chrono::duration<Rep, Period>& d)
      |                ^~~~~~~~~
```

I am actually not sure whether it's caused by boost-1.75.0 or gcc-11.

michaelld, 2021-01-26:
`std::this_thread::sleep_for` should be available in C++11 or newer < https://en.cppreference.com/w/cpp/thread/sleep_for >. I'd guess this is a gcc11 issue, not a Boost one. Any way you can try gcc10 for both Boost and UHD?

yarda, 2021-01-26:
This error is not reproducible on f33 (boost-1.73.0, gcc-10.2.1). Unfortunately, I hadn't time to investigate it more deep, but it's on my todo list :)

mbr0wn, 2021-01-29:
If you want to test, try
```diff
commit ae3d7551ded10ac51a3c2eda1e9e8ac64c4cd83b (HEAD -> fix-missing-includes-rpc, origin/fix-missing-includes-rpc)
Author: Martin Braun <martin.braun@ettus.com>
Date:   Fri Jan 29 12:23:50 2021 +0100

lib: Fix missing includes in rpc.hpp

diff --git a/host/lib/include/uhdlib/utils/rpc.hpp b/host/lib/include/uhdlib/utils/rpc.hpp
index e87a2ee32..ca89c1054 100644
--- a/host/lib/include/uhdlib/utils/rpc.hpp
+++ b/host/lib/include/uhdlib/utils/rpc.hpp
@@ -12,7 +12,11 @@
#include <rpc/client.h>
#include <rpc/rpc_error.h>
#include <boost/format.hpp>
+#include <chrono>
#include <memory>
+#include <mutex>
+#include <string>
+#include <thread>

namespace {
```

yarda, 2021-01-29:
@mbr0wn thanks, it fixed the problem for me. Previously, I was trying to add just `#include <chrono>`, but it wasn't enough.

michaelld, 2021-01-29:
Thanks for reporting back, @yarda . Good luck with your UHD / USRP work!

closed 2021-01-29 by michaelld

mbr0wn, 2021-01-29:
Yeah, thanks. We'll be pushing this fix live soon.

ashley-b, 2021-04-29:
Did this build error get patched?. I have the same issue and could not see a commit for this.
I found the following patch fixed the build for me.
```
diff --git a/host/lib/include/uhdlib/utils/rpc.hpp b/host/lib/include/uhdlib/utils/rpc.hpp
index e87a2ee32..3ce8352ee 100644
--- a/host/lib/include/uhdlib/utils/rpc.hpp
+++ b/host/lib/include/uhdlib/utils/rpc.hpp
@@ -13,6 +13,7 @@
 #include <rpc/rpc_error.h>
 #include <boost/format.hpp>
 #include <memory>
+#include <thread> /* std::this_thread::sleep_for */
 
 namespace {
 
diff --git a/host/lib/usrp/dboard/e3xx/e3xx_radio_control_impl.cpp b/host/lib/usrp/dboard/e3xx/e3xx_radio_control_impl.cpp
index 360ce3a11..cd2baa667 100644
--- a/host/lib/usrp/dboard/e3xx/e3xx_radio_control_impl.cpp
+++ b/host/lib/usrp/dboard/e3xx/e3xx_radio_control_impl.cpp
@@ -17,6 +17,7 @@
 #include <cmath>
 #include <cstdlib>
 #include <sstream>
+#include <thread> /* std::this_thread::sleep_for */
 
 using namespace uhd;
 using namespace uhd::usrp;

```

mbr0wn, 2021-05-04:
@ashley-b the rpc.hpp is fixed, but we hadn't seen the error in the exx_radio_control yet. There's some more includes missing from there, we'll push a fix for that soon.