EttusResearch/uhd #409: b205 mini cannot keep up with slow sample rates (5/10MHz) without an OVERFLOW

crroush, 2021-02-01:
## Issue Description ##
Using both UHD 3.15 and 4.0.0 I am unable to keep up with a 5/10MHz sample rate reliably.  
For simplicity I am using the rx_samples_to_file since at these lower rates it should have no issue keeping up.  Although the same setup I have for a LimeSDR using SOAPY can easily go up to 50MHz sample rate without dropping a packet and writing to disk.   So I do not believe I have an issue with my hardware.  

## Setup Details ##
`./rx_samples_to_file --rate=10e6 --duration=600 --file=/data/samp.dat  --spb=261120`
Connected to usb-3.0:

```
System:    Host: Kernel: 5.8.0-7630-generic x86_64 bits: 64 Desktop: Gnome 3.36.4 Distro: Pop!_OS 20.04 LTS 

CPU:       Topology: 16-Core (2-Die) model: AMD Ryzen 9 3950X bits: 64 type: MT MCP MCM L2 cache: 8192 KiB 
           Speed: 2195 MHz min/max: 2200/3500 MHz Core 
Drives:    Local Storage: total: 953.87 GiB used: 515.81 GiB (54.1%) 
           ID-1: /dev/nvme0n1 vendor: Micron model: 2300 NVMe 1024GB size: 953.87 GiB 
```
I have attempted to keep the samples per buffer a function of 2040 so that we see no fragmentation during the allocation process. I have tried different sizes and received the same result.  Sometimes it might let me run a bit longer but it always ends with an OVERFLOW error.

I created a flamegraph ( https://github.com/brendangregg/FlameGraph ) to see where the kernel was spending the most time, which pointed to a lot of time is being spent waiting on a conditional variable.  [https://github.com/EttusResearch/uhd/blob/master/host/lib/transport/libusb1_zero_copy.cpp#L152-L191](here)
I did remove the conditional wait and placed a do / while loop that was timed based on the timeout value passed to the function.  It gave me another minute or so before it overflowed.

Some of the other issues people have posted  pointed to playing with the num_frames / frame_size,  I upped the num_frames to 256 (per a suggestion in another post).  It allowed it to run longer but still ended in overflow.  
I also increased the frame_size by various factors (2,4,8) that had zero impact on allowing it to keep up.  
## Expected Behavior ##
B205Mini to run without dropping packets.  Since I can do this with other SDRs I would expect this one to behave the same way,.

## Actual Behaviour ##
it overflows then times out and you have to restart the collection process

## Steps to reproduce the problem ## 
This does it for me:
`./rx_samples_to_file --rate=10e6 --duration=600 --file=/data/samp.dat  --spb=261120`

mbr0wn, 2021-09-21:
A couple of comments:
- rx_samples_to_file is not a great tool, performance-wise. It's more of an example. The most obvious issue is the lack of reader/writer threads. What do you use to capture from the LimeSDR?
- At 10 MHz, I occasionally do see overruns, albeit not many. Are you seeing lots of overruns, or occasional ones?
- benchmark_rate runs without overruns for me up to 56 MHz. Can you confirm that?

    ./examples/benchmark_rate --rx_rate 56e6 --args type=b200 --duration 30 

- Since you're waiting on that condition variable, it means it's taking a long time for transfers to complete (i.e., it's stuck somewhere in libusb). Not sure what that tells us. Would you mind lettings us know which USB controller you're using?

mbr0wn, 2021-09-24:
@crroush Unless you're getting lots and of overruns, I'm going to close this. As mentioned before, this particular example is not great optimized for performance (rather, for readability) and the occasional overrun is not unusual.

Thanks for the report.

closed 2021-09-24 by mbr0wn

crroush, 2021-09-24:
@mBr0wn we were seeing it with gnuradio and soapy.  Ultimately we worked with @michaelld offline to troubleshoot the issue.  We ended up making modifications to the UHD driver side to decrease the amount of context switching on the UHD libusb interface, which helped us to get the performance we needed.