EttusResearch/uhd #410: uhd-4.0.0.0, gcc-11, boost-1.75, build error on s390x

yarda, 2021-02-01:
## Issue Description ##
Build error on s390x, other arches builds OK.

## Setup Details ##
Fedora rawhide (f34), gcc-11, boost-1.75

## Expected Behavior ##
Build OK.

## Actual Behaviour ##
```
[ 81%] Linking CXX executable test_dboard_coercion
cd /builddir/build/BUILD/uhd-4.0.0.0/host/s390x-redhat-linux-gnu/examples && /usr/bin/cmake -E cmake_link_script CMakeFiles/test_dboard_coercion.dir/link.txt --verbose=1
/usr/bin/g++ -O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1  -m64 -march=zEC12 -mtune=z13 -fasynchronous-unwind-tables -fstack-clash-protection -fvisibility=hidden -fvisibility-inlines-hidden -DNDEBUG -Wl,-z,relro -Wl,--as-needed  -Wl,-z,now -specs=/usr/lib/rpm/redhat/redhat-hardened-ld CMakeFiles/test_dboard_coercion.dir/test_dboard_coercion.cpp.o -o test_dboard_coercion  -Wl,-rpath,/builddir/build/BUILD/uhd-4.0.0.0/host/s390x-redhat-linux-gnu/lib: ../lib/libuhd.so.4.0.0 /usr/lib64/libboost_chrono.so /usr/lib64/libboost_date_time.so /usr/lib64/libboost_filesystem.so /usr/lib64/libboost_program_options.so /usr/lib64/libboost_serialization.so -lboost_thread -lpthread /usr/lib64/libboost_unit_test_framework.so /usr/lib64/libboost_system.so /usr/lib64/libboost_atomic.so -lpthread -ldl /lib64/libusb-1.0.so ../lib/rc/libuhd-resources.a 
/usr/bin/ld: ../lib/libuhd.so.4.0.0: undefined reference to `__atomic_compare_exchange_16'
/usr/bin/ld: ../lib/libuhd.so.4.0.0: undefined reference to `__atomic_store_16'
/usr/bin/ld: ../lib/libuhd.so.4.0.0: undefined reference to `__atomic_load_16'
collect2: error: ld returned 1 exit status
gmake[2]: *** [examples/CMakeFiles/tx_timed_samples.dir/build.make:118: examples/tx_timed_samples] Error 1
gmake[2]: Leaving directory '/builddir/build/BUILD/uhd-4.0.0.0/host/s390x-redhat-linux-gnu'
gmake[1]: *** [CMakeFiles/Makefile2:818: examples/CMakeFiles/tx_timed_samples.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
```

## Steps to reproduce the problem ##
Build on s390x on Fedora rawhide (f34)

## Additional Information ##

yarda, 2021-02-01:
It probably needs linking with `-latomic` on s390x.

yarda, 2021-02-02:
I used the following workaround in Fedora:
```
diff --git a/host/lib/utils/CMakeLists.txt b/host/lib/utils/CMakeLists.txt
index 7e95d8c..2ad6e40 100644
--- a/host/lib/utils/CMakeLists.txt
+++ b/host/lib/utils/CMakeLists.txt
@@ -229,3 +229,5 @@ if(ENABLE_C_API)
         ${CMAKE_CURRENT_SOURCE_DIR}/thread_priority_c.cpp
     )
 endif(ENABLE_C_API)
+
+LIBUHD_APPEND_LIBS(atomic)
```
It's not nice, but it works. LDFLAGS didn't help, because libatomic needs to be specifed after libuhd. Maybe boost pkgconfig should provide it on s390x?

jholtom, 2021-02-12:
This issue also occurs with gcc version 8.3.0 (Raspbian 8.3.0-6+rpi1), 1.67.0.1+b1, on aarch64.  Adding the LIBUHD_APPEND_LIBS(atomic) as above works to fix that issue.  Also needed to add -fpu=neon to the C/CXXFLAGS to get a working build.

drmpeg, 2021-07-29:
I'm seeing this issue when upgrading from CMake 3.18.4 (on Ubuntu 21.04) to CMake 3.21.1 (latest release).

The ```LIBUHD_APPEND_LIBS(atomic)``` fix works fine.

mbr0wn, 2021-08-25:
I'm experimenting with this patch:
```patch
From feada5b7e29d70e9b11629ebd093ce288f87b76f Mon Sep 17 00:00:00 2001
From: Martin Braun <martin.braun@ettus.com>
Date: Wed, 25 Aug 2021 11:14:30 +0200
Subject: [PATCH] cmake: Add check for libatomic linking requirement

Some archs require linking against libatomic, others don't. We add some
CMake code that checks for libatomic.so requirement if:
- We are not on MSVC, AND
- Compiling std::atomics code would cause a linker error.

We then check for the existence of libatomic.so, and fail if we can't
find it.
---
 host/cmake/Modules/UHDAtomics.cmake | 80 +++++++++++++++++++++++++++++
 host/lib/utils/CMakeLists.txt       | 14 +++++
 2 files changed, 94 insertions(+)
 create mode 100644 host/cmake/Modules/UHDAtomics.cmake

diff --git a/host/cmake/Modules/UHDAtomics.cmake b/host/cmake/Modules/UHDAtomics.cmake
new file mode 100644
index 000000000..6b43331a6
--- /dev/null
+++ b/host/cmake/Modules/UHDAtomics.cmake
@@ -0,0 +1,80 @@
+#
+# This file is mostly inspired by
+# https://github.com/llvm-mirror/llvm/blob/master/cmake/modules/CheckAtomic.cmake
+#
+# SPDX-License-Identifier: GPL-3.0-or-later
+#
+
+include(CheckCXXSourceCompiles)
+include(CheckLibraryExists)
+
+# Sometimes linking against libatomic is required for atomic ops, if
+# the platform doesn't support lock-free atomics.
+
+function(CHECK_WORKING_CXX_ATOMICS varname)
+    CHECK_CXX_SOURCE_COMPILES("
+	#include <atomic>
+	std::atomic<int> x;
+	int main() {
+	  return x;
+	}
+	" ${varname})
+endfunction(CHECK_WORKING_CXX_ATOMICS)
+
+function(CHECK_WORKING_CXX_ATOMICS64 varname)
+    CHECK_CXX_SOURCE_COMPILES("
+        #include <atomic>
+        #include <cstdint>
+        std::atomic<uint64_t> x (0);
+        int main() {
+          uint64_t i = x.load(std::memory_order_relaxed);
+          return 0;
+        }
+        " ${varname})
+endfunction(CHECK_WORKING_CXX_ATOMICS64)
+
+set(${required_var} FALSE)
+macro(CHECK_ATOMICS_LIB_REQUIRED required_var)
+    if(MSVC)
+        return()
+    else()
+        # First check if atomics work without the library.
+        CHECK_WORKING_CXX_ATOMICS(HAVE_CXX_ATOMICS_WITHOUT_LIB)
+        # If not, check if the library exists, and atomics work with it.
+        if(NOT HAVE_CXX_ATOMICS_WITHOUT_LIB)
+            check_library_exists(atomic __atomic_fetch_add_4 "" HAVE_LIBATOMIC)
+            if(HAVE_LIBATOMIC)
+                set(${required_var} TRUE)
+                set(CMAKE_REQUIRED_LIBRARIES "atomic")
+                CHECK_WORKING_CXX_ATOMICS(HAVE_CXX_ATOMICS_WITH_LIB)
+                if (NOT HAVE_CXX_ATOMICS_WITH_LIB)
+                    message(FATAL_ERROR "Host compiler must support std::atomic!")
+                endif()
+                unset(CMAKE_REQUIRED_LIBRARIES)
+            else()
+                message(
+                    FATAL_ERROR
+                    "Host compiler appears to require libatomic, but cannot find it.")
+            endif()
+        endif()
+        # Same check, but for 64-bit atomics.
+        CHECK_WORKING_CXX_ATOMICS64(HAVE_CXX_ATOMICS64_WITHOUT_LIB)
+        # If not, check if the library exists, and atomics work with it.
+        if(NOT HAVE_CXX_ATOMICS64_WITHOUT_LIB)
+            check_library_exists(atomic __atomic_load_8 "" HAVE_CXX_LIBATOMICS64)
+            if(HAVE_CXX_LIBATOMICS64)
+                set(${required_var} TRUE)
+                set(CMAKE_REQUIRED_LIBRARIES "atomic")
+                check_working_cxx_atomics64(HAVE_CXX_ATOMICS64_WITH_LIB)
+                unset(CMAKE_REQUIRED_LIBRARIES)
+                if (NOT HAVE_CXX_ATOMICS64_WITH_LIB)
+                    message(FATAL_ERROR "Host compiler must support 64-bit std::atomic!")
+                endif()
+            else()
+                message(
+                    FATAL_ERROR
+                    "Host compiler appears to require libatomic for 64-bit operations, but cannot find it.")
+            endif()
+        endif()
+    endif()
+endmacro(CHECK_ATOMICS_LIB_REQUIRED)
diff --git a/host/lib/utils/CMakeLists.txt b/host/lib/utils/CMakeLists.txt
index 0d18945ca..03f7933a0 100644
--- a/host/lib/utils/CMakeLists.txt
+++ b/host/lib/utils/CMakeLists.txt
@@ -177,6 +177,20 @@ set_source_files_properties(
     PROPERTIES COMPILE_DEFINITIONS "${LOAD_MODULES_DEFS}"
 )
 
+
+########################################################################
+# Check if we dynamically need to link against libatomic
+# "Guerney, deploy the family atomics!"
+########################################################################
+message(STATUS "")
+message(STATUS "Configuring atomics support...")
+include(UHDAtomics)
+CHECK_ATOMICS_LIB_REQUIRED(NEED_LIBATOMICS)
+if(NEED_LIBATOMICS)
+    message(STATUS "Linking against libatomic...")
+    LIBUHD_APPEND_LIBS("atomic")
+endif()
+
 ########################################################################
 # Define UHD_PKG_DATA_PATH for paths.cpp
 ########################################################################
-- 
2.25.1
```

Any ideas if I'm missing cases? I don't like the blanket-linking instruction, but I'm not sure if this code covers all the cases.

yarda, 2022-01-28:
It is still problem on the s390x, uhd-4.1.0.5.

yarda, 2022-01-28:
> It is still problem on the s390x, uhd-4.1.0.5.

gcc-12.0.1

Could you please reopen this issue or point to the commit fixing this problem(after the 4.1.0.5 release).

wordimont, 2022-01-28:
@yarda, the PR that automatically closed this issue was for this commit:
[cmake: Add check for libatomic linking requirement ](https://github.com/EttusResearch/uhd/commit/2680025c1b8ac51812931d78b2e0a4777f8e1774)

Under the v4.1.0.5 tag, that commit is here: 373184a6f543ae650f1dbc1dc74a875c6762fe76

Let us know if that's not the fix you were expecting.

yarda, 2022-01-28:
@wordimont thanks for the info. I checked it and I have the macro UHDAtomics.cmake and it's used during the configuration, but it doesn't seem to work for me on the s390x:
```
-- Performing Test HAVE_CXX_ATOMICS64_WITHOUT_LIB - Success
--   Atomics support is built-in, no linking required.
```
Which doesn't seem to be true, because the build fails on the s390x. Adding the `LIBUHD_APPEND_LIBS(atomic)` caused the build to proceed. So probably the macro is not correct.

yarda, 2022-01-28:
s390x failed build log:
https://kojipkgs.fedoraproject.org//work/tasks/3120/82083120/build.log

yarda, 2022-01-29:
> s390x failed build log: https://kojipkgs.fedoraproject.org//work/tasks/3120/82083120/build.log

The log is temporal, thus copying here the important parts of it:
```
-- Configuring atomics support...
-- Performing Test HAVE_CXX_ATOMICS_WITHOUT_LIB
-- Performing Test HAVE_CXX_ATOMICS_WITHOUT_LIB - Success
-- Performing Test HAVE_CXX_ATOMICS64_WITHOUT_LIB
-- Performing Test HAVE_CXX_ATOMICS64_WITHOUT_LIB - Success
--   Atomics support is built-in, no linking required.
...
/usr/bin/ld: ../lib/libuhd.so.4.1.0: undefined reference to `__atomic_compare_exchange_16'
/usr/bin/ld: ../lib/libuhd.so.4.1.0: undefined reference to `__atomic_store_16'
/usr/bin/ld: ../lib/libuhd.so.4.1.0: undefined reference to `__atomic_load_16'
collect2: error: ld returned 1 exit status
```

The patching part:
```
patching file host/cmake/Modules/UHDAtomics.cmake
```
is not related to this failure, because it was my second attempt to compile the code where I tried to add some more code to the detection macro not to be optimized out by the compiler (which probably isn't the case), but the result is the same even without the patch.

yarda, 2022-01-30:
It seems I got it. The problem is caused by the `boost/lockfree/queue.hpp` which is used on the different places of the uhd code, namely:
```
./host/lib/include/uhdlib/rfnoc/tx_async_msg_queue.hpp
./host/lib/transport/offload_io_service.cpp
./host/tests/common/mock_transport.hpp
```
It seems boost uses tagged pointers for the lockfree queue implementation which requires `-latomic` on the s390x, because the s390x is missing the corresponding `compare&exchange` instruction. The test which worked for me:
```
#include <boost/lockfree/queue.hpp>                                                                                                    
                                                                                                                                       
boost::lockfree::queue<int> queue(1);                                                                                                  
                                                                                                                                       
int main()                                                                                                                             
{                                                                                                                                      
   return queue.is_lock_free();                                                                                                        
}                                                                                                                                      
```
I think we need to add the test to the macro or create new macro.

yarda, 2022-01-30:
> the s390x is missing the corresponding `compare&exchange` instruction.

This maybe not true. The difference is probably caused by the content of the `/usr/include/boost/lockfree/detail/prefix.hpp`:
```
...
#if defined(__s390__) || defined(__s390x__)                                                                                            
    #define BOOST_LOCKFREE_CACHELINE_BYTES 256                                                                                         
#elif defined(powerpc) || defined(__powerpc__) || defined(__ppc__)                                                                     
    #define BOOST_LOCKFREE_CACHELINE_BYTES 128                                                                                         
#else                                                                                                                                  
    #define BOOST_LOCKFREE_CACHELINE_BYTES 64                                                                                          
#endif                                                                                                                                 
#include <boost/predef.h>

#if BOOST_ARCH_X86_64 || defined (__aarch64__)
#define BOOST_LOCKFREE_PTR_COMPRESSION 1
#endif
...
```
Which resulted in different padding and aligning of the structures. This explains why the cmake tests passed. I.e. new test is probably needed. The code from the previous comment worked for me. It compiles on the x86_64 but fails on the s390x (without the -latomic).

yarda, 2022-01-30:
```
  BOOST_LOCKFREE_PTR_COMPRESSION: use tag/pointer compression to utilize parts                                                        
                                   of the virtual address space as tag (at least 16bit)                                                
```
This is what compacts the tagged_pointer on the x86_64 and aarch64. Without it, the resulting code after the pre-processing requires 16 bytes long atomic compare&exchange and load&store instructions which are not available (even on the x86_64).

yarda, 2022-01-30:
There are two implementations with and without ptr compression which are selected by the following code from the `/usr/include/boost/lockfree/detail/tagged_ptr.hpp` wrapper:
```
#ifndef BOOST_LOCKFREE_PTR_COMPRESSION                                                                                                 
#include <boost/lockfree/detail/tagged_ptr_dcas.hpp>                                                                                   
#else                                                                                                                                  
#include <boost/lockfree/detail/tagged_ptr_ptrcompression.hpp>                                                                         
#endif                                                                                                                                 
```

yarda, 2022-01-31:
@wordimont or anbody could you please reopen the issue? Because the $subj is still not resolved, i.e. it still fails on the s390x. Or do you need new issue? Proposed test code for cmake is in the https://github.com/EttusResearch/uhd/issues/410#issuecomment-1025185119 I can also do PR if needed.

mbr0wn, 2022-02-01:
@yarda Since we don't have your Fedora build infrastructure, can you verify that this will fix the issue:
```patch
From 0d184bf708ab80f2685ecdce45359c17ac1b42cf Mon Sep 17 00:00:00 2001
From: Martin Braun <martin.braun@ettus.com>
Date: Mon, 31 Jan 2022 16:27:47 +0100
Subject: [PATCH] cmake: atomics: Improve atomics check

The existing check for built-in atomics support (added in 2680025) did
not catch all cases of requiring linkage against libatomic.so. This uses
Boost.Lockfree to catch another case, which fixes build issues on s390x.
---
 host/cmake/Modules/UHDAtomics.cmake | 5 +++++
 1 file changed, 5 insertions(+)

diff --git a/host/cmake/Modules/UHDAtomics.cmake b/host/cmake/Modules/UHDAtomics.cmake
index 4d53a2d0b..063c0658e 100644
--- a/host/cmake/Modules/UHDAtomics.cmake
+++ b/host/cmake/Modules/UHDAtomics.cmake
@@ -25,9 +25,14 @@ function(CHECK_WORKING_CXX_ATOMICS64 varname)
     CHECK_CXX_SOURCE_COMPILES("
         #include <atomic>
         #include <cstdint>
+        // Boost.Lockfree is a header-only library, and if we reach this, we
+        // have already checked for the existence of Boost.
+        #include <boost/lockfree/queue.hpp>
         std::atomic<uint64_t> x (0);
         int main() {
           uint64_t i = x.load(std::memory_order_relaxed);
+          boost::lockfree::queue<int> queue(1);
+          const bool lock_free = queue.is_lock_free();
           return 0;
         }
         " ${varname})
-- 
2.25.1
```

I basically took your code from above and added it to our existing test.

mbr0wn, 2022-02-01:
sorry, totally missed your PR.

mbr0wn, 2022-02-01:
Ignore my patch from above. We're taking your PR through our internal CI and review. Should be merged soon.

yarda, 2022-02-01:
> Ignore my patch from above. We're taking your PR through our internal CI and review. Should be merged soon.

I checked your patch, it also works OK, if you want feel free to use it instead of mine.

mbr0wn, 2022-02-01:
I actually like the separation of "does Boost need atomics" from the rest. Besides, you did the work, you should get the credit.

closed 2022-02-01 by atrnati