EttusResearch/uhd #412: -- Could NOT find DPDK (missing: DPDK_INCLUDE_DIRS DPDK_LIBRARIES) (Required is exact version "18.11")

jingxianwang, 2021-02-06:
## Issue Description ##
Compiling UHD from Source on OSX

## Setup Details ##
I run cmake ../host

Output:
-- 
-- Configuring the Python interpreter...
-- Manually determining build Python version...
-- Python interpreter: /opt/local/bin/python Version: 3.8.7
-- Override with: -DPYTHON_EXECUTABLE=<path-to-python>
-- Manually determining runtime Python version...
-- Python runtime interpreter: /opt/local/bin/python Version: 3.8.7
-- Override with: -DRUNTIME_PYTHON_EXECUTABLE=<path-to-python>
-- Finding Python Libraries...
-- Python Libraries: /opt/local/Library/Frameworks/Python.framework/Versions/3.8/Python
-- Python include directories: /opt/local/Library/Frameworks/Python.framework/Versions/3.8/Headers
-- Operating on master branch.
-- Using UHD Images Directory: /opt/local/share/uhd/images
-- Build type not specified: defaulting to release.
-- 
-- Configuring Boost C++ Libraries...
-- 
-- Checking for Boost version 1.58 or greater
--   Looking for required Boost components...
CMake Warning at /opt/local/share/cmake-3.10/Modules/FindBoost.cmake:801 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /opt/local/share/cmake-3.10/Modules/FindBoost.cmake:907 (_Boost_COMPONENT_DEPENDENCIES)
  /opt/local/share/cmake-3.10/Modules/FindBoost.cmake:1542 (_Boost_MISSING_DEPENDENCIES)
  cmake/Modules/UHDBoost.cmake:176 (find_package)
  CMakeLists.txt:296 (include)

CMake Warning at /opt/local/share/cmake-3.10/Modules/FindBoost.cmake:801 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /opt/local/share/cmake-3.10/Modules/FindBoost.cmake:907 (_Boost_COMPONENT_DEPENDENCIES)
  /opt/local/share/cmake-3.10/Modules/FindBoost.cmake:1542 (_Boost_MISSING_DEPENDENCIES)
  cmake/Modules/UHDBoost.cmake:176 (find_package)
  CMakeLists.txt:296 (include)

CMake Warning at /opt/local/share/cmake-3.10/Modules/FindBoost.cmake:801 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /opt/local/share/cmake-3.10/Modules/FindBoost.cmake:907 (_Boost_COMPONENT_DEPENDENCIES)
  /opt/local/share/cmake-3.10/Modules/FindBoost.cmake:1542 (_Boost_MISSING_DEPENDENCIES)
  cmake/Modules/UHDBoost.cmake:176 (find_package)
  CMakeLists.txt:296 (include)

CMake Warning at /opt/local/share/cmake-3.10/Modules/FindBoost.cmake:801 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /opt/local/share/cmake-3.10/Modules/FindBoost.cmake:907 (_Boost_COMPONENT_DEPENDENCIES)
  /opt/local/share/cmake-3.10/Modules/FindBoost.cmake:1542 (_Boost_MISSING_DEPENDENCIES)
  cmake/Modules/UHDBoost.cmake:176 (find_package)
  CMakeLists.txt:296 (include)

CMake Warning at /opt/local/share/cmake-3.10/Modules/FindBoost.cmake:801 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /opt/local/share/cmake-3.10/Modules/FindBoost.cmake:907 (_Boost_COMPONENT_DEPENDENCIES)
  /opt/local/share/cmake-3.10/Modules/FindBoost.cmake:1542 (_Boost_MISSING_DEPENDENCIES)
  cmake/Modules/UHDBoost.cmake:176 (find_package)
  CMakeLists.txt:296 (include)
[... 328 lines cut ...]

Run Build Command:"/usr/bin/make" "cmTC_06747/fast"
/Library/Developer/CommandLineTools/usr/bin/make -f CMakeFiles/cmTC_06747.dir/build.make CMakeFiles/cmTC_06747.dir/build
Building CXX object CMakeFiles/cmTC_06747.dir/CheckIncludeFile.cxx.o
/Library/Developer/CommandLineTools/usr/bin/clang++   -I/opt/local/include/libusb-1.0  -fvisibility=hidden -fvisibility-inlines-hidden  -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk   -std=gnu++14 -o CMakeFiles/cmTC_06747.dir/CheckIncludeFile.cxx.o -c /Users/jx/Documents/uhd/build/CMakeFiles/CMakeTmp/CheckIncludeFile.cxx
/Users/jx/Documents/uhd/build/CMakeFiles/CMakeTmp/CheckIncludeFile.cxx:1:10: fatal error: 'winsock2.h' file not found
#include <winsock2.h>
         ^~~~~~~~~~~~
1 error generated.
make[1]: *** [CMakeFiles/cmTC_06747.dir/CheckIncludeFile.cxx.o] Error 1
make: *** [cmTC_06747/fast] Error 2

Determining if the include file atlbase.h exists failed with the following output:
Change Dir: /Users/jx/Documents/uhd/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_a27df/fast"
/Library/Developer/CommandLineTools/usr/bin/make -f CMakeFiles/cmTC_a27df.dir/build.make CMakeFiles/cmTC_a27df.dir/build
Building CXX object CMakeFiles/cmTC_a27df.dir/CheckIncludeFile.cxx.o
/Library/Developer/CommandLineTools/usr/bin/clang++   -I/opt/local/include/libusb-1.0  -fvisibility=hidden -fvisibility-inlines-hidden  -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk   -std=gnu++14 -o CMakeFiles/cmTC_a27df.dir/CheckIncludeFile.cxx.o -c /Users/jx/Documents/uhd/build/CMakeFiles/CMakeTmp/CheckIncludeFile.cxx
/Users/jx/Documents/uhd/build/CMakeFiles/CMakeTmp/CheckIncludeFile.cxx:1:10: fatal error: 'atlbase.h' file not found
#include <atlbase.h>
         ^~~~~~~~~~~
1 error generated.
make[1]: *** [CMakeFiles/cmTC_a27df.dir/CheckIncludeFile.cxx.o] Error 1
make: *** [cmTC_a27df/fast] Error 2

michaelld, 2021-02-08:
UHD _can_ configure on OSX from 10.4/5 (PPC 32/64 && Intel 32/64) to current (Intel64 / ARM64), with the correct dependencies installed. Hence I'm closing this issue.

Please send an email to support@ettus.com with your issue & status. Please include some more information, so that I can help you better:

* Version of MacOS & Xcode ... for example run this script `echo "macOS $(sw_vers -productVersion) $(sw_vers -buildVersion)"; echo "Xcode $(xcodebuild -version)"`

* Version of UHD you're trying to build and how: from source? inside MacPorts? current GIT master or some specific branch or tag?

* The full configure / CMake info output / debug log.

NOTE: It is normal on OSX for `virtualenv` to fail. Here's what I see from within MacPorts
```
:info:configure -- Python checking for virtualenv
:info:configure -- Python checking for virtualenv - "assert hasattr(sys, 'real_prefix')" failed
```
This failure does not impact whether UHD can build or not; it just changes how UHD does some internal CMake things.

closed 2021-02-08 by michaelld

michaelld, 2021-02-08:
Related: Here's what the [DPDK folks note on their main webpage
](https://www.dpdk.org/)
```
Designed to run on x86, POWER and ARM processors, it runs mostly in Linux userland, with a FreeBSD port available for a subset of DPDK features.
```
To the best of my knowledge, DPDK doesn't build or work on MacOS / OSX of any version. In theory it _might_ be portable since they have a FreeBSD port, and Darwin / OSX is a variant on *BSD ... I really don't know! My strong advice is to not even try DPDK on macOS at this time _unless this is the end-goal of your work_ ; for casual UHD users just trying to get UHD running, just use MacPorts unless you can't for some reason.

jingxianwang, 2021-02-08:
Thank you so much Michael, that's helpful. I have a followup question: I was able to use UHD properly via MacPorts Installation. I have a dumb question, how to compile and build my own cpp application? I was able to do that on linux by adding it in cmakelist.txt and make. I am not sure how to do it on Mac when I use MacPorts Installation.

michaelld, 2021-02-08:
@jingxianwang Look in the source code directory `host/examples` and look at those individual examples for the `CMakeLists.txt` that can be used for out-of-tree builds ... roughly. Should be roughly the same as on any other OS, since CMake is so cross-platform portable.

jingxianwang, 2021-02-08:
The problem is it's not clear where the CMakeLists.txt is after the MacPorts installation. I can only find Unix executable files in the folder of /usr/local/lib/uhd/examples.