EttusResearch/uhd #413: uhd-4.0.0.0: error building zpu wireshark dissector with gcc-11

yarda, 2021-02-08:
## Issue Description ##
Error when building zpu wireshark dissector.

## Setup Details ##
Fedora 34 (rawhide), gcc-11, uhd-4.0.0.0, -DENABLE_ZPU=ON

## Expected Behavior ##
No error

## Actual Behaviour ##
```
[ 93%] Built target zpu
In file included from /usr/include/glib-2.0/glib/gmacros.h:241,
                 from /usr/lib64/glib-2.0/include/glibconfig.h:9,
                 from /usr/include/glib-2.0/glib/gtypes.h:32,
                 from /usr/include/glib-2.0/glib/galloca.h:32,
                 from /usr/include/glib-2.0/glib.h:30,
                 from /builddir/build/BUILD/uhd-4.0.0.0/tools/dissectors/epan/rfnoc/packet-rfnoc.cpp:17:
/usr/include/c++/11/type_traits:56:3: error: template with C linkage
   56 |   template<typename _Tp, _Tp __v>
      |   ^~~~~~~~
/builddir/build/BUILD/uhd-4.0.0.0/tools/dissectors/epan/rfnoc/packet-rfnoc.cpp:10:1: note: 'extern "C"' linkage started here
   10 | extern "C" {
      | ^~~~~~~~~~
In file included from /usr/include/glib-2.0/glib/gmacros.h:241,
                 from /usr/lib64/glib-2.0/include/glibconfig.h:9,
                 from /usr/include/glib-2.0/glib/gtypes.h:32,
                 from /usr/include/glib-2.0/glib/galloca.h:32,
                 from /usr/include/glib-2.0/glib.h:30,
                 from /builddir/build/BUILD/uhd-4.0.0.0/tools/dissectors/epan/rfnoc/packet-rfnoc.cpp:17:
/usr/include/c++/11/type_traits:71:3: error: template with C linkage
```

## Steps to reproduce the problem ##
Try to rebuild with -DENABLE_ZPU=ON on Fedora rawhide.

## Additional Information ##

yarda, 2021-02-08:
It seems with other dissectors the error is the same.

mbr0wn, 2021-09-23:
@yarda Thanks, as usual, for reporting this issues. It turns out glib.h should not be included with `extern "C"`. A fix is on the way.

closed 2021-09-28 by atrnati