EttusResearch/uhd #416: Typo in AN-400 causes rfnoc_image_builder to fail when building RFNoC image

atrnati, 2021-02-18:
## Issue Description ##
Mike on the USRP-users mailing list was following AN-400 to add an FPGA block to the E310 RFNoC image:

> I'm building a new FPGA image for E310 based on the yaml file
> "e310_rfnoc_image_core.yml".  I'm adding a FFT block per the
> instructions in "Getting_Started_with_RFNoC_in_UHD_4.0".
> 
> However, when I run:
> 
> rfnoc_image_builder -y ./e310_with_fft.yml -t E310
> 
> I get the error:
> 
> [ERR] 1 unresolved clk domain(s)
> [ERR]     fft0:ce
> [ERR] Please specify the clock(s) to connect
> 
> Even though the clk_domains is configured as:
> 
> clk_domains:
>   - { srcblk: _device_, srcport: radio, dstblk: radio0, dstport: radio }
>   - { srcblk: _device_, srcport: ce,    dstblk: fft0,   dstport:ce }
> 
> If I remove the FFT block the compile process begins properly.

The issue is the source block for the FFT's `ce` clock domain. `{ srcblk: _device_, srcport: ce }` should be `{ srcblk: _device_, srcport: rfnoc_chdr }` instead.

## Setup Details ##
* UHD 4.0

## Expected Behavior ##
AN-400 should be accurate and following its instructions to the letter should result in success.

## Actual Behaviour ##
The error in AN-400 results in the error described above, and the user is unsuccessful in creating the custom RFNoC image.

## Steps to reproduce the problem ##
Follow the instructions of AN-400 to the letter.

michaelld, 2021-02-18:
Should both of the `srcblk: _device_, srcport: ce` entries be updated to be the same?

atrnati, 2021-02-18:
> Should both of the `srcblk: _device_, srcport: ce` entries be updated to be the same?

Yes indeed.

michaelld, 2021-02-18:
Done

michaelld, 2021-02-18:
Direct link to AN-400, for reference: https://kb.ettus.com/Getting_Started_with_RFNoC_in_UHD_4.0

closed 2021-02-18 by michaelld

atrnati, 2021-02-18:
LOL, that was quick. Thanks for taking care of that! 👍

michaelld, 2021-02-18:
YW! For something this simple, it's best to just get it done ASAP IMHO!

f380cedric, 2021-02-18:
Is is a typo?

On x3x0 devices, ce_clk exists and is derived from CLK_OUT4 (214.286 MHz) while rfnoc_chdr_clk is derived from CLK_OUT1 (187.500 MHz) [[1]](https://github.com/EttusResearch/uhd/blob/3b9ced8f07c068faf1f494ce170cb44edaa47075/fpga/usrp3/top/x300/x300.v#L266).
However, ce_clk does not exists on e3x0.

wordimont, 2021-02-22:
@f380cedric you are correct. I'm not sure why we didn't add CE clock to E-series. CE clock is the preferred clock for other target types since RFNoC CHDR clock is too slow for some device configurations.