EttusResearch/uhd #417: rfnoc_image_builder crashes in python 3.5.2

jelder19, 2021-02-26:
## Issue Description ##
When running rfnoc_image_builder using python 3.5.2, crashes with the following error:
*** TypeError: '_sre.SRE_Match' object is not subscriptable
line 260 of image_builder.py -> if ident[0] in block["parameters"]:

## Setup Details ##
UHD latest commit of master branch
Ubuntu 16.04
python 3.5.2

## Expected Behavior ##
works

## Actual Behaviour ##
crashes

## Steps to reproduce the problem ##
cd ~/uhd/fpga/usrp3/top/x300
rfnoc_image_builder -y ./x310_rfnoc_image_core.yml -t X310_XG

## Additional Information ##
Turns out the regex module behavior changed in 3.7. in 3.5, you can't subscript the match result.
Replacing 'ident[0]' with 'ident.group(0)' on lines 260-262 fixes the problem for me.
I'm not sure if this works in 3.7+, so you guys will have to figure that out.

mbr0wn, 2021-06-17:
@jelder19 many thanks for the bug report. Using `group(0)` is correct and works fine across version. It is the right answer. We'll apply this fix soon.

mbr0wn, 2021-06-29:
@jelder19 we actually found a bunch more issues related to Python 3.5. However, we also decided to no longer support 3.5 in the upcoming release (UHD 4.1), among other things because we really want to start using some Python 3.6 features.

Here are the patches that will let you run UHD with Python 3.5:
```patch
From 570c377db41e1350167a6ede76ea1784ecc595ad Mon Sep 17 00:00:00 2001
From: Martin Braun <martin.braun@ettus.com>
Date: Mon, 28 Jun 2021 13:12:27 +0200
Subject: [PATCH 1/2] uhd: Remove Python 3.6-isms

As much as it pains my dear heart, we currently promise Python 3.5
support in our dependency list. This removes some usage of f-strings
from UHD which currently break Py3.5-support.
---
 host/lib/include/uhdlib/usrp/common/rpc.py    |  4 ++--
 host/python/uhd/usrp/cal/switch.py            |  5 +++--
 host/python/uhd/usrp/cal/usrp_calibrator.py   |  6 +++--
 host/tests/devtest/multi_usrp_test.py         | 22 +++++++++++--------
 .../run_X4xx_max_rate_tests.py                |  4 ++--
 host/tests/verify_fbs_test.py                 |  2 +-
 6 files changed, 25 insertions(+), 18 deletions(-)

diff --git a/host/lib/include/uhdlib/usrp/common/rpc.py b/host/lib/include/uhdlib/usrp/common/rpc.py
index 4ca30b07d..88f266951 100644
--- a/host/lib/include/uhdlib/usrp/common/rpc.py
+++ b/host/lib/include/uhdlib/usrp/common/rpc.py
@@ -14,13 +14,13 @@ class Function:
         self.does_return = return_type != "void"
         self.return_type = return_type
         self.arg_names = [arg[1] for arg in args]
-        self.rpcname = f"\"{function_name}\""
+        self.rpcname = "\"{}\"".format(function_name)
         self.args = [" ".join(arg) for arg in args]
         self.has_rpcprefix = False
         self.no_claim = no_claim
 
     def enable_rpcprefix(self):
-        self.rpcname = f"_rpc_prefix + \"{self.name}\""
+        self.rpcname = "_rpc_prefix + \"{}\"".format(self.name)
         self.has_rpcprefix = True
 
 class Interface:
diff --git a/host/python/uhd/usrp/cal/switch.py b/host/python/uhd/usrp/cal/switch.py
index b17318b5e..4c255e4b5 100644
--- a/host/python/uhd/usrp/cal/switch.py
+++ b/host/python/uhd/usrp/cal/switch.py
@@ -53,8 +53,9 @@ class ManualSwitch(SwitchBase):
         if self.mode == 'auto':
             return # no need to wait for manual connection
         dev_type = "signal generator" if self.direction == 'rx' else "power meter"
-        input(f"[{self.direction}] Connect your {dev_type} to device channel {chan}, "
-              f"antenna {antenna}. Then, hit Enter.")
+        input("[{}] Connect your {} to device channel {}, "
+              "antenna {}. Then, hit Enter."
+              .format(self.direction, dev_type, chan, antenna))
 
 class NISwitch(SwitchBase):
     """
diff --git a/host/python/uhd/usrp/cal/usrp_calibrator.py b/host/python/uhd/usrp/cal/usrp_calibrator.py
index c129fb17d..2a20aaf85 100644
--- a/host/python/uhd/usrp/cal/usrp_calibrator.py
+++ b/host/python/uhd/usrp/cal/usrp_calibrator.py
@@ -307,7 +307,8 @@ class USRPCalibratorBase:
             # result of the equation is in dBm again. We omit the subtract-by-zero
             # since our variables don't have units.
             results[gain] = usrp_input_power - recvd_signal_power
-            self.log(f"{gain:4.2f} dB => {results[gain]:+6.2f} dBm")
+            self.log("{:4.2f} dB => {:+6.2f} dBm"
+                     .format(gain, results[gain]))
             # If we get too close to the noise floor, we stop
             if recvd_power - self._noise[freq][gain] <= 1.5:
[... 113 lines cut ...]
---
 host/python/uhd/imgbuilder/image_builder.py | 7 ++++---
 1 file changed, 4 insertions(+), 3 deletions(-)

diff --git a/host/python/uhd/imgbuilder/image_builder.py b/host/python/uhd/imgbuilder/image_builder.py
index 9ce57d0e0..a99a51076 100755
--- a/host/python/uhd/imgbuilder/image_builder.py
+++ b/host/python/uhd/imgbuilder/image_builder.py
@@ -259,9 +259,10 @@ class ImageBuilderConfig:
                             # place. That may result in an exception being
                             # thrown from eval(), but we'll catch it and
                             # report an error a bit later on.
-                            if ident[0] in block["parameters"]:
-                                val = str(block["parameters"][ident[0]])
-                                num_ports = re.sub(ident[0], val, num_ports)
+                            parameter = ident.group(0)
+                            if parameter in block["parameters"]:
+                                val = str(block["parameters"][parameter])
+                                num_ports = re.sub(parameter, val, num_ports)
 
                         # Now, with identifiers resolved to parameter values,
                         # attempt to evaluate the expression. If eval() fails,
-- 
2.25.1
```

mbr0wn, 2021-06-29:
Closing, because we will be using Python 3.6 going forward.

closed 2021-06-29 by mbr0wn