EttusResearch/uhd #420: UHD 4.0: self-calibration of N320/N321 not working

karel, 2021-03-04:
[UHD's calibration utilities](https://files.ettus.com/manual/page_calibration.html) do not work with UHD 4.0 and N320/N321. It seems that the calibration files will also not load, should they exist from previous UHD versions and be converted.

Firstly, the following comparison fails, as "n3xx" is compared against "n320".

https://github.com/EttusResearch/uhd/blob/3b9ced8f07c068faf1f494ce170cb44edaa47075/host/utils/usrp_cal_utils.hpp#L58

After altering that comparison to look for "n320", calibration utilities terminate with an error. Seemingly the functionality is not implemented in `rfnoc` namespace as it is in `usrp`?

https://github.com/EttusResearch/uhd/blob/3b9ced8f07c068faf1f494ce170cb44edaa47075/host/lib/rfnoc/radio_control_impl.cpp#L801

Having downgraded back to UHD 3.15, I can not provide exact outputs and logs. However, the issue was verified by another user on USRP IRC and should be fairly straightforward to reproduce.

mbr0wn, 2021-03-08:
@karel yup, good find, many thanks. Something must have slipped through the cracks during the 4.0 porting.

karel, 2021-03-26:
With the patches to "rhodium_radio_control.cpp", "rhodium_radio_control.hpp", and "rhodium_radio_control_init.cpp", the calibration utilities do run but with seemingly poor results (based on the calibration utilities reported results). 

See these [USRP N321 self-calibration results](https://gist.github.com/karel/55621cc47ae5e8bfe74afe62161b4ab3) with the exact same configuration.

michaelld, 2021-03-26:
FYI I provided @karel with the (very simple) patch to "rhodium_radio_control_init.cpp" from https://github.com/EttusResearch/uhddev/pull/5377 even though it's not technically part of UHD yet.

@karel what patches are you referring to for "rhodium_radio_control.cpp" and "rhodium_radio_control.hpp"?

karel, 2021-03-26:
Subsequently @mbr0wn also shared the "[PATCH] n320: Add correction APIs to radio_control object".

michaelld, 2021-03-26:
Gotcha. Thx!

arakawa2019, 2021-03-31:
Hi, I'm also facing with not loading calibration files on n320 & UHD 4.0. 

You mentioned EttusResearch/uhddev for possible modifications, but I cannot reach the Web site. 
Is this a private repository? Can you provide the (very simple) patch to us?

arakawa2019, 2021-03-31:
In my environment, db_serial_u8 is empty and db_serial is set to ''unknown'' at _update_corrections of rhodium_radio_control.cpp.
Hard-coding the serial number successfully loads calibration files (confirmed by UHD_LOG).  So, my issue is solved.

I have generated calibration files using calibration utilities with UHD 3.15. I didn't run the utilities with UHD 4.0.

michaelld, 2021-04-01:
@arakawa2019 thanks for the updates; I'm glad you worked out how to load calibration!

mbr0wn, 2021-05-04:
65617b2f67dd7c6176f467d2f2920e2026583ba3 is the last commit required.

closed 2021-05-04 by mbr0wn