EttusResearch/uhd #421: [ERROR] [STREAMER] The receive packet handler caught a value exception. B210

elmaghba, 2021-03-06:
## Issue Description ##

In my basic flowgraph, I have a USRP source that is connected to a head block to control the capturing time. Then, the head is connected to two different low-pass filters which each one of them is connected to a file sink block. The low-pass filters have the same sample rate as the USRP source block. 
When I run the flowgraph, it prints some zeros in the terminal, then, it gives me the following error: 

USRP Source Block caught rx error code: 15
[ERROR] [STREAMER] The receive packet handler caught a value exception.
ValueError: bad vrt header or packet fragment

## Setup Details ##
I'm running the gnuradio/uhd on a VM virtual box-Linux Ubuntu 18.
USRP B210

## Expected Behavior ##
I captured the data for 4 devices, then, it stopped working! 

## Actual Behaviour ##
Prints (00000000000000000000), then: 
USRP Source Block caught rx error code: 15
[ERROR] [STREAMER] The receive packet handler caught a value exception.
ValueError: bad vrt header or packet fragment

elmaghba, 2021-05-20:
I'm still getting the same issue. 
If there is any further information I need to provide, please let me know!

Any help is really appreciated!

mbr0wn, 2021-09-17:
@elmaghba Can you try this outside of a VM? We have often heard about USB issues when running in virtualbox. Also, can you post a flow graph here that exhibits the issue? And which UHD version were you using?

fatafalihhilmi, 2022-08-21:
I'm still getting the same issue.
If there is any further information I need to provide, please let me know!
 @mbr0wn  I use 4.2.0.1 UHD version

Aspect1398, 2023-08-22:
[ERROR] [STREAMER] The receive packet handler caught a value exception.
ValueError: bad vrt header or packet fragment
[ERROR] [STREAMER] recv packet demuxer unexpected sid 0x22000b

Aspect1398, 2023-08-22:
I am running eNB and it starts correctly, works well. Then out of nowhere, the error pops up and my eNB is dead.

yoursunny, 2024-05-15:
I'm trying USRP B210 with OpenAirInterface5G gNB.
I'm experience the same issue:

```text
[0m[93m[NR_MAC]   Normalized power 4 based on current resources (RBs 5, MCS 0) exceed reported PHR -7 (normalized value)
[0m[93m[NR_MAC]   Normalized power 4 based on current resources (RBs 5, MCS 0) exceed reported PHR -7 (normalized value)
[0m[93m[NR_MAC]   Normalized power 4 based on current resources (RBs 5, MCS 0) exceed reported PHR -7 (normalized value)
[0m[1;31m[HW]   [recv] received 3548 samples out of 23040
[0m[1;31m[HW]   ERROR_CODE_BAD_PACKET

[ERROR] [STREAMER] The receive packet handler caught a value exception.
ValueError: bad vrt header or packet fragment
[0m[1;31m[PHY]   rx_rf: Asked for 23040 samples, got 3548 from USRP
[0m[1;31m[PHY]   problem receiving samples
[ERROR] [STREAMER] recv packet demuxer unexpected sid 0x49c0264
[0m[1;31m[HW]   [recv] received 0 samples out of 23040
[0m[1;31m[HW]   ERROR_CODE_TIMEOUT
```

Environment:

* Xeon Gold 6240, "performance" governor on all cores
* Ubuntu 22.04
* linux-image-5.15.0-107-lowlatency
* `oaisoftwarealliance/oai-gnb:2024.w17` Docker image (container has full privileges)
* UHD version 4.4.0.HEAD-0-g5fac246b (4.4.0)

mbr0wn, 2024-06-25:
This is too hard to pinpoint / reproduce / too rare. Closing.

closed 2024-06-25 by mbr0wn