EttusResearch/uhd #423: rfnoc_chdr_utils.vh not easily included from OOT development

rkossler, 2021-03-10:
## Issue Description ##

The include file rfnoc_chdr_utils.vh is not easily included from an OOT RFNoC development because the include path is relative to the OOT dir.

## Setup Details ##

N/A

## Expected Behavior ##

N/A

## Actual Behaviour ##

N/A

## Steps to reproduce the problem ##

Try to include the file from an example OOT folder. My expectation is that you will find that you need to use either an absolute path or else a fairly complex relative path.  Either solution is not very robust.

## Additional Information ##

I do not know of the solution or even if there is a solution.  But, the current situation is not convenient.

mbr0wn, 2021-09-06:
@rkossler This was great feedback for general usability. We'll be adding `UHD_FPGA_DIR` as a definition to the synthesis tools which can be used for including UHD code.

wordimont, 2021-09-08:
These are examples of how to include an in-tree header file in your OOT block. Tools vary somewhat in how they support using macros in `include statements.

This works in Vivado:
```
`include `"`UHD_FPGA_DIR/usrp3/lib/rfnoc/core/rfnoc_chdr_utils.vh`"
```
Some tools allow this:
```
`define INCLUDE_UHD_FILE(REL_PATH) `"`UHD_FPGA_DIR/REL_PATH`"
`include `INCLUDE_UHD_FILE(usrp3/lib/rfnoc/core/rfnoc_chdr_utils.vh)
```
This should work in most tools:
```
`define RFNOC_CHDR_UTILS_PATH `"`UHD_FPGA_DIR/usrp3/lib/rfnoc/core/rfnoc_chdr_utils.vh`"
`include `RFNOC_CHDR_UTILS_PATH
```

closed 2021-09-08 by wordimont