EttusResearch/uhd #424: trx_uhd_read: error code = 0xf  when run Amarisoft 5G SA

cn0xroot, 2021-03-16:
Info:

```
gnuradio-config-info -v
3.7.14.0

uhd_find_devices 
[INFO] [UHD] linux; GNU C++ version 7.5.0; Boost_106501; UHD_3.11.0.0-0-unknown
--------------------------------------------------
-- UHD Device 0

Amarisoft /lteenb-linux-2021-02-26

```

./lteenb config/B210SA.cfg

UHD: 1
UHD: 1
UHD: 1
UHD: 1
UHD: 1
UHD: 1
UHD: 1
UHD: 1
UHD: 1
trx_uhd_read: error code = 0xf
[ERROR] [STREAMER] The receive packet handler caught a value exception.
ValueError: bad vrt header or packet fragment
terminate called after throwing an instance of 'boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::lock_error> >'

cn0xroot, 2021-03-16:
config file :  https://github.com/cn0xroot/Amarisoft_USRP/blob/main/SA_MIMO.cfg

mbr0wn, 2024-03-19:
We can't repro. Also, not a lot of info here.