EttusResearch/uhd #425: Fix uhd as cmake subproject

marcobergamin, 2021-03-29:
## Issue Description ##
uhd can't be built as cmake subproject

## Expected Behavior ##
Possibility to build uhd both as stand-alone project and as subproject

## Actual Behaviour ##
Does not build as subproject because uhd calculates some paths relative to the root CMakeLists.txt location and not relative to the root CMakeLists.txt of the uhd project.

## Steps to reproduce the problem ##
1. Create a project with it's own CMakelists.txt
2. Inside the CMakeLists.txt:
```
add_subdirectory(uhd/host) 
```
3. Try to build the target uhd

## Proposed fix ##
replace CMAKE_SOURCE_DIR with CMAKE_CURRENT_SOURCE_DIR
replate CMAKE_BINARY_DIR with CMAKE_CURRENT_BINARY_DIR

This fix does not change the existing behaviour because when uhd is built as stand-alone project
* CMAKE_SOURCE_DIR  has the same value of CMAKE_CURRENT_SOURCE_DIR
* CMAKE_BINARY_DIR has the same value of CMAKE_CURRENT_BINARY_DIR

michaelld, 2021-03-29:
To be more complete for anyone reading this:

In CMake 3.0, a build variable [`<PROJECT-NAME>_SOURCE_DIR` was introduced](https://cmake.org/cmake/help/v3.0/variable/PROJECT-NAME_SOURCE_DIR.html) that refers to the "Top level source directory for the named project". In CMake 3.0.2 the build variable [`CMAKE_SOURCE_DIR` was introduced](https://cmake.org/cmake/help/v3.0/variable/CMAKE_SOURCE_DIR.html) that refers to "The path to the top level of the source tree". For a standalone project -- one that is not added as a subdirectory of another -- these variables have the same value. The same holds true for the `*_BINARY_DIR` variables. On the other hand if using the former vairables then the project _can_ be included as a subdirectory that the top level source / binary directories would be correctly used whereas when using the latter the very top level project's source / binary directories would be used & UHD would clearly fail the `cmake` configuration step.

As the [CMake "project" for UHD is `UHD`](https://github.com/EttusResearch/uhd/blob/master/host/CMakeLists.txt#L17), the former variable is `UHD_SOURCE_DIR` ... UHD [requires CMake 3.8](https://github.com/EttusResearch/uhd/blob/master/host/CMakeLists.txt#L13) so these variables are valid.

As noted: This fix does not change the existing behavior because when UHD is built as stand-alone project; it just adds the capability for UHD to be included as a subdirectory by other projects.

@marcobergamin It won't be difficult to create a PR for this issue. Would you mind doing so?

marcobergamin, 2021-03-29:
Yes, I'll do it asap. Thanks for the integration

michaelld, 2021-03-29:
@marcobergamin I don't guarantee that the PR will be accepted, but providing it greatly increases the chances! Thx!

closed 2021-09-10 by atrnati