EttusResearch/uhd #429: Python API:  Number of dropped samples is incorrectly reported to be zero

bhorsfield, 2021-04-14:
## Issue Description ##
I want to use the Python API to capture short bursts of Rx samples from my USRP B210.  As a starting point, I have been evaluating the example script “benchmark_rate.py”.  This has revealed the following bug:  

- The number of dropped samples that is reported in the summary at the end of the script is always zero, even if a large number of overruns occurred during measurement. 

A typical output from the Python script is as follows:

[16:02:55.54] [INFO] (MainThread) Benchmark rate summary:
    Num received samples:     239622480
    **Num dropped samples:      0    [THIS NUMBER IS ALWAYS ZERO]**
    Num overruns detected:    16
    Num transmitted samples:  0
    Num sequence errors (Tx): 0
    Num sequence errors (Rx): 32
    Num underruns detected:   0
    Num late commands:        0
    Num timeouts (Tx):        0
    Num timeouts (Rx):        0

If I run the C++ version of this script, I get the correct output (i.e. a non-zero value for the number of dropped samples):

Benchmark rate summary:
  Num received samples:     241293678
  **Num dropped samples:      4265641    [NON-ZERO, AS EXPECTED]**
  Num overruns detected:    12
  Num transmitted samples:  0
  Num sequence errors (Tx): 0
  Num sequence errors (Rx): 24
  Num underruns detected:   0
  Num late commands:        0
  Num timeouts (Tx):        0
  Num timeouts (Rx):        0

## Setup Details ##

USRP B210 with internal GPSDO: GPSTCXO , Firmware Rev 0.929a, FPGA ver 16.0
Ubuntu 18.04; GNU C++ version 7.5.0; Boost_106501; UHD_4.0.0.0-46-g57ca4235
HP Omen laptop, Intel Core i7-8750H CPU, 32GB RAM, 512GB SSD
Operating over USB 3.0

## Expected Behavior ##
The Python script "benchmark_rate.py" should return the number of dropped samples if an overrun occurs during an Rx measurement. 

## Actual Behaviour ##
Script always returns zero for the number of dropped samples, even if an overrun has occurred.

## Steps to reproduce the problem ##

Run the example Python script "benchmark_rate.py" from the command line with the following settings:
`$ ./benchmark_rate.py --rx_rate 25e6 --duration 10 --channels 0 1`  

Compare this with the C++ version:
`$ ./benchmark_rate --rx_rate 25e6 --duration 10 --channels 0,1`

## Additional Information ##

After further investigation I have discovered that the values of `metadata.time_spec.get_real_secs()` and `last_overflow.get_real_secs()` in the method `benchmark_rx_rate()` are always identical to each other.  This is definitely incorrect, and will give a result of zero for the number of dropped samples every time.

bhorsfield, 2021-04-15:
Note that I am using Python 3.7.7.

mbr0wn, 2021-09-22:
This is a small, subtle Python-usage bug. @bhorsfield, thanks for reporting. You have a good eye!
```patch
commit 1a6619da7a4853c779800f0a244a7a00bba5d780 (HEAD -> mbr0wn/fix-python-benchmark-rate)
Author: Martin Braun <martin.braun@ettus.com>
Date:   Wed Sep 22 16:17:08 2021 +0200

python: Fix dropped-sample calculation in benchmark_rate.py

This fixes a subtle bug, where a variable to cache the timestamp of an
error gets bound to the metadata instead of creating a copy thereof.
Without this fix, the calculation of dropped samples would always be 0,
because the difference in timestamps would incorrectly be always zero.
This fix will now make a copy of the timestamp.

Shoutout to GitHub user bhorsfield for finding this issue.

diff --git a/host/examples/python/benchmark_rate.py b/host/examples/python/benchmark_rate.py
index a2ef1047b5..6d1f4626ec 100755
--- a/host/examples/python/benchmark_rate.py
+++ b/host/examples/python/benchmark_rate.py
@@ -9,6 +9,7 @@
Benchmark rate using Python API
"""

+import copy
import argparse
from datetime import datetime, timedelta
import sys
@@ -210,16 +211,18 @@ def benchmark_rx_rate(usrp, rx_streamer, random, timer_elapsed_event, rx_statist
# Reset the overflow flag
if had_an_overflow:
had_an_overflow = False
-                num_rx_dropped += uhd.types.TimeSpec(
-                    metadata.time_spec.get_real_secs() - last_overflow.get_real_secs()
-                ).to_ticks(rate)
+                num_rx_dropped += (metadata.time_spec - last_overflow).to_ticks(rate)
elif metadata.error_code == uhd.types.RXMetadataErrorCode.overflow:
had_an_overflow = True
-            last_overflow = metadata.time_spec
+            # Need to make sure that last_overflow is a new TimeSpec object, not
+            # a reference to metadata.time_spec, or it would not be useful
+            # further up.
+            last_overflow = uhd.types.TimeSpec(
+                metadata.time_spec.get_full_secs(),
+                metadata.time_spec.get_frac_secs())
# If we had a sequence error, record it
if metadata.out_of_sequence:
num_rx_seqerr += 1
-                logger.warning("Detected RX sequence error.")
# Otherwise just count the overrun
else:
num_rx_overruns += 1
```

bhorsfield, 2021-09-24:
Hi @mbr0wn, 

Thanks for your efforts in chasing up this problem.  I have implemented your changes to `benchmark_rate.py`, and can confirm that the reported number of dropped samples when an overrun occurs is now non-zero.

However, I have discovered an odd quirk in the new code:  If I perform an Rx measurement on only one channel, the number of dropped samples is always **negative**.  With a two-channel measurement, the number of dropped samples is always **positive**.

For example:
1-channel measurement:  `$ ./benchmark_rate.py --rx_rate 50e6 --duration 10 --channels 0`
Result:  Num dropped samples:  -22190  **(SHOULD BE A POSITIVE NUMBER)**

2-channel measurement:  `$ ./benchmark_rate.py --rx_rate 25e6 --duration 10 --channels 0 1`
Result:  Num dropped samples:  3435876  **(OK)**

Do you have any idea why this is occurring?

mbr0wn, 2021-09-24:
Is this still on a B210? In 3a4073799db9cf314b57eb20bb8f8fc085a76631, we added some code to catch these cases (but I honestly can't remember right now how/why they occurred).

mbr0wn, 2021-09-24:
My way testing of this was to `#define ERROR_INJECT_DROPPED_PACKETS` in super_recv_packet_handler.hpp, which did not cause this issue.

mbr0wn, 2021-09-24:
Come to think of it... how do you end up with so many dropped packets? 
What USB controller do you have?

bhorsfield, 2021-09-28:
My USB controller is:  Intel(R) USB 3.1 eXtensible Host Controller 

In the examples I provided, I deliberately chose a sampling rate that would cause lots of dropped packets.  Past experience has taught me that 25 Msps on 2 channels is simply too fast for USB 3 (at least on my machine), even with a RAM disk and the CPU governor set to "performance" on all 12 cores.

Could the negative number of dropped samples be due to the counter variable `num_rx_dropped` overflowing, e.g. because it is defaulting to a 16-bit signed integer type rather than 32-bit?

bhorsfield, 2021-09-28:
Note that all of my measurements have been performed using my B210 unit.

mbr0wn, 2021-09-28:
I don't think Python uses 16-bit integers... does it? That would surprise me. In any case, I don't think that's the issue. I have a hunch that `metadata.time_spec - last_overflow` is negative, although I don't know how/why.

The USB controller seems fine... Intels usually don't give us grief. I'm still unhappy about the many dropped packets.

mbr0wn, 2021-09-28:
Note that in dual-chan mode, we use timed commands to restart the streamer. I wonder if there's some oddity on the FPGA that causes timestamps to be ordered in a weird way.

closed 2021-09-28 by atrnati