EttusResearch/uhd #430: Python API:  Metadata object often fails to record overflow events during a receive measurement

bhorsfield, 2021-04-15:
## Issue Description ##
I am using the Python API to record short bursts of Rx samples from my B210.  As each packet comes in, I check the metadata object so see if any errors have occurred, and print a warning message on the screen if they have.  

Unfortunately, it appears that some overflow events are not being recorded by the metadata object.  In a significant number of cases, the metadata object returns `uhd.types.RXMetadataErrorCode.none`, even when the Fastpath logger prints 'O' to the console.  This means I do not have a reliable way to programmatically detect an overflow event. 

## Setup Details ##
USRP B210 with internal GPSDO: GPSTCXO , Firmware Rev 0.929a, FPGA ver 16.0
Ubuntu 18.04; GNU C++ version 7.5.0; Boost_106501; UHD_4.0.0.0-46-g57ca4235
Python 3.7.7
HP Omen laptop, Intel Core i7-8750H CPU, 32GB RAM, 512GB SSD
Operating over USB 3.0

## Expected Behavior ##
Whenever an overflow occurs during the Rx sampling process, a 'O' should be printed to the console by the Fastpath logger, and the metadata object should return something other than `uhd.types.RXMetadataErrorCode.none`.  My script should then print a warning message on the screen reading `Error: Packet-X caused an overflow`. 

## Actual Behaviour ##
Whenever an overflow occurs during the Rx measurement process, an 'O' is always printed to the console by the Fastpath logger as expected, but the metadata object frequently returns `uhd.types.RXMetadataErrorCode.none`.  In some cases the Fastpath logger has returned 8-10 overflows, while the metadata object finds zero errors of any kind.  As a result, my script is unable to reliably detect overflow events, meaning that I cannot guarantee the integrity of my sampled data. 

A typical output from the script is as follows:

[INFO] [B200] Actually got clock rate 25.000000 MHz.
Using Device: Single USRP:
  Device: B-Series Device
  Mboard 0: B210
  RX Channel: 0
    RX DSP: 0
    RX Dboard: A
    RX Subdev: FE-RX2
  RX Channel: 1
    RX DSP: 1
    RX Dboard: A
    RX Subdev: FE-RX1
  TX Channel: 0
    TX DSP: 0
    TX Dboard: A
    TX Subdev: FE-TX2
  TX Channel: 1
    TX DSP: 1
    TX Dboard: A
    TX Subdev: FE-TX1

OOOOOOOO    **[NOTE: SHOULD SEE A WARNING MESSAGE AFTER EACH 'O']**

Elapsed time: 154.186 seconds
Time per batch: 154.186 milliseconds

## Steps to reproduce the problem ##
Run the attached Python script using the command `$ ./python_api_overflow_bug.py`.  The problem will usually manifest itself within the first 30 seconds.  If problem does not occur, increase the parameter `target_num_samples` from 50,000 to 80,000 (or even higher). 
[python_api_overflow_bug.zip](https://github.com/EttusResearch/uhd/files/6315850/python_api_overflow_bug.zip)

## Additional Information ##
I should stress that I am not concerned about the overflows per se.  They do not surprise me, given the high sampling rate I am using.  Rather, I am concerned about the fact that the Python script cannot detect an overflow event programmatically.

jrocnuck, 2021-04-24:
This may not fix your problem, but it may help others who land here looking at your issue.

I have seen similar problems with UHD 4 and an X300 radio on Fedora 33 Linux.
The scenario is to run a receiver for about 150 seconds with any sample rate / freq / gain (and also with mtu 8000 or 1742).  Then stop the receiver for about 150 seconds and attempt to restart it (all in the same running process).  Instantly I get an async message followed by the timeout / overflow issue.    The radio basically will never clear its state in the firmware from this point forward no matter what stream command is issued.

I have partially figured out what happens.  In my case, I have the standard firewalld (with default rules/settings) running and once I issue the restart of the radio (after the long wait), I get a burst of data packets and then I see my host barking ICMP DESTINATION UNREACHABLE packets back to the x300, however those are blocked by the firewall, resulting in a repeatable NON-RECOVERABLE state.

When I disable the firewall, I do not see these ICMP DESTINATION unreachable messages going from the host to the x300.

I have dug fairly deep in all the diagnostic logs, etc. on the host and cannot see any cases where UDP packet errors/drops occur.  I haven't enabled extensive logging in the firewall yet to see if it's causing the issues of the destination UDP port/address being unreachable.  At this point, if I had to guess, the firewall is causing enough of a delay to cause the overflow situation.  The bad news is this means if a process ever overflowed long enough, the radio would become unrecoverable.

I am still shocked that this problem of NON-RECOVERY has not been fixed in the drivers after all this time.  I've seen messages in forums dating back to 2017/2018 with this state in the drivers.
The bottom line when a stop stream command is issued, this driver needs to clear any bad state in the control processing period and start fresh.  But from what I can tell, a stop command does not clear previous run inconsistent state.

From a reliability standpoint, that's a show stopper (basically unacceptable) for deployments running in real time where an entire radio (multiple channels) will require reset of all channels to recover one.

mbr0wn, 2021-09-21:
@bhorsfield OK, this was a real headscratcher for me. Well, I could have been smarter debugging... but anyway, I have a response for you: There is no bug here. Rather, it's an idiosyncrasy of how the B200 works (and some other USRPs).

Let me explain: When the device overruns, it will put a special packet into the pipeline, which tells UHD "there's no data here, there's an overrun". It will do this when all the FIFOs between your application and the ADC are full (well, when they are full minus one packet, which we need for the overrun message).

The first time `recv()` will tell you there was an overrun is when it has received all the valid samples in the queue. However, the queue is bigger than your number of packets. That's why you see the O but not the overrun error code: You're getting it in the flush procedure. If you add this:

```python
        while samps:
            if metadata.error_code == uhd.types.RXMetadataErrorCode.overflow:
                print("Error: Packet-%d caused an overflow!\n" % pkt_idx)
            samps = rx_streamer.recv(recv_buffer, metadata, timeout=1)
        # Set Rx streamer to None to help with garbage collection
        rx_streamer = None
```

...you will see the error code.

I will try and improve the docs about this. But it's not a bug.

Thanks for the report though. I added some additional testing while trying to figure out the issue here, so that's an improvement to UHD.