EttusResearch/uhd #432: why is there no master cmakelist.txt ?

nielsvanetten, 2021-04-15:
## Issue Description ##

## Setup Details ##

## Expected Behavior ##

## Actual Behaviour ##

## Steps to reproduce the problem ##

## Additional Information ##

brentstapleton, 2021-04-15:
Hi @nielsvanetten, there are many CMakeLists.txt files in uhd. Please take a look at the [README.md](https://github.com/EttusResearch/uhd/blob/master/README.md), which in turn links to the [instructions to build from source](https://files.ettus.com/manual/page_build_guide.html).

I think what you're asking for here is the CMakeLists.txt to build the UHD host applications and library, which can be [found here](https://github.com/EttusResearch/uhd/blob/master/host/CMakeLists.txt). Hope this helps. Otherwise, please add more details for what you're looking for.

mbr0wn, 2021-04-16:
Thanks @brentstapleton! @nielsvanetten, I hope this clears things up for you.

closed 2021-04-16 by mbr0wn