EttusResearch/uhd #437: Explicit boost/math/special_functions.hpp required with Boost 1.76

StefanBruens, 2021-05-05:
https://github.com/EttusResearch/uhd/blob/0813efb9a116d73300f6690b223295cf10a2f723/host/lib/usrp/usrp2/usrp2_impl.cpp#L920

Instead of the boost functions, std::copysign and std::round can be used as a replacement.

mbr0wn, 2021-06-22:
Fix is on the way!

StefanBruens, 2021-06-22:
> Fix is on the way!

Fix is available in PR #439 for 6 weeks ...

loqs, 2021-06-23:
@mbrown does the fix cover host/lib/usrp/cores/rx_dsp_core_3000.cpp and host/lib/usrp/cores/rx_frontend_core_3000.cpp in addition to host/lib/usrp/usrp2/usrp2_impl.cpp?

mbr0wn, 2021-06-28:
> @mbr0wn does the fix cover host/lib/usrp/cores/rx_dsp_core_3000.cpp and host/lib/usrp/cores/rx_frontend_core_3000.cpp in addition to host/lib/usrp/usrp2/usrp2_impl.cpp?

Yes. Will go public this week.

mbr0wn, 2021-06-29:
Fix is public.

closed 2021-06-29 by mbr0wn

loqs, 2021-06-29:
> > @mbr0wn does the fix cover host/lib/usrp/cores/rx_dsp_core_3000.cpp and host/lib/usrp/cores/rx_frontend_core_3000.cpp in addition to host/lib/usrp/usrp2/usrp2_impl.cpp?
> 
> Yes. Will go public this week.

@mbr0wn Build is still failing rx_dsp_core_3000.cpp and rx_frontend_core_3000.cpp.  What is supposed to be importing boost::math::sign ?

loqs, 2021-07-01:
Fixed by https://github.com/EttusResearch/uhd/files/6750118/patch-UHD-4.1_build_with_Boost_1.76.0.diff.txt https://github.com/EttusResearch/uhd/issues/450#issuecomment-872461864
Just leaves the left over headers.

michaelld, 2021-07-01:
@loqs can you explain "Just leaves the left over headers"?