EttusResearch/uhd #441: replay_block_control.hpp header no longer copied to install

ciarantoner, 2021-05-11:
## Issue Description ##
The file 'replay_block_control.hpp' is no longer copied to the install area. 

## Setup Details ##
This was a previous issue https://github.com/EttusResearch/uhd/issues/419 that was resolved however this commit:
3cc77a8e3f4823e8e969fb13bf782cde02f1880b 
seems to have removed 'replay_block_control.hpp' from the CMakeLists file therefore replay_block_control.hpp is not copied to the install area.

This is on the UHD-4.0 

## Expected Behavior ##
replay_block_control.hpp to be copied

## Actual Behaviour ##
replay_block_control.hpp is missing

## Steps to reproduce the problem ##
Build uhd from UHD-4.0 branch and look for replay_block_control.hpp in the install area

## Additional Information ##

michaelld, 2021-05-14:
@ciarantoner we are discussing internally. someone will reply here (sooner or later) with the decision. In the meantime, you can always go with the GIT master branch since it contains this fix & lots of others & is the future of UHD ...

mbr0wn, 2021-06-17:
Fixed by 24773a9afbc2901e7b38c679c5fd140c1526d76f

closed 2021-06-17 by mbr0wn