EttusResearch/uhd #442: run sudo sysctl -w net.core.rmem_max=2426666 is invalid!

shuimoshusheng, 2021-05-17:
on ubuntu 18.04.3,linux kernel is  5.4.0-73-generic, uhd is UHD-3.15.LTS ,and usrp x310 output：

[WARNING] [STREAMER] Requested send_frame_size of 8000 exceeds the maximum possible on this stream. Using 1472
[WARNING] [UDP] The recv buffer could not be resized sufficiently.
Target sock buff size: 2426666 bytes.
Actual sock buff size: 212992 bytes.
See the transport application notes on buffer resizing.
Please run: sudo sysctl -w net.core.rmem_max=2426666

when i run this command, is invaild!

neelpandeya, 2021-05-17:
What is the exact command that you try to run, and what is the exact output?

Please post this issue on the usrp-users mailing list.

mbr0wn, 2021-06-17:
Closing due to lack of activity. The command is fine -- there might be other circumstances going wrong here.

closed 2021-06-17 by mbr0wn