EttusResearch/uhd #443: pip install uhd  Could not find a version that satisfies the requirement uhd (from versions: none)

mstram, 2021-05-20:
win10 
python 3.9.5

>>> import pysdr

C:\Users\USER\AppData\Local\Programs\Python\Python39\lib\site-packages\pysdr\themes.py:7: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  black_and_white = Theme(json=yaml.load("""
Traceback (most recent call last):

  File "<stdin>", line 1, in <module>

  File "C:\Users\USER\AppData\Local\Programs\Python\Python39\lib\site-packages\pysdr\__init__.py", line 8, in <module>
    from pysdr.pyuhd_wrapper import usrp_source

  File "C:\Users\USER\AppData\Local\Programs\Python\Python39\lib\site-packages\pysdr\pyuhd_wrapper.py", line 3, in <module>
    from uhd import libpyuhd
ModuleNotFoundError: No module named 'uhd'

mbr0wn, 2021-05-20:
Looks like pysdr has uhd as a dependency -- we don't ship UHD through pip though. You might have to manually install UHD first, then PySDR through pip.

mstram, 2021-05-20:
Ok, thx.

I downloaded the https://files.ettus.com/binaries/uhd/latest_release/Windows-10-x64/uhd_4.0.0.0-release_Winx86_VS2019.exe installer

I'm not really sure which one I need.

Ran the installer, clicked on 'add to system path'.

Run python 3.9.5
 import uhd
ModuleNotFoundError: No module named 'uhd'

Is that just a "generic python import" error ?

I only use python ocassionally, and have a vague idea of setting up python paths / modules etc

manderseck, 2021-09-07:
@mstram Sorry for getting back to you so late. I fear for using the Python API you need to compile it as it is not included in the installer. Once you have compiled everything following the [manual](https://kb.ettus.com/Building_and_Installing_the_USRP_Open_Source_Toolchain_(UHD_and_GNU_Radio)_on_Windows) you might still stumble upon the "No module named 'uhd'" issue. To resolve this please check the [Python API instructions](https://files.ettus.com/manual/page_python.html). I'm closing this issue now. Feel free to re-open if there are any issues remaining.

closed 2021-09-07 by manderseck