EttusResearch/uhd #444: B205 mini transmitting file error with sending files not repeatedly

mym2424, 2021-05-22:
## Issue Description ##

When I use usrp B205 mini to transmit a binary file with module File Source and UHD:USRP Sink, the transmission  failed with the  File Source option 'repeat=No'. In WX GUI Scope Sink, only part of the waveform can be displayed and transmission stopped halfway through.
If I replace B205 by N210, there is no such problem here.

## Setup Details ##

 Ubuntu 16.04.7; GNU Radio Companion 3.7.11,; GNU C++ version 5.4.0 20160609; Boost_105800; UHD_3.11.0.HEAD-0-g13c32cef

## Expected Behavior ##

 With the option 'repeat=No', after the file is transmitted once, the scope sink should display nothing, just like I run the flow graph with N210.
## Actual Behavior ##

[image]
The figure above is the case ‘Repeat = Yes', and there is no problem. However, the figure below is the case ‘Repeat = No', the waveform stops and seems stuck.
No error message in terminal.

## Steps to reproduce the problem ##

Run this flow graph with usrp B205 mini. Change the ’Repeat' option to ‘No'.
[image]

## Additional Information 
I guess the issue comes from the USRP Sink module rather than the file source module. 
> ##

mbr0wn, 2021-09-17:
We're going to have to close this; the issue is not entirely clear, and also, the UHD and GR versions used are very old. I also can't really reproduce this.

Some notes, though:
- If you do `repeat=no`, then you are requesting the flow graph to terminate. This will also eventually freeze the GUI widget.
- What's displayed on the GUI widget is not necessarily what's being output on the device. I would reckon if you attach a scope to the analog output of the different USRPs, you would get the full transmission in both cases (N210 and B200).
- I can't explain why the behaviour is different between N210 and B200. Did you check this multiple times, to see if it's deterministic? If no, then maybe this was just random behaviour. If yes, then maybe the N210 subtly schedules transfers differently from B200 that interacts with the GR scheduler in a way that makes the amount of data plotted different.

mbr0wn, 2021-09-17:
If you still have an issue, you can reopen this, but please add some more detail so we can reproduce. Also, please try a more recent version of UHD.

closed 2021-09-17 by mbr0wn