EttusResearch/uhd #445: 'ssize_t' redefinition

gvanem, 2021-05-27:
Compiling UHD on Win-10 and using an *old* Python 3.6, caused
this error when compiling `sim_find.cpp`:
```
F:\ProgramFiler\Python36\include\pyconfig.h(174): error C2371: 'ssize_t': redefinition; different basic types
F:\gv\dx-radio\UHD\host\include\uhd/config.hpp(44): note: see declaration of 'ssize_t'
```

This Python 3.6 does not have a test for `_SSIZE_T_DEFINED` like your `include/uhd/config.hpp` has.
So maybe there is a need to rearrange some headers and/or test the Python version first?

Comparing with Python 3.9, it seems no problem since it uses `Py_ssize_t` through-out it's code.

mbr0wn, 2021-08-31:
Actually, I'm a bit baffled that this works even with newer versions of Python, because the Python-typedef for `ssize_t` is always unconditional, even in newer versions.

mbr0wn, 2021-08-31:
Some more detail:

- Python will, like UHD, see if we're on a non-POSIX system and then define `ssize_t` via typedef. This happens here:
https://github.com/python/cpython/blob/3.6/PC/pyconfig.h#L171
- This logic is unchanged between 3.6 and 3.9 (and even `main`). In 3.10, they removed all references to `ssize_t`, making this issue go away. But for us, that means 3.7 through 3.9 is a problem.
- Here's the Python commit: https://github.com/python/cpython/commit/c994ffe69553cbb34f1cea6a4494d6e7657f41b0. Sounds like it's a 32-bit issue.
- Somehow, the `SSIZE_T` in `BaseTsd.h` and whatever Python defines must be different. @gvanem are you doing 32-bit compiles or 64 bit?
- Python does not check if `ssize_t` is already defined. If it's already defined, it better match the Python version.
- UHD checks if `_SSIZE_T_DEFINED` exists, which I believe is a MinGW thing. Python defines `HAVE_SSIZE_T`, but without checking it.
- The only fix I see is to make sure that Python includes always come first, and that we also check for `HAVE_SSIZE_T`.

gvanem, 2021-08-31:
> are you doing 32-bit compiles or 64 bit?

32-bit mostly.

gvanem, 2021-08-31:
I see now way out of this except editing my `pyconfig.h` to say:
```
#ifndef _SSIZE_T_DEFINED
#define _SSIZE_T_DEFINED
  #ifdef MS_WIN64
  typedef __int64 ssize_t;
  #else
  typedef _W64 int ssize_t;
  #endif
#endif
```

mbr0wn, 2021-09-01:
We can also work around this in UHD, by including Python first, and then also checking for `HAVE_SSIZE_T` (see above).

mbr0wn, 2021-09-17:
@gvanem First, thanks again for this great report. However, we won't be fixing this. I hope this is not a big inconvenience for you. Some comments:
- 32-Bit compiles on Windows (and in this case, mixing 32 and 64 Bit) are not very common use cases.
- On Windows, users are usually not so tied to their Python version as they are on Linux, where this doesn't seem to be an issue.
  - Therefore, one fix is to use the latest Python version. Or, to patch Python like you did.
- The fix itself would be a bit messy, screw with our clang-format, and could easily break elsewhere.

Again, thanks for the report. I have renamed it and hope that others with this problem will find this thread.

closed 2021-09-17 by mbr0wn