EttusResearch/uhd #446: X300 FPGA Compatibility Issue (number 33, but got 36)

cloudlakecho, 2021-05-29:
I updated the latest FPGA image in USRP and installed latest Gnuradio, but when I try to use the USRP using a Python code I receive FPGA image conflict. 

## Issue Description ##
My Python  code (error section)
```
import uhd
.
.
.
self.uhd_usrp_source_0 = uhd.usrp_source(
            ",".join(("addr=10.1.11.202", "")),
            uhd.stream_args(
                cpu_format="fc32",
                args='',
                channels=list(range(0,1)),
            ),
        )
```

Error message
```
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

-- X300 initialization sequence...
-- Determining maximum frame size... 1472 bytes.
-- Setup basic communication...
Traceback (most recent call last):
  File "Rx_Warmup_Automated.py", line 172, in <module>
    main()
  File "Rx_Warmup_Automated.py", line 142, in main
    tb = top_block_cls()
  File "Rx_Warmup_Automated.py", line 64, in __init__
    channels=list(range(0,1)),
  File "/usr/lib/python3/dist-packages/gnuradio/uhd/__init__.py", line 125, in constructor_interceptor
    return old_constructor(*args)
  File "/usr/lib/python3/dist-packages/gnuradio/uhd/uhd_swig.py", line 3259, in make
    return _uhd_swig.usrp_source_make(device_addr, stream_args, issue_stream_cmd_on_start)
RuntimeError: RuntimeError: Expected FPGA compatibility number 33, but got 36:
The FPGA image on your device is not compatible with this host code build.
Download the appropriate FPGA images for this version of UHD.
```

## Setup Details ##
USRP X300
Gnuraido 3.8.2.0

## Steps to reproduce the problem ##
My entire Python code is at [Rx_Warmup_Automated.py](https://gist.github.com/cloudlakecho/e570b6a51c1bd91afaa808643d5cc3bb#file-rx_warmup_automated-py-L59)

## Additional Information ##
I could ping the USRP , execute `uhd_find_devices` and `uhd_usrp_probe --args addr=10.1.11.202` without a warning.

Is something incorrect in Python tool `uhd` or `gnuradio`?

dwidmann, 2021-06-15:
This sort of problem tends to arise when you have more than one installation of UHD on your machine, or if you have multiple machines accessing the same USRP. For reference, the first part of your error message says that you're using UHD 3.10.0.0 (which is very old now), and FPGA version 33. The USRP responded saying that it was flashed with FPGA version 36, which doesn't match version 33, so they can't communicate. You might want to make sure you have your paths set up correctly (e.g. PATH, LD_LIBRARY_PATH, PYTHONPATH ...)

cloudlakecho, 2021-06-15:
@dwidmann Thanks for reviewing the error and suggestion. As you mentioned, I check the UHD version and it is 3.10.0.0. 
```
$ dpkg-query -l | grep uhd
ii  libgnuradio-uhd:amd64                           3.8.2.0-0~gnuradio~bionic-3                      amd64        gnuradio universal hardware driver functions
ii  libuhd003.010.003:amd64                         3.10.3.0-2                                       amd64        universal hardware driver for Ettus Research products - library
ii  uhd-host                                        3.10.3.0-2                                       amd64        universal hardware driver for Ettus Research products - host apps
```
I will upgrade UHD and try again. 

Additionally, these are my environmental settings: 
```
$ echo $PATH
/home/ccho/anaconda3/condabin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
$ echo $LD_LIBRARY_PATH

$ echo $PYTHONPATH
/usr/lib/python3/dist-packages/gnuradio

```
If you think something incorrect in here, please let me know, thanks.

cloudlakecho, 2021-06-17:
@mbr0wn Thanks for the suggestion. I actually updated the FPGA image before put this issue. 

@dwidmann @mbr0wn I appreciate the suggestion. I upgrade the UHD