EttusResearch/uhd #447: uhd-4.0.0.0, gcc-10.2.0-r5, boost-1.76.0-r1, build error

rogovst, 2021-06-02:
## Issue Description ##
Build error

## Setup Details ##
Portage 3.0.18 (python 3.9.4-final-0, default/linux/amd64/17.1, gcc-10.2.0, glibc-2.33, 5.10.27-gentoo x86_64)
boost-1.76.0-r1

## Expected Behavior ##
Successful build

## Actual Behaviour ##
Build fails with error:
`
[ 66%] Building CXX object lib/CMakeFiles/uhd.dir/usrp/b100/clock_ctrl.cpp.o
cd /var/tmp/portage/net-wireless/uhd-4.0.0.0/work/uhd-4.0.0.0_build/lib && /usr/bin/x86_64-pc-linux-gnu-g++ -DBOOST_ALL_NO_LIB -DBOOST_ASIO_DISABLE_STD_EXPERIMENTAL_STRING_VIEW -DBOOST_ASIO_DISABLE_STD_STRING_VIEW -DBOOST_CHRONO_DYN_LINK -DBOOST_DATE_TIME_DYN_LINK -DBOOST_ERROR_CODE_HEADER_ONLY -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_SERIALIZATION_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -DBOOST_UNIT_TEST_FRAMEWORK_DYN_LINK -DHAVE_CONFIG_H -DUHD_DLL_EXPORTS -DUHD_LOG_CONSOLE_COLOR -DUHD_LOG_CONSOLE_LEVEL=2 -DUHD_LOG_FILE_LEVEL=2 -DUHD_LOG_MIN_LEVEL=1 -I/var/tmp/portage/net-wireless/uhd-4.0.0.0/work/uhd-4.0.0.0_build/include -I/var/tmp/portage/net-wireless/uhd-4.0.0.0/work/uhd-4.0.0.0/host/include -I/var/tmp/portage/net-wireless/uhd-4.0.0.0/work/uhd-4.0.0.0/host/lib/include -I/var/tmp/portage/net-wireless/uhd-4.0.0.0/work/uhd-4.0.0.0/host/lib/deps/flatbuffers/include -I/var/tmp/portage/net-wireless/uhd-4.0.0.0/work/uhd-4.0.0.0_build/lib/ic_reg_maps -I/var/tmp/portage/net-wireless/uhd-4.0.0.0/work/uhd-4.0.0.0/host/lib/convert -I/var/tmp/portage/net-wireless/uhd-4.0.0.0/work/uhd-4.0.0.0_build/lib/convert -I/var/tmp/portage/net-wireless/uhd-4.0.0.0/work/uhd-4.0.0.0/host/lib/usrp -I/var/tmp/portage/net-wireless/uhd-4.0.0.0/work/uhd-4.0.0.0/host/lib/usrp/common/ad9361_driver -I/var/tmp/portage/net-wireless/uhd-4.0.0.0/work/uhd-4.0.0.0/host/lib/usrp/common -I/var/tmp/portage/net-wireless/uhd-4.0.0.0/work/uhd-4.0.0.0_build/lib/transport/nirio/lvbitx -I/usr/include/libusb-1.0 -I/var/tmp/portage/net-wireless/uhd-4.0.0.0/work/uhd-4.0.0.0/host/lib/deps/rpclib/include -I/var/tmp/portage/net-wireless/uhd-4.0.0.0/work/uhd-4.0.0.0_build/_cmrc/include  -DNDEBUG -march=native -pipe -O2 -fvisibility=hidden -fvisibility-inlines-hidden -fPIC   -Wall -Wextra -Wsign-compare -std=gnu++14 -o CMakeFiles/uhd.dir/usrp/b100/clock_ctrl.cpp.o -c /var/tmp/portage/net-wireless/uhd-4.0.0.0/work/uhd-4.0.0.0/host/lib/usrp/b100/clock_ctrl.cpp
/var/tmp/portage/net-wireless/uhd-4.0.0.0/work/uhd-4.0.0.0/host/lib/usrp/usrp2/usrp2_impl.cpp: In member function ‘double usrp2_impl::set_tx_dsp_freq(const string&, double)’:
/var/tmp/portage/net-wireless/uhd-4.0.0.0/work/uhd-4.0.0.0/host/lib/usrp/usrp2/usrp2_impl.cpp:920:37: error: ‘boost::math’ has not been declared
  920 |     const int sign         = boost::math::sign(new_freq);
      |                                     ^~~~
/var/tmp/portage/net-wireless/uhd-4.0.0.0/work/uhd-4.0.0.0/host/lib/usrp/usrp2/usrp2_impl.cpp:921:46: error: ‘boost::math’ has not been declared
  921 |     const int zone         = std::min(boost::math::iround(new_freq / tick_rate), 2);
      |                                              ^~~~

`

## Steps to reproduce the problem ##
emerge =net-wireless/uhd-4.0.0.0

## Additional Information ##

mbr0wn, 2021-06-17:
Could reproduce on F34. Will have a fix coming soon.

mbr0wn, 2021-06-17:
Actually a dupe of #437

closed 2021-06-17 by mbr0wn