EttusResearch/uhd #450: uhd 4.1.0.0 can't build with error: ‘boost::math’ has not been declared

ZeroChaos-, 2021-07-01:
```
[ 26%] Building CXX object lib/CMakeFiles/uhd.dir/usrp/cores/spi_core_3000.cpp.o
cd /var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0_build/lib && /usr/bin/x86_64-pc-linux-gnu-g++ -DBOOST_ALL_NO_LIB -DBOOST_ASIO_DISABLE_STD_EXPERIMENTAL_STRING_VIEW -DBOOST_ASIO_DISABLE_STD_STRING_VIEW -DBOOST_CHRONO_DYN_LINK -DBOOST_DATE_TIME_DYN_LINK -DBOOST_ERROR_CODE_HEADER_ONLY -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_SERIALIZATION_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -DBOOST_UNIT_TEST_FRAMEWORK_DYN_LINK -DHAVE_CONFIG_H -DUHD_DLL_EXPORTS -DUHD_LOG_CONSOLE_COLOR -DUHD_LOG_CONSOLE_LEVEL=2 -DUHD_LOG_FILE_LEVEL=2 -DUHD_LOG_MIN_LEVEL=1 -I/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0_build/include -I/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/include -I/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/lib/include -I/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0_build/lib/include -I/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/lib/deps/flatbuffers/include -I/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0_build/lib/ic_reg_maps -I/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/lib/convert -I/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0_build/lib/convert -I/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/lib/usrp -I/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/lib/usrp/common/ad9361_driver -I/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/lib/usrp/common -I/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0_build/lib/transport/nirio/lvbitx -I/usr/include/libusb-1.0 -I/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/lib/deps/rpclib/include -I/usr/include/python3.9 -I/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/lib/deps/pybind11/include -I/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0_build/_cmrc/include  -DNDEBUG -Os -march=native -mtune=native -pipe -frecord-gcc-switches -fvisibility=hidden -fvisibility-inlines-hidden -fPIC   -Wall -Wextra -Wsign-compare -std=gnu++14 -o CMakeFiles/uhd.dir/usrp/cores/spi_core_3000.cpp.o -c /var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/lib/usrp/cores/spi_core_3000.cpp
/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/lib/usrp/cores/rx_dsp_core_3000.cpp: In member function ‘virtual void rx_dsp_core_3000_impl::set_mux(const uhd::usrp::fe_connection_t&)’:
/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/lib/usrp/cores/rx_dsp_core_3000.cpp:86:39: error: ‘boost::math’ has not been declared
   86 |             int if_freq_sign = boost::math::sign(fe_conn.get_if_freq());
      |                                       ^~~~
```

michaelld, 2021-07-01:
@ZeroChaos what version of Boost are you trying to use?

michaelld, 2021-07-01:
@ZeroChaos- trying to again with maybe the correct tag: which version of Boost?

ZeroChaos-, 2021-07-01:
dev-libs/boost-1.76.0

michaelld, 2021-07-01:
ACK ... let me try with that Boost version & see what's what ... I thought we fixed compatibility with Boost 1.76.0 with this release ... but maybe not

chenrui333, 2021-07-01:
Run into this same issue on the homebrew side as well
- Homebrew/homebrew-core#80331

michaelld, 2021-07-01:
@chenrui333 thanks for that info ... I'm setting up local testing to verify what is noted here & for Brew there

michaelld, 2021-07-01:
yes I see this exact same error too with UHD 4.1 release + Boost 1.76.0 ... sigh ... let me fix it up & put in an internal PR to fix it

ZeroChaos-, 2021-07-01:
Not for nothing, but a lot of issues around this have been reported for weeks, including fixes.  I can't help but think that opening up the dev process to the community a bit would really help things like this not happen....

michaelld, 2021-07-01:
Truth! Not within my control.

michaelld, 2021-07-01:
[patch-UHD-4.1_build_with_Boost_1.76.0.diff.txt](https://github.com/EttusResearch/uhd/files/6750118/patch-UHD-4.1_build_with_Boost_1.76.0.diff.txt)

michaelld, 2021-07-01:
PR submitted ... now we wait ... I just attached the -p1 patch that I believe correctly replicates the algorithm using std cmath

ZeroChaos-, 2021-07-01:
Hold up on a 4.1.0.1 release.  I think I have another build error but it's hard to read and it will take me some time to replicate -j1

michaelld, 2021-07-01:
OK ... if you do find one that is completely different than this one please open a new issue & provide details there ... most of R&D is on vacation & I will be too very soon ... so don't be surprised if you get no responses for a bit

ZeroChaos-, 2021-07-01:
Yeah, I will open a new bug for unrelated issue once I confirm it; just didn't want a second unbuildable release today.

ZeroChaos-, 2021-07-01:
I can't complete build, or run tests, but I think this issue is solved by the patch.

I have opened a new issue for the next build failure which I believe is not related https://github.com/EttusResearch/uhd/issues/451

loqs, 2021-07-01:
This was supposed to have already been fixed https://github.com/EttusResearch/uhd/issues/437#issuecomment-869510823
With this fix uhd now builds with boost 1.76.0.  There are just the left over headers remaining.
Did you not test with boost 1.76.0 before closing #437?

michaelld, 2021-07-01:
This issue was not fixed in the PR that addressed #437  . I do not know the extent of testing with various Boost versions. I tested with 1.71.0 until this PR came up, and then switched to 1.76.0 to create the PR addressing that issue. IMHO less Boost and more std is the best way to go when reasonably possible.

michaelld, 2021-07-01:
@loqs can you explain "There are just the left over headers remaining."? What headers? What PR or issue here?

loqs, 2021-07-01:
> @loqs can you explain "There are just the left over headers remaining."? What headers? What PR or issue here?

Sorry for the lack of detail:
https://github.com/EttusResearch/uhd/commit/3796175f32f0cc24c16809d8175d423bc7053de9 removed use of boost::math::iround and boost::math:special but left headers for those functions
https://github.com/EttusResearch/uhd/blob/3796175f32f0cc24c16809d8175d423bc7053de9/host/lib/usrp/usrp2/usrp2_impl.cpp#L950
and
https://github.com/EttusResearch/uhd/blob/3796175f32f0cc24c16809d8175d423bc7053de9/host/lib/usrp/usrp2/usrp2_impl.cpp#L951

Now those headers were not actually providing the implementation for the removed functions due to ordering but they are no longer needed.

I think has come about as #437 was titled as covering all uses boost/math/special_functions.hpp but the related PR only covered usrp2_impl.cpp.

michaelld, 2021-07-01:
gotcha ... OK yes that makes sense. Thanks for the explanation. We'll get a PR in place to remove those extraneous headers ... should not prevent the build at least ;)

mbr0wn, 2021-07-06:
I just pushed dc187b5597779b14d0de4087db4aa54752a15d07 which might help -- I don't have the full overview as of now.

mbr0wn, 2021-08-31:
I'm pretty sure this is done now. We're removing yet more `boost::math` just to be sure, but I don't think we have any open issues right now.

closed 2021-08-31 by mbr0wn