EttusResearch/uhd #451: uhd 4.1.0.0 unable to build tests

ZeroChaos-, 2021-07-01:
when testing is enabled, build failure results
```
[ 81%] Building CXX object tests/CMakeFiles/tx_streamer_test.dir/tx_streamer_test.cpp.o
cd /var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0_build/tests && /usr/bin/x86_64-pc-linux-gnu-g++ -DBOOST_ALL_NO_LIB -DBOOST_ASIO_DISABLE_STD_EXPERIMENTAL_STRING_VIEW -DBOOST_ASIO_DISABLE_STD_STRING_VIEW -DBOOST_CHRONO_DYN_LINK -DBOOST_DATE_TIME_DYN_LINK -DBOOST_ERROR_CODE_HEADER_ONLY -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_SERIALIZATION_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_TEST_DYN_LINK -DBOOST_TEST_MAIN -DBOOST_THREAD_DYN_LINK -DBOOST_UNIT_TEST_FRAMEWORK_DYN_LINK -DHAVE_CONFIG_H -DUHD_LOG_CONSOLE_COLOR -DUHD_LOG_CONSOLE_LEVEL=2 -DUHD_LOG_FILE_LEVEL=2 -DUHD_LOG_MIN_LEVEL=1 -I/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0_build/include -I/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/include -I/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/lib/include -I/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0_build/lib/include -I/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/tests/common -I/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0_build/lib/rfnoc -I/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/lib/rfnoc -I/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0_build/lib/ic_reg_maps -I/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/lib/deps/rpclib/include -I/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/lib/deps/flatbuffers/include -I/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0_build/_cmrc/include  -DNDEBUG -Os -march=native -mtune=native -pipe -frecord-gcc-switches -fvisibility=hidden -fvisibility-inlines-hidden   -Wall -Wextra -Wsign-compare -std=gnu++14 -o CMakeFiles/tx_streamer_test.dir/tx_streamer_test.cpp.o -c /var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/tests/tx_streamer_test.cpp
/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/tests/tx_streamer_test.cpp:145:10: error: ‘complex’ is not a member of ‘std’
  145 |     std::complex<uint16_t>*,
      |          ^~~~~~~
/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/tests/tx_streamer_test.cpp:10:1: note: ‘std::complex’ is defined in header ‘<complex>’; did you forget to ‘#include <complex>’?
    9 | #include <boost/test/unit_test.hpp>
  +++ |+#include <complex>
   10 | #include <iostream>
/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/tests/tx_streamer_test.cpp:145:10: error: ‘complex’ is not a member of ‘std’
  145 |     std::complex<uint16_t>*,
      |          ^~~~~~~
/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/tests/tx_streamer_test.cpp:145:10: note: ‘std::complex’ is defined in header ‘<complex>’; did you forget to ‘#include <complex>’?
/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/tests/tx_streamer_test.cpp:145:26: error: template argument 2 is invalid
  145 |     std::complex<uint16_t>*,
      |                          ^
/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/tests/tx_streamer_test.cpp:145:28: error: expected unqualified-id before ‘,’ token
  145 |     std::complex<uint16_t>*,
      |                            ^
/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/tests/tx_streamer_test.cpp:146:5: error: ‘int size_t’ redeclared as different kind of entity
  146 |     size_t,
      |     ^~~~~~
In file included from /usr/lib/gcc/x86_64-pc-linux-gnu/10.3.0/include/g++-v10/cstddef:50,
                 from /usr/include/boost/config/compiler/gcc.hpp:165,
                 from /usr/include/boost/config.hpp:39,
                 from /var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/include/uhd/config.hpp:10,
                 from /var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/include/uhd/exception.hpp:10,
                 from /var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/tests/common/../common/mock_link.hpp:10,
                 from /var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/tests/tx_streamer_test.cpp:7:
/usr/lib/gcc/x86_64-pc-linux-gnu/10.3.0/include/stddef.h:209:23: note: previous declaration ‘typedef long unsigned int size_t’
  209 | typedef __SIZE_TYPE__ size_t;
      |                       ^~~~~~
/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/tests/tx_streamer_test.cpp:147:32: error: invalid declarator before ‘>’ token
  147 |     boost::shared_array<uint8_t>>
      |                                ^~
/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/tests/tx_streamer_test.cpp: In member function ‘void test_send_one_channel_one_packet::test_method()’:
/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/tests/tx_streamer_test.cpp:210:22: error: ‘complex’ is not a member of ‘std’
  210 |     std::vector<std::complex<float>> buff(20);
      |                      ^~~~~~~
/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/tests/tx_streamer_test.cpp:210:22: note: ‘std::complex’ is defined in header ‘<complex>’; did you forget to ‘#include <complex>’?
/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/tests/tx_streamer_test.cpp:210:22: error: ‘complex’ is not a member of ‘std’
/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/tests/tx_streamer_test.cpp:210:22: note: ‘std::complex’ is defined in header ‘<complex>’; did you forget to ‘#include <complex>’?
/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/tests/tx_streamer_test.cpp:210:35: error: template argument 1 is invalid
  210 |     std::vector<std::complex<float>> buff(20);
      |                                   ^~
/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/tests/tx_streamer_test.cpp:210:35: error: template argument 2 is invalid
/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/tests/tx_streamer_test.cpp:211:33: error: request for member ‘size’ in ‘buff’, which is of non-class type ‘int’
  211 |     for (size_t i = 0; i < buff.size(); i++) {
      |                                 ^~~~
/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/tests/tx_streamer_test.cpp:212:13: error: invalid types ‘int[size_t {aka long unsigned int}]’ for array subscript
  212 |         buff[i] = std::complex<float>(i * 2, i * 2 + 1);
      |             ^
/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/tests/tx_streamer_test.cpp:212:24: error: ‘complex’ is not a member of ‘std’
  212 |         buff[i] = std::complex<float>(i * 2, i * 2 + 1);
      |                        ^~~~~~~
/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/tests/tx_streamer_test.cpp:212:24: note: ‘std::complex’ is defined in header ‘<complex>’; did you forget to ‘#include <complex>’?
/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/tests/tx_streamer_test.cpp:212:32: error: expected primary-expression before ‘float’
  212 |         buff[i] = std::complex<float>(i * 2, i * 2 + 1);
      |                                ^~~~~
/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/tests/tx_streamer_test.cpp:224:55: error: request for member ‘front’ in ‘buff’, which is of non-class type ‘int’
  224 |         const size_t num_sent  = streamer->send(&buff.front(), num_samps, metadata, 1.0);
      |                                                       ^~~~~
/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/tests/tx_streamer_test.cpp:229:14: error: ‘complex’ is not a member of ‘std’
  229 |         std::complex<uint16_t>* data;
      |              ^~~~~~~
/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/tests/tx_streamer_test.cpp:229:14: note: ‘std::complex’ is defined in header ‘<complex>’; did you forget to ‘#include <complex>’?
/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/tests/tx_streamer_test.cpp:229:30: error: expected primary-expression before ‘>’ token
  229 |         std::complex<uint16_t>* data;
[... 285 lines cut ...]
  640 |     num_sent = streamer->send(buff.data(), 10, metadata2, 1.0);
      |                                    ^~~~
/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0/host/tests/tx_streamer_test.cpp:647:9: error: ‘pop_send_packet’ was not declared in this scope
  647 |         pop_send_packet(send_links[0]);
      |         ^~~~~~~~~~~~~~~
[... cut ...]
      |                                                                                    ^
make[2]: *** [tests/CMakeFiles/tx_streamer_test.dir/build.make:82: tests/CMakeFiles/tx_streamer_test.dir/tx_streamer_test.cpp.o] Error 1
make[2]: Leaving directory '/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0_build'
make[1]: *** [CMakeFiles/Makefile2:3793: tests/CMakeFiles/tx_streamer_test.dir/all] Error 2
make[1]: Leaving directory '/var/tmp/portage/net-wireless/uhd-4.1.0.0/work/uhd-4.1.0.0_build'
make: *** [Makefile:182: all] Error 2
```

michaelld, 2021-07-01:
@ZeroChaos- if you add in the noted `#include <complex>` as noted in the error message does that allow UHD to fully build?

ZeroChaos-, 2021-07-01:
Affirmative, after that tests not only build, they pass.

michaelld, 2021-07-02:
> Affirmative, after that tests not only build, they pass.

wonderful! OK so that file just needs the header added to it, and along with the other PR to fix Boost 1.76.0 usage, we should be set ... the extra headers will be good to have removed, but not critical to the build failing. Thanks!

michaelld, 2021-07-02:
PR is in ... now we wait

closed 2021-07-02 by mbr0wn