EttusResearch/uhd #453: uhd-4.0.0.0: Broken Windows Support due to use of libusb_set_option()

tckillian, 2021-07-02:
## Issue Description ##
In UHD 4.0 and 4.1, none of the example executables will run (using UHD.dll) since libusb does not properly export the C++ symbol for libusb_set_option().

## Setup Details ##
UHD.dll relies on libusb-1.0.dll for the C++ symbols for libusb_set_option(). This symbol was briefly exported correctly in libusb 1.0.22-rc1 (when libusb_set_option() was first introduced), but by 1.0.22-rc2 it was broken and remains broken. There are no released binaries from libusb that will work with UHD 4.0 and 4.1, since they make use of that function.

Proper Symbol Export:
https://github.com/libusb/libusb/blob/v1.0.22-rc1/libusb/libusb-1.0.def#L156

Improper Symbol Export:
https://github.com/libusb/libusb/blob/master/libusb/libusb-1.0.def#L158

## Expected Behavior ##
Example binaries distributed with Windows releases of UHD run.

## Actual Behavior ##
Example binaries immediately terminate, without printing anything.

## Additional Information ##
I'd recommend reverting commit 0597c128475385daf9d0abafc60ec861c951eeaa  (by @mbr0wn ) until libusb fixes their issues and makes a new binary release.

manderseck, 2021-08-24:
Hi @tckillian,
Sorry for the delayed response. Since you've opened an [issue for libusb](https://github.com/libusb/libusb/issues/944) at the same day there has been some movement in the meantime there. In the current master of libusb (which you link as "improper Symbol Export") it now looks exactly like what you linked as "Proper Symbol Export". The latest release (1.0.24) however still has [a broken symbol export](https://github.com/libusb/libusb/blob/v1.0.24/libusb/libusb-1.0.def#L158). So they deprecated `libusb_set_debug()` but didn't export `libusb_set_option()` properly ever since then in a released version. By reverting commit 0597c12 we would still not be backwards compatible when libusb releases 1.0.25 and we re-do the change. @mbr0wn, which options do you see? Increase the `LIBUSB_API_VERSION` in your change, so that `libusb_set_option()` only gets used from libusb version 1.0.25 on?

mbr0wn, 2021-08-24:
Yeah, we can do that. However, this seems to work on Linux. I'm running 1.0.23, which has the minimum API version, and it doesn't choke.
Since we don't know for sure that 1.0.25 will fix this issue, how about we revert to using `libusb_set_debug()` for Windows builds?

mbr0wn, 2021-08-24:
@tckillian Also, thanks for this great report, and for even suggesting a fix.

manderseck, 2021-08-25:
@tckillian For fixing the issue I tried to reproduce the issue here on Win10 x64 but failed to see the issue. Are you working on 32 bit?

mcuee, 2021-09-05:
https://github.com/libusb/libusb/issues/944
I have re-opened issue #944. I am not so sure if the fix is correct or not and I am not so sure if libusb_set_option can be exported correctly or not due to the use of variadic.

mcuee, 2021-09-05:
> @tckillian For fixing the issue I tried to reproduce the issue here on Win10 x64 but failed to see the issue. Are you working on 32 bit?

It will only affect 32bit Windows. As far as I know, the intention in libusb project is to make sure the interoperability of MinGW and MSVC created DLL file.

mcuee, 2021-09-07:
@mbr0wn @tckillian Please help to give the latest git a try to see if that fixes this issue.

If yes, please try the following patch to libusb to see if it still works.
https://github.com/libusb/libusb/issues/944#issuecomment-913997187
```
diff --git a/libusb/libusb-1.0.def b/libusb/libusb-1.0.def
index e357ef5..648ae6a 100644
--- a/libusb/libusb-1.0.def
+++ b/libusb/libusb-1.0.def
@@ -155,7 +155,6 @@ EXPORTS
   libusb_set_interface_alt_setting
   libusb_set_interface_alt_setting@12 = libusb_set_interface_alt_setting
   libusb_set_option
-  libusb_set_option@8 = libusb_set_option
   libusb_set_pollfd_notifiers
   libusb_set_pollfd_notifiers@16 = libusb_set_pollfd_notifiers
   libusb_setlocale
```

mcuee, 2021-09-07:
I can not seem to be able to reproduce but I do not have the device. Using libusb-1.0.24 release (MinGW32 dll or VS2019 32bit DLL).

```
C:\Program Files (x86)\UHD\bin> .\uhd_usrp_probe.exe
[INFO] [UHD] Win32; Microsoft Visual C++ version 1925; Boost_107000; UHD_4.1.0.1-release
Error: LookupError: KeyError: No devices found for ----->
Empty Device Address

C:\Program Files (x86)\UHD\bin> $Env:LIBUSB_DEBUG=3
C:\Program Files (x86)\UHD\bin> .\uhd_usrp_probe.exe
[INFO] [UHD] Win32; Microsoft Visual C++ version 1925; Boost_107000; UHD_4.1.0.1-release
libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
libusb: error [load_usbdk_helper_dll] Failed to load UsbDkHelper.dll: [126] The specified module could not be found.
libusb: info [windows_init] UsbDk backend is not available
libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
libusb: error [load_usbdk_helper_dll] Failed to load UsbDkHelper.dll: [126] The specified module could not be found.
libusb: info [windows_init] UsbDk backend is not available
libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
libusb: error [load_usbdk_helper_dll] Failed to load UsbDkHelper.dll: [126] The specified module could not be found.
libusb: info [windows_init] UsbDk backend is not available
Error: LookupError: KeyError: No devices found for ----->
Empty Device Address
```

More detailed debug log.

<details>
<summary> full debug log </summary>

```
C:\Program Files (x86)\UHD\bin> $Env:LIBUSB_DEBUG=4
C:\Program Files (x86)\UHD\bin> .\uhd_usrp_probe.exe
[INFO] [UHD] Win32; Microsoft Visual C++ version 1925; Boost_107000; UHD_4.1.0.1-release
[timestamp] [threadID] facility level [function call] <message>
--------------------------------------------------------------------------------
[ 0.000290] [000035a8] libusb: debug [libusb_init] created default context
[ 0.000435] [000035a8] libusb: debug [libusb_init] libusb v1.0.24.11584
[ 0.000868] [000035a8] libusb: debug [get_windows_version] Windows 8 64-bit
[ 0.000998] [000035a8] libusb: debug [htab_create] using 1021 entries hash table
[ 0.017040] [000035a8] libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
[ 0.018940] [000035a8] libusb: debug [winusbx_init] libusbK DLL found, version: 3.1.0.0
[ 0.025112] [000035a8] libusb: error [load_usbdk_helper_dll] Failed to load UsbDkHelper.dll: [126] The specified module could not be found.
[ 0.025407] [000035a8] libusb: info [windows_init] UsbDk backend is not available
[ 0.025763] [000035a8] libusb: debug [usbi_add_event_source] add HANDLE 00000130 events 0
[ 0.025957] [000035a8] libusb: debug [usbi_io_init] using timer for timeouts
[ 0.026136] [000035a8] libusb: debug [usbi_add_event_source] add HANDLE 000002AC events 0
[ 0.025991] [000028d0] libusb: debug [windows_iocp_thread] I/O completion thread started
[ 0.026501] [000035a8] libusb: debug [libusb_get_device_list]
[ 0.026787] [000061d4] libusb: debug [libusb_handle_events_timeout_completed] doing our own event handling
[ 0.026973] [000061d4] libusb: debug [handle_events] event sources modified, reallocating event data
[ 0.027155] [000061d4] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() for 2 HANDLEs with timeout in 100ms
[ 0.050650] [000035a8] libusb: debug [winusb_get_device_list] allocating new device for session [92]
[ 0.050884] [000035a8] libusb: debug [winusb_get_device_list] allocating new device for session [93]
[ 0.051334] [000035a8] libusb: debug [get_api_type] driver(s): BTHUSB
[ 0.051515] [000035a8] libusb: debug [get_api_type] lower filter driver(s): ibtusb
[ 0.051579] [000035a8] libusb: debug [winusb_get_device_list] allocating new device for session [15]
[ 0.051950] [000035a8] libusb: debug [get_api_type] driver(s): WinUSB
[ 0.052086] [000035a8] libusb: debug [get_api_type] matched driver name against WinUSB
[ 0.052199] [000035a8] libusb: debug [winusb_get_device_list] allocating new device for session [23]
[ 0.052890] [000035a8] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.053101] [000035a8] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.053256] [000035a8] libusb: debug [winusb_get_device_list] allocating new device for session [72]
[ 0.053637] [000035a8] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.053837] [000035a8] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.053993] [000035a8] libusb: debug [winusb_get_device_list] allocating new device for session [73]
[ 0.054372] [000035a8] libusb: debug [get_api_type] driver(s): usbccgp
[ 0.054576] [000035a8] libusb: debug [get_api_type] matched driver name against Composite API
[ 0.054686] [000035a8] libusb: debug [winusb_get_device_list] allocating new device for session [74]
[ 0.055138] [000035a8] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_A36D&SUBSYS_091A1028&REV_10\3&11583659&0&A0' bus number 1
[ 0.055336] [000035a8] libusb: debug [enumerate_hcd_root_hub] assigning HCD 'PCI\VEN_8086&DEV_15DB&SUBSYS_091A1028&REV_02\71D9F51854B3020000' bus number 2
[... 294 lines cut ...]
[ 2.719084] [000035a8] libusb: debug [libusb_get_device_descriptor]
[ 2.719203] [000035a8] libusb: debug [libusb_get_device_descriptor]
[ 2.720334] [000035a8] libusb: debug [libusb_get_device_descriptor]
[ 2.720432] [000035a8] libusb: debug [libusb_get_device_descriptor]
[ 2.720524] [000035a8] libusb: debug [libusb_get_device_descriptor]
[ 2.720600] [000035a8] libusb: debug [libusb_get_device_descriptor]
[... cut ...]
[ 2.769774] [00001b2c] libusb: debug [usbi_wait_for_events] WaitForMultipleObjects() returned 258
[ 2.770144] [000035a8] libusb: debug [libusb_exit]
[ 2.770210] [000035a8] libusb: debug [libusb_exit] destroying default context
[ 2.770281] [000035a8] libusb: debug [usbi_remove_event_source] remove HANDLE 0000032C
[ 2.770367] [000035a8] libusb: debug [usbi_remove_event_source] remove HANDLE 000002AC
[ 2.770660] [000062d8] libusb: debug [windows_iocp_thread] I/O completion thread exiting
Error: LookupError: KeyError: No devices found for ----->
Empty Device Address
```

</details>

mcuee, 2021-09-07:
Same for the examples, tested using libusb-1.0.24 release DLL (VS2019 version and MinGW32 version). My Windows 10 is 64bit but that should not matter.

```
C:\Program Files (x86)\UHD\lib\uhd\examples> $Env:LIBUSB_DEBUG=3
C:\Program Files (x86)\UHD\lib\uhd\examples> .\usrp_list_sensors.exe

Creating the USRP device with:
[INFO] [UHD] Win32; Microsoft Visual C++ version 1925; Boost_107000; UHD_4.1.0.1-release
libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
libusb: error [load_usbdk_helper_dll] Failed to load UsbDkHelper.dll: [126] The specified module could not be found.
libusb: info [windows_init] UsbDk backend is not available
libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
libusb: error [load_usbdk_helper_dll] Failed to load UsbDkHelper.dll: [126] The specified module could not be found.
libusb: info [windows_init] UsbDk backend is not available
libusb: info [winusbx_init] WinUSB DLL available (with isoch support)
libusb: error [load_usbdk_helper_dll] Failed to load UsbDkHelper.dll: [126] The specified module could not be found.
libusb: info [windows_init] UsbDk backend is not available
Error: LookupError: KeyError: No devices found for ----->
Empty Device Address
```

manderseck, 2021-09-07:
@mcuee I tried to reproduce the issue in Windows 10 64bit but with UHD and all dependencies in 32bit. I tried it by compiling UHD from Git and I also installed the 4.1.0.1 release installer. I have libusb in version 1.0.23 and 1.0.24 and link to them via the PATH variable. My test device is a B210. I could not reproduce the issue in any of the possible combinations.
I'm still not completely sure if I need a native 32bit Windows box to reproduce the issue. Can you confirm this please?

mcuee, 2021-09-07:
@manderseck I also can not reproduce the issue as mentioned "Example binaries immediately terminate, without printing anything". Take note I do not have the device to test.

mcuee, 2021-09-07:
@tckillian Can you try again? Take note you need to have uhd.dll and libusb-1.0.dll in the path. If you do not have uhd.dll in path, it will indeed "immediately terminate, without printing anything".

manderseck, 2021-09-07:
@mcuee Sorry, I mixed you up with @tckillian and was surprised thus that you couldn't reproduce the issue anymore, either. So yes, I totally agree with you: uhd.dll and libusb-1.0.dll must be available through the Path variable, otherwise there is the chance that we won't print anything.
Since after all trials to reproduce the issue I was not able to do so, and we found an explanation for the observed behavior I will close the issue.
@tckillian Please re-open if you disagree with this.

closed 2021-09-07 by manderseck