EttusResearch/uhd #454: N300 works only in version UHD4.0

nielsvanetten, 2021-07-02:
title says it all

version 3.9 and 4.1 give runtime errors

mbr0wn, 2021-07-02:
@nielsvanetten The device version must match the host version (within limits, they do not always have to be identical). If your N300 is running version 4.0, then you can't use UHD 4.1 on your host (you will get a compat error). UHD 3.9 doesn't have N300 support, so it simply won't find it. UHD 3.15 does have N300 support, but uses a vastly different architecture, and will thus also report a compat error.

Please update your device if you want to use 4.1: https://files.ettus.com/manual/page_usrp_n3xx.html#n3xx_getting_started_fs_update

Or: https://files.ettus.com/manual/page_usrp_n3xx.html#n3xx_rasm_mender

closed 2021-07-02 by mbr0wn

nielsvanetten, 2021-07-02:
I did update the N300 to the latest version as per:
https://kb.ettus.com/USRP_N300/N310/N320/N321_Getting_Started_Guide