EttusResearch/uhd #466: N320 front end corrections API implementation is inconsistent with documentation

benjaminogles, 2021-07-20:
## Issue Description ##

UHD 4.1 provides implementations of the front end corrections API on the N320 Rhodium radio control interface. However, the implementations seem to mishandle the boolean argument in the overloads

- https://github.com/EttusResearch/uhd/blob/f633b497ff931c3c180d00148fc66e4feb8b8b6a/host/lib/usrp/dboard/rhodium/rhodium_radio_control.cpp#L621
- https://github.com/EttusResearch/uhd/blob/f633b497ff931c3c180d00148fc66e4feb8b8b6a/host/lib/usrp/dboard/rhodium/rhodium_radio_control.cpp#L639

which `multi_usrp` documentation suggests should enable automatic front end corrections. Instead, they cast the boolean arguments to a correction constant of `1.0` and forward them to the RX front end

- https://github.com/EttusResearch/uhd/blob/f633b497ff931c3c180d00148fc66e4feb8b8b6a/host/lib/usrp/cores/rx_frontend_core_3000.cpp#L152
- https://github.com/EttusResearch/uhd/blob/f633b497ff931c3c180d00148fc66e4feb8b8b6a/host/lib/usrp/cores/rx_frontend_core_3000.cpp#L169

## Setup Details ##

- UHD tag v4.1.0.1
- N320

## Expected Behavior ##

- `multi_usrp_rfnoc_ptr->set_rx_dc_offset(true, 0)` should enable automatic DC offset correction on the correct front end. https://github.com/EttusResearch/uhd/blob/f633b497ff931c3c180d00148fc66e4feb8b8b6a/host/lib/usrp/cores/rx_frontend_core_3000.cpp#L147 The USRP manual says that automatic DC offset correction on the N320 should be implemented as a single tap IIR filter.
- `multi_usrp_rfnoc_ptr->set_rx_iq_balance(true, 0)` should throw or log a warning and return.

## Actual Behaviour ##

- `multi_usrp_rfnoc_ptr->set_rx_dc_offset(true, 0)` sets the fixed DC offset correction to `1.0`.
- `multi_usrp_rfnoc_ptr->set_rx_iq_balance(true, 0)` sets the fixed I/Q imbalance correction to `1.0`.

## Steps to reproduce the problem ##

## Additional Information ##

I am guessing many users will generate calibration tables and avoid this issue altogether but I think it is worth fixing for the sake of applications that use the `multi_usrp` interface and expect consistent semantics between USRP types.

mbr0wn, 2021-08-23:
Many thanks for this detailed report. We'll check it out.

mbr0wn, 2021-08-24:
I think this should do the trick. Still needs some verification and a run through our CI and review pipeline.

```patch
From 5a31fbb480b2bd0bee89c106dabd7a53f0d1d2a7 Mon Sep 17 00:00:00 2001
From: Martin Braun <martin.braun@ettus.com>
Date: Tue, 24 Aug 2021 20:32:29 +0200
Subject: [PATCH 1/2] rh: Fix auto DC-offset correction and auto-IQ balance
 APIs

N320 doesn't have an automatic RX IQ balance correction, so that API is
removed.
The auto-DC offset correction was calling into the manual DC offset
correction code, which means auto-DC offset correction was never enabled
for N320.
---
 host/lib/usrp/dboard/rhodium/rhodium_radio_control.cpp | 7 +------
 host/lib/usrp/dboard/rhodium/rhodium_radio_control.hpp | 1 -
 2 files changed, 1 insertion(+), 7 deletions(-)

diff --git a/host/lib/usrp/dboard/rhodium/rhodium_radio_control.cpp b/host/lib/usrp/dboard/rhodium/rhodium_radio_control.cpp
index 5f80b84a1..6e24be21f 100644
--- a/host/lib/usrp/dboard/rhodium/rhodium_radio_control.cpp
+++ b/host/lib/usrp/dboard/rhodium/rhodium_radio_control.cpp
@@ -620,7 +620,7 @@ void rhodium_radio_control_impl::set_tx_iq_balance(
 
 void rhodium_radio_control_impl::set_rx_dc_offset(const bool enb, size_t)
 {
-    _rx_fe_core->set_dc_offset(enb);
+    _rx_fe_core->set_dc_offset_auto(enb);
 }
 
 void rhodium_radio_control_impl::set_rx_dc_offset(
@@ -636,11 +636,6 @@ meta_range_t rhodium_radio_control_impl::get_rx_dc_offset_range(size_t) const
         .get();
 }
 
-void rhodium_radio_control_impl::set_rx_iq_balance(const bool enb, size_t)
-{
-    _rx_fe_core->set_iq_balance(enb);
-}
-
 void rhodium_radio_control_impl::set_rx_iq_balance(
     const std::complex<double>& correction, size_t)
 {
diff --git a/host/lib/usrp/dboard/rhodium/rhodium_radio_control.hpp b/host/lib/usrp/dboard/rhodium/rhodium_radio_control.hpp
index a99289f21..36978ffa8 100644
--- a/host/lib/usrp/dboard/rhodium/rhodium_radio_control.hpp
+++ b/host/lib/usrp/dboard/rhodium/rhodium_radio_control.hpp
@@ -170,7 +170,6 @@ public:
     void set_rx_dc_offset(const bool enb, size_t chan = ALL_CHANS) override;
     void set_rx_dc_offset(const std::complex<double>& offset, size_t chan) override;
     meta_range_t get_rx_dc_offset_range(size_t chan) const override;
-    void set_rx_iq_balance(const bool enb, size_t chan) override;
     void set_rx_iq_balance(const std::complex<double>& correction, size_t chan) override;
 
     /************************************************************************
-- 
2.25.1
```

benjaminogles, 2021-08-24:
That patch looks right to me. I guess I'll let you close this issue when it is merged. Thank you for the fix!

mbr0wn, 2021-08-25:
FYI, the way we use GitHub, this issue will be auto-closed when we merge the fix, but it'll take a few more days before it goes public after that.

closed 2021-09-17 by mbr0wn