EttusResearch/uhd #468: always_inline fails on armv7l

gigs94, 2021-07-29:
## Issue Description ##

https://github.com/gnuradio/pybombs/issues/525
On armv7l the neon instructions should be forced to have the fpu set to neon.  Add the -mfpu=neon line below to the host/lib/convert/CMakeLists.txt file:
```
if(NEON_SIMD_ENABLE AND HAVE_ARM_NEON_H AND
   (${CMAKE_SIZEOF_VOID_P} EQUAL 4))
    enable_language(ASM)

    LIBUHD_APPEND_SOURCES(
        ${CMAKE_CURRENT_SOURCE_DIR}/convert_with_neon.cpp
        ${CMAKE_CURRENT_SOURCE_DIR}/convert_neon.S
    )
    SET ( CMAKE_CXX_FLAGS "-mfpu=neon" )
endif()
```

## Setup Details ##

## Expected Behavior ##

It should build without having to manually modify the build configuration

## Actual Behaviour ##

[  2%] Building CXX object lib/CMakeFiles/uhd.dir/convert/convert_with_neon.cpp.o
In file included from /home/pi/pybombs/bladeRF/src/uhd/host/lib/convert/convert_with_neon.cpp:10:
/usr/lib/gcc/arm-linux-gnueabihf/8/include/arm_neon.h: In member function \u2018virtual void __convert_fc32_1_sc16_item32_le_1_PRIORITY_SIMD::operator()(const input_type&, const output_type&, size_t)\u2019:
/usr/lib/gcc/arm-linux-gnueabihf/8/include/arm_neon.h:6740:1: error: inlining failed in call to always_inline \u2018float32x4_t vdupq_n_f32(float32_t)\u2019: target specific option mismatch
 vdupq_n_f32 (float32_t __a)
 ^~~~~~~~~~~
/home/pi/pybombs/bladeRF/src/uhd/host/lib/convert/convert_with_neon.cpp:27:33: note: called from here
     float32x4_t Q0 = vdupq_n_f32(float(scale_factor));
                      ~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~
In file included from /home/pi/pybombs/bladeRF/src/uhd/host/lib/convert/convert_with_neon.cpp:10:
/usr/lib/gcc/arm-linux-gnueabihf/8/include/arm_neon.h:10844:1: error: inlining failed in call to always_inline \u2018void vst1_s16(int16_t*, int16x4_t)\u2019: target specific option mismatch
 vst1_s16 (int16_t * __a, int16x4_t __b)
 ^~~~~~~~
/home/pi/pybombs/bladeRF/src/uhd/host/lib/convert/convert_with_neon.cpp:56:17: note: called from here
         vst1_s16((reinterpret_cast<int16_t*>(&output[i + 6])), D15);
         ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/pi/pybombs/bladeRF/src/uhd/host/lib/convert/convert_with_neon.cpp:10:
/usr/lib/gcc/arm-linux-gnueabihf/8/include/arm_neon.h:9027:1: error: inlining failed in call to always_inline \u2018int16x4_t vrev32_s16(int16x4_t)\u2019: target specific option mismatch
 vrev32_s16 (int16x4_t __a)
 ^~~~~~~~~~
/home/pi/pybombs/bladeRF/src/uhd/host/lib/convert/convert_with_neon.cpp:55:37: note: called from here
         int16x4_t D15   = vrev32_s16(D14);
                           ~~~~~~~~~~^~~~~
In file included from /home/pi/pybombs/bladeRF/src/uhd/host/lib/convert/convert_with_neon.cpp:10:
/usr/lib/gcc/arm-linux-gnueabihf/8/include/arm_neon.h:7552:1: error: inlining failed in call to always_inline \u2018int16x4_t vmovn_s32(int32x4_t)\u2019: target specific option mismatch
 vmovn_s32 (int32x4_t __a)
 ^~~~~~~~~
/home/pi/pybombs/bladeRF/src/uhd/host/lib/convert/convert_with_neon.cpp:54:36: note: called from here
         int16x4_t D14   = vmovn_s32(Q13);
                           ~~~~~~~~~^~~~~
In file included from /home/pi/pybombs/bladeRF/src/uhd/host/lib/convert/convert_with_neon.cpp:10:
/usr/lib/gcc/arm-linux-gnueabihf/8/include/arm_neon.h:7440:1: error: inlining failed in call to always_inline \u2018int32x4_t vcvtq_s32_f32(float32x4_t)\u2019: target specific option mismatch
 vcvtq_s32_f32 (float32x4_t __a)
 ^~~~~~~~~~~~~
/home/pi/pybombs/bladeRF/src/uhd/host/lib/convert/convert_with_neon.cpp:53:40: note: called from here
         int32x4_t Q13   = vcvtq_s32_f32(Q11);
                           ~~~~~~~~~~~~~^~~~~
In file included from /home/pi/pybombs/bladeRF/src/uhd/host/lib/convert/convert_with_neon.cpp:10:
/usr/lib/gcc/arm-linux-gnueabihf/8/include/arm_neon.h:1172:1: error: inlining failed in call to always_inline \u2018float32x4_t vmulq_f32(float32x4_t, float32x4_t)\u2019: target specific option mismatch
 vmulq_f32 (float32x4_t __a, float32x4_t __b)
 ^~~~~~~~~
/home/pi/pybombs/bladeRF/src/uhd/host/lib/convert/convert_with_neon.cpp:52:36: note: called from here
         float32x4_t Q11 = vmulq_f32(Q10, Q0);
                           ~~~~~~~~~^~~~~~~~~
In file included from /home/pi/pybombs/bladeRF/src/uhd/host/lib/convert/convert_with_neon.cpp:10:
[... 302 lines cut ...]
 vld1q_f32 (const float32_t * __a)
 ^~~~~~~~~
/home/pi/pybombs/bladeRF/src/uhd/host/lib/convert/convert_with_neon.cpp:30:36: note: called from here
         float32x4_t Q4  = vld1q_f32(reinterpret_cast<const float*>(&input[i + 2]));
                           ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/pi/pybombs/bladeRF/src/uhd/host/lib/convert/convert_with_neon.cpp:10:
/usr/lib/gcc/arm-linux-gnueabihf/8/include/arm_neon.h:10392:1: error: inlining failed in call to always_inline \u2018float32x4_t vld1q_f32(const float32_t*)\u2019: target specific option mismatch
 vld1q_f32 (const float32_t * __a)
 ^~~~~~~~~
/home/pi/pybombs/bladeRF/src/uhd/host/lib/convert/convert_with_neon.cpp:29:36: note: called from here
         float32x4_t Q1  = vld1q_f32(reinterpret_cast<const float*>(&input[i]));
                           ~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[2]: *** [lib/CMakeFiles/uhd.dir/build.make:471: lib/CMakeFiles/uhd.dir/convert/convert_with_neon.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:760: lib/CMakeFiles/uhd.dir/all] Error 2
make: *** [Makefile:163: all] Error 2
[ERROR] Build failed. See output above for error messages.
[ERROR] Problem occurred while building package uhd:
Build failed.
[ERROR] Error installing package uhd. Aborting.

## Steps to reproduce the problem ##

build on a raspberry pi 4

## Additional Information ##

drmpeg, 2021-08-02:
You can take care of this with the CMake command line. Here's an example tuned for a Beagleboard-X15.
```
cmake -DCMAKE_CXX_FLAGS:STRING="-march=armv7ve -mfloat-abi=hard -mfpu=neon-vfpv4 -mtune=cortex-a15 -Wno-psabi" -DCMAKE_C_FLAGS:STRING="-march=armv7ve -mfloat-abi=hard -mfpu=neon-vfpv4 -mtune=cortex-a15 -Wno-psabi" -DCMAKE_ASM_FLAGS:STRING="-march=armv7ve -mfloat-abi=hard -mfpu=neon-vfpv4 -mtune=cortex-a15 -g" ../
```

mbr0wn, 2021-09-03:
Combining this with https://github.com/EttusResearch/uhd/issues/350.

closed 2021-09-03 by mbr0wn