EttusResearch/uhd #472: USRP X310 LFTX, Twin TX DB

vogon82, 2021-08-04:
## Issue Description ##

## Setup Details ##

## Expected Behavior ##

## Actual Behaviour ##

## Steps to reproduce the problem ##

## Additional Information ##

vogon82, 2021-08-04:
As I understand I'm not able to set two TX streams. It looks like you were making headway into making two streams with RFnoc but that is just a future feature. Right now there isn't resources for configuring the TX into two different streamers. But at the same time there is for the RX (Twin RX DB). There is no Twin TX DB.  When and how can I get this 'FUTURE' feature, which is mentioned in your schemas and datasheets as 'PRESENT'???))

michaelld, 2021-08-05:
@vogon82 Can you point us to where in our documentation ("schemas and datasheets") this specific use case is listed as 'PRESENT'? Those need to be corrected, as clearly 2xTX using a single daughterboard is not currently supported.

vogon82, 2021-08-11:
When we bought the LFTX daughter boards, it was written in their specifications that they had two transmitting channels. We discussed this with the manager before the purchase. In correspondence with representatives of NI, they said that they are working on this and in the future the work of two channels will be implemented.
[image]
[image]

When using the driver version 3.15, the uhd_usrp_probe program shows the presence of two transmission channels. Version 4.0 is not available.

michaelld, 2021-08-11:
@vogon82 was the exchange with the NI / manager folks in email? If so, can you please send it to me : michael.dickens@ettus.com ... also send me the purchase info on those specific daughterboards ... this should not have happened & I'll see what I can track down internally to help you ... but I need that info before I can do anything & I make no promises that I -can- actually do anything!

mbr0wn, 2026-11-16:
I'm closing this -- double Tx per X310 radio is not on our roadmap, and this seems to go back to a miscommunication.

closed 2026-11-16 by mbr0wn