EttusResearch/uhd #473: [UHD 4.1.0.1/N320] RFNoC custom block -> Tx Radio does not transmit any signal

a-panella, 2021-08-05:
## Issue Description ##
Tx Radio randomly deasserts AXI Stream ready signal after some amount of time of continuous streaming. This results in invalid output on the TX/RX SMA of the Tx radio. See screenshot for clarification
[image]

## Setup Details ##
Device: Ettus N320
UHD Version/info: 
[INFO] [UHD] linux; 
GNU C++ version 9.2.0; 
Boost_107100; 
UHD_4.1.0.1-0-gf633b497

## Expected Behavior ##
I would expect ready to stay constantly ready, as when talking with Ettus support team, the Tx radios can support full throughput.

## Actual Behaviour ##
Observed in ILAs, the transmit radio will randomly stop asserting ready signal and it will never recover.

## Steps to reproduce the problem ##
Either build from my provided source files (OOT RFNoC module, source provided, and SW for RFNoC C++ API)
Alternatively, you can run my bitstream and .ltx probes file on Vivado hardware manager directly (on an N320 box only)
In any case, load the bitstream onto an N320, and then run the demonstrator_test binary to setup radios and observe the ILA signals

## Additional Information ##
This was tested with SPP = 200
I will assist in any way to fix this issue if required

wordimont, 2021-08-30:
@a-panella I just came across this issue. It's not clear to me which signals are shown in the ILA capture. Can you provide the signal paths for what's in the screen shots?

At the input to the radio block, it is expected for tready to deassert because RFNoC transfers data faster than the radio can accept. This is necessary to avoid underflow. Once its buffers fill up, the radio will deassert tready and you'll see tready start to toggle.

tready might also deassert if an error occurs, like underflow, or if it reaches the end of a burst. Adding the "state" in radio_tx_core to the ILA capture would show if this is happening.

Please let me know if you're still working on this issue.

a-panella, 2021-08-30:
@wordimont First of all, thanks for the response. The signals captured in the ILA are some of my internal signals like sine wave gen, and input tlast from the rx radio (all in green) The yellow and red signals are part of the Tx Radio's AXI-S interface. I would not have expected tready from the Tx Radio to deassert in the middle of streaming. To me this looks like a drop of throughput. 

Not shown here is the Rx Radio interface. My custom block, that sits in between Rx and Tx Radio, is ready to accept data all the time. The data coming into the custom block is valid every cycle @ 200 MHz (that is my user clock freq). 

Re: your comment that RFNoC transfers data faster than the radio can accept: Is this always the case? I was deliberately setting up my system so that my block and radios to have a input rate/output rate 1:1. It is sounding like I misunderstood the way the radios work - From what you are saying, the Tx radio buffer fills up and backs off the input stream, so that the read side has time to read. This wouldn't explain why I saw nothing on my CRO but I will test it again. I will also throw a probe onto the tx radio state to make sure its operating correctly

wordimont, 2021-08-30:
@a-panella There are multiple clocks in RFNoC. The data is transferred between blocks using the "rfnoc_chdr_clk", which I believe is 200 MHz on N310, and the 64-bit RFNoC bus will transfer two 32-bit samples per clock cycle. So that's 400 MS/s. The radio can only transmit as fast as the master clock rate, which I'm guessing is 200 MHz in your case, so 200 MS/s. So depending on which clock domain you're looking at, you may see tready deassert.

If your block is generating 1 sample per clock in the radio clock domain, which it sounds like you're trying to do, then you're right that you should end up with a 1:1 input/output rate. 

You mentioned the yellow and red signals are in the "Tx Radio" but can you be more precise? For example, which module and what are the Verilog signal names? I can't find any signals with "_channel_" in the name in the radio code.

If you continue debugging with the ILA, I suggest you take a look at s_axis_t* in radio_tx_core.v to see if its tready is dropping, or if "state" leaves the ST_TRANSMIT state unexpectedly. That module is basically the end of the chain before the data goes to the radio. Somewhere between your code and radio_tx_core is where the logic is deasserting tready.

a-panella, 2021-08-31:
@wordimont yes, I'm running a master clock @ 200MHz. The signal names are my own signal names (from within my custom block) which map directly to the radio axi stream interface. i.e the signals should be identical to the s_axis_t* signals you are referring to. After probing the tx_radio_core.v state machines:

Error state machine: 
- Never enters error state (good)
Transmit state machine:
- Stays transmitting (good)

> ...64-bit RFNoC bus will transfer two 32-bit samples per clock cycle. So that's 400 MS/s. The radio can only transmit as fast as the master clock rate, which I'm guessing is 200 MHz in your case, so 200 MS/s. So depending on which clock domain you're looking at, you may see tready deassert...

I notice that if I set my custom block to receive at 200 MHz, the Rx Radio always presents valid data. Does this mean that the receiver can provide 400 MSPS but the transmitter can't output at 400 MSPS? Additionally, if I look in my custom_block .yml definition, I have set NIPC to 1, and item_width to 32. Does this mean that I am wasting half of the throughput, because as you say above, RFNoC bus is 64 bits? Should I set NIPC to 2 then to concat two 32 bit samples?

It is now seeming like there never was anything wrong at all and I misinterpreted the way the tx_radio is supposed to work...

wordimont, 2021-08-31:
@a-panella The input to the block will be rfnoc_chdr_clk (200 MHz) at 2 samples/cycle. But the input to your user code within the block will be user0_clk (200 MHz also) but at 1 sample/cycle the way you configured the YAML, because NIPC is 1.

Yes, if you change NIPC to 2 then you'll get a 64-bit bus to your logic, with 2 samples per cycle. You can do this if you like, just keep in mind that it might use more logic and can make the logic more complicated.

a-panella, 2021-09-06:
@wordimont I've played around with my design a bit more and am still unable to get the 1:1 rates I would like. Could I please confirm that the following is sensible? To achieve the 1:1 input/output rate I would like, with my custom block running at 200MHz @ 1 NIPC where each item is 32 bits (sc32). I could:
- Change my custom block to run on 64 bits, as the RFNoC data bus is 64 bits wide
- Change my image core yml to have rfnoc_chdr clock @ 100MHz. Because RFNoC spec is 64 bits wide, this will be serialised to 32 bits at twice the rate (200MHz) which should be correct
- Use the DDC to halve the sampling rate on input, which should bring it down to 200 MHz.

My issue here is that nothing I am doing is affecting the tready : not(tready) ratio coming from the tx_radio...
If I am attempting to send twice as much data to the block how come its not ready only half the time? Looks closer to one third or one quarter readiness on the ILA.

What is the easiest way that I can input data @ 200 MSPS, process it at 200 MSPS and retransmit it at 200 MSPS ? Is there something here I am missing? The radio in question that is not ready enough of the time also has the following warning...
```
[WARNING] [0/Radio#1] Cannot change tick rate to 245.76 MHz, this clock is not configurable by the graph!
```
I am not trying to change tick rate to 245 MHz, I don't know what is causing the API to think I'm trying to do that. I have set my radio block controllers with set_rate(200e6,chan) and master clock is defaulting to 200 MHz because I am not setting it. Why is the block not able to handle this? How do I either increase the rate that the tx_radio can receive or decrease the input rate? I'm assuming to increase Tx Radio rate, I need to increase the master clock (I do not want to do this, so can I drop the input sampling rate instead)??

wordimont, 2021-09-06:
@a-panella To respond to your bullets:

- You can change your block to run on 64 bits.
- The RFNoC clocks are not currently configurable in the YAML. rfnoc_chdr_clk will always be 200 MHz (unless you manually modify how it's generated in the Verilog code) and radio_clk will always match the radio tick rate. Also, please note that rfnoc_chdr_clk and radio_clk aren't generated from the same source, so the 200 MHz of rfnoc_chdr_clk isn't exactly the same as the 200 MHz of the radio_clk.
- The data comes in from the RX port of the radio at 200 MHz, so I would expect the DDC bring the sample rate down to 100 MHz. And I assume you'd want a DUC to bring the rate back up to 200 MHz for transmit. I think this is possible if 100 MHz is an acceptable sample rate.

From your YAML, it looks like you have the RX radio connected to your block, and then your block does some processing, then sends the data on to the TX radio. So I'm assuming you want to be able to consume the RX input at the radio's sample rate and generate date to transmit at that same rate. Let me know if I'm misunderstanding your use case. 

To get the 1:1 rate you need to process data as fast as the RX radio generates it, and produce new data as fast as the TX radio consumes it. The radio block controls the rate, and your block just has to keep up. Ideally, you should be able to run your block's DSP on the radio_clk at 1 sample/cycle. However, some IP will insert bubble cycles, in which case those occasional idle cycles will cause your block to fall behind the radio sample rates. If that's the case, then you would need to change your block to process data as 64-bits (2 samples/cycle). If you process 2 samples/cycle then you don't have to use radio_clk, since rfnoc_chdr_clk will also be fast enough.

I think a simulation would be helpful in proving this out and debugging it. You could instantiate your block and the radio block to make sure the rates work out. You might look at rfnoc_block_radio_tb to see an example of simulating the radio block.

I agree that the tready ratio doesn't seem right. I would expect it to be toggling 50% of the time based on what you've said, so I think I'm missing something. Again, I think a simulation would shed some light on what's happening.

a-panella, 2021-09-06:
Thanks for the replies. Good to know that rfnoc_chdr_clk is not modifiable. I changed my yml to have chdr_clk at 100 MHz and got the same behaviour. So it makes sense that the frequency doesn't change. 

For our application we would rather a sample rate at 200 MHz. 

You are correct re: what my use case is, it receives data, processes it and resends data N cycles later. That's why I want the rates to match, because I want the latency to be constant (or as close to constant as the bubble cycles will allow). To achieve this, I've put a FWFT FIFO on the output of my block, feeding the troublesome Tx Radio. 

Thank you for suggesting simulation - I have simulated my block on its own, and verified it works. But I haven't simulated it with RFNoC blocks too, as I don't have a decent mixed-lang simulator. If you think it is worthwhile, I will simulate my design with the radios and Vivado XSim. Will let you know how I go.

Last note for now: I am using a user_clk at 200 MHz. If this is generated from a different source than the radios clk, then maybe I should swap over to radio_clk instead, as drift could be doing weird things. Would not expect there to be enough drift to cause this kind of issue in this short amount of time though, instead I would expect drift issues to occur over hours instead of seconds.

wordimont, 2021-09-06:
The YAML in your bug_report_example.zip shows radio_clk being connected to user0_clk on your block.
```
- { srcblk: _device_, srcport: radio, dstblk: custom_block0, dstport: user0 }
```
radio_clk is the same clock signal that goes to the radios for TX/RX, so that's the one you should use if you want to use 200 MHz and process 1 sample/cycle.

a-panella, 2021-09-06:
Is that not what the above line is doing? I would think the arbitrarily named input port named _user0_ is fed a clock signal which is sourced from srcport radio? Does it matter what name I give it? I thought if I feed it the radio clk it would work?

wordimont, 2021-09-07:
Yes, I was just confirming that this is how you should connect radio_clk to your block. And you are correct, renaming it on your block just changes the port name.

a-panella, 2021-09-07:
Ah, my mistake. I guess I will continue with simulation then

a-panella, 2021-09-17:
@wordimont I think I have found the reason for the weird sampling behaviour. I believe my .yaml file is somehow not being referenced.
My yaml file for my design *and* in the minimum reproducible example provided above has user0_clk freq set to 
```
- name: user0
    freq: "range(5e9,5e9)"
```
which I stupidly set because internally I was thinking 5e-9 for a 200Mhz clock... of course the N320 refclk probably does not support this so it is clipping to the max, which is near 250 MHz. This is why the radio is complaining about the tick rate being set to 245.x MHz even though I am not setting it to that. I thought I had found the issue when I corrected the freq to range(200e6, 200e6) but the bitfile produced the same results. I now think that there is something wrong with the way my .yaml file is imported into the build. If I cd into my rfnoc/build dir, I can do 
```
cmake -DUHD_FPGA_DIR=<path/to/uhd-fpga> ../ 
make 
make install
```
make install successfully copies the yml into the usr/local/share/rfnoc/... dir, but fails with copying my custom block files, as the regex in cmake.install file is set to find .v (verilog) files only, instead of .vhd. I notice that the noc_shell_custom_block.v and rfnoc_block_custom_block.v files are generated when I run 
```
rfnoc_create_verilog.py -c my/yaml/path/custom_block.yml -d my/rfnoc/path
```
but if I build the image core afterwards, the clock frequency is unchanged. Is the build process specified at https://kb.ettus.com/Getting_Started_with_RFNoC_in_UHD_4.0 still current for OOT modules?

So, my new question is, what is the correct way to make sure the yml is rolled into the FPGA build? does it matter that the .vhd files are not copied in?

wordimont, 2021-09-17:
The RFNoC clocks are not currently configurable in the YAML, so that "freq " setting is ignored by the FGPA build. If you connect the user0_clk to radio_clk then it will be 200 MHz in your case, since that's what radio_clk is.

I think the regex that captures which files get installed is here:
https://github.com/EttusResearch/uhd/blob/master/host/examples/rfnoc-example/CMakeLists.txt#L118

However, I don't think this affects the FPGA build process. Installing the example is not needed to build the FPGA, so I don't think it matters that the VHD files are not copied in.

When you build the FPGA, the rfnoc_image_builder prints which block YAML files are being referenced (see the `[INF] Adding block description from ...` lines). You can also add the `-l DEBUG` argument to rfnoc_image_builder. For example, if you're using CMake, modify this line:
https://github.com/EttusResearch/uhd/blob/master/host/examples/rfnoc-example/CMakeLists.txt#L141
so that the line becomes:
```
COMMAND ${_rfnoc_image_builder_exe} -l DEBUG -F ${UHD_FPGA_DIR} -y ${CMAKE_CURRENT_SOURCE_DIR}/${_rfnoc_image_core_SRC} -I ${CMAKE_SOURCE_DIR}
```
Then when you build the FPGA, it will also print at the beginning which image core YAML file is being used (See the `[DEB] Load configuration` line that gets printed). It will also print the location of the generated rfnoc_image_core.v file, which you can inspect to make sure it includes your block as you expect and verify that it is instantiated correctly with the clocks you expect connected.

a-panella, 2021-09-21:
@wordimont If we set the input/output rates on all DDCs and Radios to 245.x MHz (the maximum for this board) then we get no overflows. I think somewhere in the design, one of the RFNoC blocks is running at 245.x Mhz regardless of what you set it to...
For now I can work around this by running our system at 245 MHz.

a-panella, 2021-09-23:
**Closing issue: This was user error.** 
It was a little complicated. It seems that to do loopback, property propagation doesn't work. I was told about this a while ago by Jonathon Pendlum from Ettus, and was given a workaround (create DDC and DUC manually and set the rates) However, my .yaml was pretty much copied from the n320 image core, which was incorrect for running loopback with a custom block in the middle. Switching the yml connections around fixed intermittant signals from being fed in. The final issue (still existing) is that AFAIK in the N320 boxes, there is one block running at max rate or something. Even if you set the other blocks to some lesser sampling rate, eventually you get overflow/dropped data. This produces weird output signals. The work around we are currently doing is clocking our design to 247 MHz, which gets clipped back down to 245.x MHz. This stops the overflow/data rate issue.

Closing this issue as it is no longer relevant and was user error.

closed 2021-09-23 by a-panella

mbr0wn, 2021-09-28:
@a-panella Thanks for updating this thread.