EttusResearch/uhd #475: B210 streaming mode broken with UHD-4.1

donlzx, 2021-08-07:
## Issue Description ##
B210 stop streaming after an overflow occurs.

## Setup Details ##
GNU Radio UHD source module using B210 in streaming mode.

## Expected Behavior ##
With UHD-3.15.LTS, the receiver will continue to work even overflow occurs once in a while.

## Actual Behaviour ##
With UHD-4.1, the receiver will pause/stop because of not receiving any more IQ samples. Sometime the receiver even failed to start due to overflow immediately in the beginning.

## Additional Information ##
These two commits break the current B210 behavior.

* b200: temporary change to remove stream restart after overflow (https://github.com/EttusResearch/uhd/commit/0326bdcc964917501bb0111d3ce65467edb437f3)
* b200: flush receive buffer after getting an overflow (https://github.com/EttusResearch/uhd/commit/78ec29b6becd53f79a0ae282ae22c38bedb887d1)

muaddib1984, 2021-08-11:
i can confirm this bug in 4.0, 4.1 and master

michael-west, 2021-08-11:
@donlzx @muaddib1984 We are aware of the issue and already have a fix in progress.  It should be available within 2-4 weeks.

muaddib1984, 2021-08-11:
Great thank you. 

<end transmission>

closed 2021-09-01 by manderseck

manderseck, 2021-09-01:
@donlzx @muaddib1984 The fix is available at the master branch: https://github.com/EttusResearch/uhd/commit/175e553c44996925dc0f3e135f78716ce6f0e4ea
We are working on porting this back to the 4.0 and 4.1 branches.