EttusResearch/uhd #477: 0xc000007b error running Windows 10

careyer, 2021-08-13:
## Issue Description ##
Starting any application part of the USRP Hardware Driver (UHD™) Software (e.g. uhd_find_devices.exe or uhd_usrp_probe.exe) under Windows 10 (x64) gives the follwoing windows error: "Error 0xc00007b “Application was unable to start correctly”

## Setup Details ##

UHD 4.1.0.1 Release
Windows 10 21H1 (x64)
Intel Core i7-3770K

## Expected Behavior ##

Expecting e.g. uhd_find_devices.exe to find my B210 SDR and display all its information/configuration.

## Actual Behaviour ##

As of now starting e.g. uhd_find_devices.exe will first complain about a missing libusb-1.0.dll. Once this problem has been solved I continously get confronted with the "Error 0xc00007b “Application was unable to start correctly" windows error message. I am experiancing this very same problem on five PCs (only a single Laptop correctly runs the software).
I have tried my best to figure out the difference between the working and non-working configuration for weeks now but to no avail. 
[image]

## Steps to reproduce the problem ##

1.) Install USRP Hardware Driver (UHD™) Software
2.) Run e.g. hd_find_devices.exe

## Additional Information ##

AFAICT all nessesarry runtime environments are installed (they have all be re-installed from scratch respectively)
[image]

Contacted Ettus Research Support via Email/Webpage already but to no avail. 
My inquiry sadly got ignored... :-/

I am thankful for any adivce/help on this issue!
Thanks and kee up the excellent work
Best regards
Thomas

manderseck, 2021-08-23:
Hi @careyer,
Apologies for not getting back to you earlier. I have seen that kind of issues when UHD-related applications try to run with DLLs of the wrong bitness. I just managed to get this error back on my computer by changing my Path variable to point to the 32bit version of libusb. Also the boost libraries can potentially be linked incorrectly. Please check to where the Path variable points, change this if necessary and let me know if that resolves your issue. 
Best regards,
Martin

manderseck, 2021-08-25:
@careyer, I'm closing this issue for now. If you should still encounter this issue after checking the Path variable please feel free to re-open. Thanks!

closed 2021-08-25 by manderseck

careyer, 2021-08-25:
Thank you very much @manderseck for comming back to me on this issue! I am sorry for answering with some delay... I wasoin the go and had no access to my mail. I copied the libusb-1.0.dll from my working laptop (there wverthing works as expected) to my other computers (by overwriting the files there) and the problem is solved. 

However I have no idea where from I sourced that libusb-1.0.dll in the correct version. Do you know of a trustworthy source where to get the 32-bit and 64-bit libs from? I think I took mine from this https://wiki.spench.net/wiki/ExtIO_USRP but obviously it was the wrong one.

Maybe you can consider shipping the .dll with your software if the licence allowss it? I put it into the bin directory and it works like a charm.

Best regards and thanks again for stepping in! =)

manderseck, 2021-08-25:
Hi @careyer, in the [manual for building and installing UHD on Windows](https://kb.ettus.com/Building_and_Installing_the_USRP_Open_Source_Toolchain_(UHD_and_GNU_Radio)_on_Windows#LibUSB) the official source for libusb is mentioned: 
https://sourceforge.net/projects/libusb/files/libusb-1.0/
In these packages both bitnesses are included.

I need to check if shipping libusb together with the binary installer would be possible.

Best regards,
Martin

careyer, 2021-08-25:
Thank you very much!