EttusResearch/uhd #479: Can't activate external GPSDO and NMEA0183

msfu101, 2021-08-18:
Hi,
i*m new to usrp.
Got a USRP B210, an external GPSDO with external active GPS Antenna and NMEA0183.
Please, can you tell me how to activate the ext.GPSDO or the whole setup?
I googled it and took a look at the mailing list, but cant find an answer. i think it has to do with uhd_usrp_probe --args but dont know the Arguments. i know its a nooby question, but who wasnt a noob once?;)
thanks in advance!
msfu

manderseck, 2021-08-23:
Hi @msfu101,
According to the [manual](https://files.ettus.com/manual/page_gpsdo_b2x0.html#gpsdob_using) there is no need to activate a GPSDO manually, it will be detected during startup.
To verify that your hardware is setup properly you can run `sync_to_gps` which you can find in the examples directory (e.g. /usr/local/lib/uhd/examples/ in Linux). This will use the multi_usrp API calls `set_clock_source("gpsdo")` and `set_time_source("gpsdo")`. If this doesn't work you most likely have an issue in your hardware setup.

mbr0wn, 2021-08-24:
@msfu101 Closing this thread for now -- if @manderseck's answer doesn't suffice, please re-open.

closed 2021-08-24 by mbr0wn

msfu101, 2021-08-31:
Okay, i think there is really no need to activate the external GPSDO,it is active. 
But there is a Problem with the lock, the unit is locked and doesn't want to communicate with the b210 and the vk-172 ublox 7 USB-GPS-Mouse. The USB-Receiver is registered by the machine and sents a little data to /dev/ttyACM0, but no Locationdata or even PPS. And i can*t get the Data from the NMEA0183. 
Sometimes the User is the biggest problem... but i'll try hard. do you got a clue how to use this multi-usrp API Calls the right way and sync my GPS? 
Thank you

manderseck, 2021-08-31:
I recommend switching over to the [usrp-users mailinglist](https://kb.ettus.com/Mailing_Lists#usrp-users) for any further questions. Please name the components you are using in detail there. The issues you are having might be related to the GPS mouse but it doesn't look like an issue in UHD.

mbr0wn, 2021-08-31:
@msfu101 The only GPSDO that works directly with the B210 is this one: https://www.ettus.com/all-products/gpsdo-tcxo-module/ (Jackson Labs). Is that what you're using?

msfu101, 2021-09-09:
the USB_Receiver is a stempedu ublox 7 type vk-172 GPS Mouse from amazon
an usrp b210 
the external PLL_GPSDO Bg7tbl, a chinese product OXCO 10MHz 1PPS Sine Wave
a 5V active GPS/GNSS Antenna 
and there is a NMEA0183 from topgnss.com connected to the GPSDO  via RS232 DB9.
that*s the whole setup.
how can i get the whole GPS Setup to work together?
maybe its a faiilure in gpsd and not in the hardware...
i don't really know, 
i am trying to use Crocodilehunter, everything works but not the ext. GPSDO.

mbr0wn, 2021-09-10:
@msfu101 : Using separate hardware for GPS is outside the scope of what we can provide help with. I suggest getting your device going without the USRP before trying to use it together.