EttusResearch/uhd #490: N310: device arg 'rfic_digital_loopback' enables loopback regardless of value

gavingrant, 2021-09-22:
## Issue Description ##
Refer to N3xx device argument `rfic_digital_loopback` in: https://files.ettus.com/manual_archive/v3.14.1.1/html/page_usrp_n3xx.html#n3xx_usage_device_args

If the value is set to `0`, internal FPGA loopback is still enabled.

## Setup Details ##
- N310
- UHD-4.0.0.0, UHD-4.1.0.1

## Expected Behavior ##
Setting flag to false should disable internal FPGA loopback.

## Actual Behaviour ##
Internal loopback is enabled.

## Steps to reproduce the problem ##
uhd_usrp_probe --args addr=192.168.20.2,rfic_digital_loopback=0

manderseck, 2021-09-30:
@gavingrant Thanks for creating this issue. I have checked the behavior and you are right, it doesn't matter what you write as value to the `rfic_digital_loopback` argument, it will always resolve to True. I need to check if this is done wrong in other places, too. For the time being a workaround is to just omit the `rfic_digital_loopback` argument or to not set any value after the equal sign. You will see the difference in a shorter execution time because we don't need to run a full re-init sequence on the device if we don't enable the RFIC digital loopback.

closed 2021-10-12 by atrnati

manderseck, 2021-10-13:
@gavingrant This issue was closed automatically because I created a fix for this and it was merged into our internal repo. In a few days this will be merged into the public repo, too. So thanks again for reporting!