EttusResearch/uhd #492: rfnoc_radio_loopback with rx and tx from same block gives error: read property `scaling@OUTPUT_EDGE:0'

TSlivede, 2021-09-23:
## Issue Description ##

```
rfnoc_radio_loopback --rx-blockid '0/Radio#1' --tx-blockid '0/Radio#1' --duration 2
```
doesn't work and prints error:
```
[ERROR] [RFNOC::GRAPH::DETAIL] Adding edge 0/DDC#1:0 -> 0/DUC#1:0 without disabling property_propagation_active will lead to unresolvable graph!
Error: RfnocError: Adding edge without disabling property_propagation_active will lead to unresolvable graph!
```
If rx-blockid and tx-blockid are different, it works as expected (My real usecase is not the example. I am using all 4 rx-channels. The example just seems like the easiest way to reproduce the problem).

As the error message mentioned that property_propagation could be disabled, I tried modifying the example. I made my own copy of [rfnoc_radio_loopback.cpp](https://github.com/EttusResearch/uhd/blob/master/host/examples/rfnoc_radio_loopback.cpp), copied [`connect_through_blocks`](https://github.com/EttusResearch/uhd/blob/master/host/lib/utils/graph_utils.cpp#L86-L159) directly into it and changed the last [`graph->connect` call](https://github.com/EttusResearch/uhd/blob/master/host/lib/utils/graph_utils.cpp#L154) to disable property propagation (I set `skip_property_propagation` to `true`).

This solves the unresolvable graph error, but leads to another error:
```
Error: RuntimeError: AccessError: Attempting to read property `scaling@OUTPUT_EDGE:0' before it was initialized!
```
This error appears to occur at the `graph->commit` call.

## Setup Details ##

- USRP:
    - N310
    - UHD 4.0.0.0-0-g90ce6062 (installed via mender, downloaded with `uhd_images_downloader` from ettus)
    - FPGA-image: `uhd_image_loader --args "type=n3xx,fpga=HG"`
- host:
    - Ubuntu 20
    - UHD 4.0.0.0-154-gb061af4f

## Expected Behavior ##

I expected to be able to use rx and tx from the same radio control block in loopback configuration (receive and directly send out again), at least when I disable property propagation.

## Actual Behaviour ##

I got the errors mentioned above.

## Steps to reproduce the problem ##

<details><summary>Modified rfnoc_radio_loopback code:</summary>

```
//
// Copyright 2016 Ettus Research LLC
// Copyright 2018 Ettus Research, a National Instruments Company
//
// SPDX-License-Identifier: GPL-3.0-or-later
//

// Example UHD/RFNoC application: Connect an rx radio to a tx radio and
// run a loopback.

#include <uhd/rfnoc/block_id.hpp>
#include <uhd/rfnoc/mb_controller.hpp>
#include <uhd/rfnoc/radio_control.hpp>
#include <uhd/rfnoc_graph.hpp>
#include <uhd/types/tune_request.hpp>
#include <uhd/utils/graph_utils.hpp>
#include <uhd/utils/math.hpp>
#include <uhd/utils/safe_main.hpp>
#include <boost/format.hpp>
#include <boost/program_options.hpp>
#include <chrono>
#include <csignal>
#include <iostream>
#include <thread>
#include <numeric>

namespace po = boost::program_options;
[... 324 lines cut ...]
    while (not stop_signal_called) {
        std::this_thread::sleep_for(100ms);
        if (total_time > 0.0) {
            elapsed_time += 0.1;
            if (elapsed_time > total_time) {
                break;
            }
        }
    }

    // Stop radio
    stream_cmd.stream_mode = uhd::stream_cmd_t::STREAM_MODE_STOP_CONTINUOUS;
    std::cout << "Issuing stop stream cmd..." << std::endl;
    rx_radio_ctrl->issue_stream_cmd(stream_cmd, rx_chan);
    std::cout << "Done" << std::endl;
    // Allow for the samples and ACKs to propagate
    std::this_thread::sleep_for(100ms);

    return EXIT_SUCCESS;
}

```
</details>

## Additional Information ##

TSlivede, 2021-09-29:
I now have a workaround: I manually set the uninitialized scaling properties by hand. This is most certainly not the intended way, but it works.

To do this I copied [`node_accessor.hpp`](https://github.com/EttusResearch/uhd/blob/master/host/lib/include/uhdlib/rfnoc/node_accessor.hpp) and [`prop_accessor.hpp`](https://github.com/EttusResearch/uhd/blob/master/host/lib/include/uhdlib/rfnoc/prop_accessor.hpp) to a place where they are accessible for my project (definitly not recommended...) and included them into my modified `rfnoc_radio_loopback` example. Then directly after the line I already modified (`graph->connect(src_to_sep_id, src_to_sep_port, sep_to_dst_id, sep_to_dst_port, true);`, 6th to last line in `my_connect_through_blocks()` in posted code above) I added:

```
        const uhd::rfnoc::res_source_info src_edge_info{uhd::rfnoc::res_source_info::OUTPUT_EDGE, src_to_sep_port};
        const uhd::rfnoc::res_source_info dst_edge_info{uhd::rfnoc::res_source_info::INPUT_EDGE, sep_to_dst_port};
        auto fix_scaling = [&](auto id, auto edge){
            auto props_ptrs = uhd::rfnoc::node_accessor_t{}.filter_props(
                graph->get_block(id).get(),
                [&](auto x) { return x->get_id() == "scaling"
                                  && x->get_src_info() == edge; });
            for (auto &&prop_ptr : props_ptrs)
            {
                auto prop = dynamic_cast<uhd::rfnoc::property_t<double> *>(prop_ptr);
                if(!prop->is_valid()){
                    uhd::rfnoc::prop_accessor_t{}.set_access(prop, uhd::rfnoc::property_base_t::RW);
                    prop->set(1.0);
                }
            }
        };
        fix_scaling(src_to_sep_id, src_edge_info);
        fix_scaling(sep_to_dst_id, dst_edge_info);
```

As I said: this works, but whatever the actual problem is should probably be properly fixed somewhere in the uhd library.

mbr0wn, 2024-06-21:
`connect_through_blocks()` now also has that argument, and that's the correct way to handle this.

closed 2024-06-21 by mbr0wn

tomwts, 2024-09-30:
I experienced the same issue running the same example app on X310 and RFNOC 4.7 release. As long as the Rx and Tx radio block is different, the app ran fine, but it failed to initialize when Tx and Rx radio block Id is the same.
Is this issue still occurred on RFNOC 4.7 release?

[INFO] [UHD] linux; GNU C++ version 13.2.0; Boost_108300; UHD_4.7.0.0-0-ga5ed1872
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 8000 bytes.
[INFO] [X300] Radio 1x clock: 200 MHz
Using RX radio 0/Radio#1, channel 0
Using TX radio 0/Radio#1, channel 0
[ERROR] [RFNOC::GRAPH::DETAIL] Cannot create route from device:1/sep:2 and device:1/sep:2, no route was found!
Error: RuntimeError: Cannot create route from device:1/sep:2 and device:1/sep:2, no route was found!